fruit-bat/pico-zxspectrum #7: file missing

shaeon, 2022-03-09:
When i try compile i got this error:

PicoDVI/software/apps/zxspectrum/src/ZxSpectrumAy.h:3:20: fatal error: memory.h: No such file or directory

Can you upload memory.h file?

fruit-bat, 2022-03-09:
Hmm, should probably have been including string.h
I've changed a couple of lines, so pull the latest code and try again.
Let me know how you get on :-)

shaeon, 2022-03-09:
Now i got error in other file:

`/home/shaeon/pipico/PicoDVI/software/apps/zxspectrum/src/ZxSpectrumMenu.cpp: In member function  void ZxSpectrumMenu::loadDirAlphabetical(const char*, PicoSelect*)':
/home/shaeon/pipico/PicoDVI/software/apps/zxspectrum/src/ZxSpectrumMenu.cpp:433:3: error: 'sort' is not a member of 'std'
   std::sort(fnames.begin(), fnames.end(), [](const std::string& a, const std::string& b) -> bool {`

shaeon, 2022-03-09:
I add:

shaeon, 2022-03-09:
I add next line:

`#include <algorithm>`

And this fix the upper problem but i got again memory.h missing file error:

`/home/shaeon/pipico/PicoDVI/software/apps/pico-dvi-menu/src/PicoCharScreen.cpp:2:20: fatal error: memory.h: No such file or directory
compilation terminated.
`

fruit-bat, 2022-03-09:
Pull [pico-zxspectrum](https://github.com/fruit-bat/pico-zxspectrum) and [pico-dvi-menu](https://github.com/fruit-bat/pico-dvi-menu)

shaeon, 2022-03-10:
Now its compiling but i got nothing at screen. I need check it. I got pico-sorcerer-2 working well (even with usb-hub) but with zxspectrum not. 

I guest you can change url in readme from:
git clone git@github.com:fruit-bat/xxxxxxxxxxx
to 
git clone https://github.com/fruit-bat/

Thanks for all. :)

fruit-bat, 2022-03-10:
Just done a clean build and deployed to my RP2040 and it is working ok for me. I am building on a raspberry pi 64bit but have also built on a 32bit environment and it worked ok. Might be worth connecting to the serial port of seeing if it says anything useful. If not you could try dropping the uf2 in the chat and I could give it a go (not sure what sized doc you are allowed to attach here).

closed 2022-03-10 by shaeon