fruit-bat/pico-zxspectrum #8: error: 'CFG_TUH_ENDPOINT_MAX' undeclared here

shaeon, 2022-04-13:
Hello Again, 

I'm using your pre-built binary, and its work on my new pcb. I got sound, video, keyboard, all less sd-card access because i havent sd-card detect line. 

When i'm trying compile (in the past I got it) now i'm getting next error:
/pipico/PicoDVI/software/apps/pico-emu-utils/src/hid_host_info.c:26:38: error: 'CFG_TUH_ENDPOINT_MAX' undeclared here (not in a function)

[image]

whats i'm doing wrong?

Best Regards,
Shaeon.

shaeon, 2022-04-13:
I got compile addin in file pico-emu-utils/src/hid_host_info.c
after the line #include "hid_host_info.h" next:
`#define` CFG_TUH_ENDPOINT_MAX 16`

fruit-bat, 2022-04-13:
README has been updated. Let me know how you get on :-)

shaeon, 2022-04-14:
Its working on!

Here its my notes about compile, maybe you can change git clone git@ because i cant get it by git clone https:// (and other slash after):

`cd ~

mkdir pipico
cd pipico
git clone https://github.com/raspberrypi/pico-sdk

export PICO_SDK_PATH=~/pipico/pico-sdk
cd $PICO_SDK_PATH/lib/
mv tinyusb tinyusb.orig
git clone https://github.com/fruit-bat/tinyusb.git
cd tinyusb
git checkout hid_micro_parser

cd ~/pipico/
git clone https://github.com/Wren6991/PicoDVI

cd PicoDVI/software/apps
git clone https://github.com/fruit-bat/pico-zxspectrum.git zxspectrum
git clone https://github.com/fruit-bat/no-OS-FatFS-SD-SPI-RPi-Pico.git
git clone https://github.com/fruit-bat/pico-dvi-menu
git clone https://github.com/fruit-bat/pico-emu-utils

Change in file "zxspectrum/src/hw_config.c" next line:
.card_detected_true = 1 
with
.card_detected_true = -1

Add to CMakeLists.txt file next lines (others can comment out with #):
add_subdirectory(pico-dvi-menu)
add_subdirectory(pico-emu-utils)
add_subdirectory(zxspectrum)
add_subdirectory(no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI)

cmake -DPICO_COPY_TO_RAM=0 ..
make clean
make -j4 ZxSpectrum

cp apps/zxspectrum/ZxSpectrum.uf2 /media/pi/RPI-RP2/`

I used a recolocator pcb in the atlas proyect:
https://github.com/theexperimentgroup/Atlas-FPGA

I change fpga for pi pipico as main bus. I'll publish scheme next monday at work for my pipico "recolocator" 

[image]

[image]

Thanks for your work, share, and fast fixing! :)

closed 2022-04-14 by shaeon

shaeon, 2022-04-14:
I forgot tell you, you can use my notes if think can help more people, its free / open, you can do all you want with them :)

fruit-bat, 2022-04-21:
Thanks for the pics and info 🙂
At some point I will add build notes with https: as an alternative to git@