fruit-bat/pico-zxspectrum #10: cannot find hid_rip.h

xrayer, 2022-04-21:
Hi, I'm trying to compile the current sources with tinyusb and libdvi but I got error missing hid_rip.h referenced by hid_host_joy.h of pico-emu-utils - where can I get this file?

fruit-bat, 2022-04-22:
I think you are using the wrong version of TinyUSB. Have you replaced TinyUSB in the Pico SDK as per the first couple of paragraphs in the build instructions? If you have, check you don't have multiple copies in your build path.

xrayer, 2022-04-22:
I read about this in the readme but git failed to clone the repo for me:

git clone git@github.com:fruit-bat/tinyusb.git
Cloning into 'tinyusb'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

So I downloaded the zip package from this page but file is missing there...

fruit-bat, 2022-04-22:
Try cloning using https....

export PICO_SDK_PATH=~/pipico/pico-sdk
cd $PICO_SDK_PATH/lib/
mv tinyusb tinyusb.orig
git clone https://github.com/fruit-bat/tinyusb.git
cd tinyusb
git checkout hid_micro_parser

fruit-bat, 2022-04-22:
...or download the zip for the hid_micro_parser branch

xrayer, 2022-04-22:
Thanks for the hint, https works. I got it compiled, so closing now :)

closed 2022-04-22 by xrayer