fruit-bat/pico-zxspectrum #9: PICOMPUTER / RetroVGA support

bobricius, 2022-04-21:
Hi If you are able to add support my boards
I can send you this hardware
https://hackaday.io/project/184965-picomputer-zx-pocket-zx-spectrum (ST7789 IPS LCD)
https://hackaday.io/project/183398-retrovga-raspbery-pico-multi-retro-computer (VGA output)
Currently this board use MCUME https://github.com/Jean-MarcHarvengt/MCUME
But there is only Spectrum48K

fruit-bat, 2022-04-21:
If you can get some hardware to me I could have a go. As long as you don't mind if if takes a while. Circuit diagrams and any accompanying docs would be helpful.

fruit-bat, 2022-05-26:
Still work to be done on key mappings and documentation and other board types.

fruit-bat, 2022-05-28:
> Why not add ps2 support too?
> 
> Some examples of use it: https://github.com/tvlad1234/pico-ps2Driv https://github.com/lurk101/pico-ps2kbd/blob/master/example.c

Shaeon,

I've opened this as a new issue https://github.com/fruit-bat/pico-zxspectrum/issues/12 in case it gets lost in this conversation. Looks like a nice idea :-)

Thanks,

fruit-bat