fruit-bat/pico-zxspectrum #12: ps2 keyboard support

fruit-bat, 2022-05-28:
Some examples of use it:

https://github.com/tvlad1234/pico-ps2Driv
https://github.com/lurk101/pico-ps2kbd/blob/master/example.c

shaeon, 2022-05-28:
It will be Great! When you add support ps2 I'll use on our board!

fruit-bat, 2022-06-26:
I've ordered some ps/2 breakouts so I can have a look at this. It may require some extra circuitry as some keyboards/mice had internal pull up resistors on the open collector outputs which would damage the rp2040 (the pull up would be to 5v). Maybe just a resistor and a zener? I'll have a think.

shaeon, 2022-06-26:
I used next adapter on many fpga 3,3v gpio without problems:

https://www.waveshare.com/wiki/VGA_PS2_Board

I contact to waveshare and they told me i need used 3,3v. This scheme it's that i used on my mother board with pi pico :)

fruit-bat, 2022-06-26:
Ah, thanks for that. I take it you still power your keyboard at 5v?

I think most PS/2 keyboards do not contain a pull-up to 5v, and when this is the case the Zener diode is redundant (you would know if your keyboard has a pull-up resistor cause it would have most likely broken the GPIO pin on your RP2040/fpga).

Are you intending to use HDMI or VGA for the video?
If you let me know your setup I could have a go at a making a build target that matches your circuit.

shaeon, 2022-06-26:
Almost all ps2 keyboards that i used work well with 3,3V. I would like setup ps2 keyboard and HDMI output video. 

Waveshare ps2+vga adapter usually i set for testing environment.

fruit-bat, 2022-07-08:
Parts arrived and successfully receiving key-codes from the keyboard using:
https://github.com/lurk101/ps2kbd-lib
Need to map PS/2 keys to HID keycodes etc...

fruit-bat, 2022-07-10:
There is now basic support for PS/2 keyboards (set 2), with some caveats:
- Using USB and PS/2 keyboards simultaneously make not work quite right. This is fixable when I get time.
- CAPS/Num lock do nothing
- Indicator LEDs are not used

I've uploaded the new HDMI uf2 files containing the new behavior.

Let me know how you get on with it... which may encourage me to finish the code!

fruit-bat, 2022-07-27:
Hope you are happy with the PS/2 driver. I'll close this issue but feel free to open new issues if you are having trouble with it.

closed 2022-07-27 by fruit-bat