fruit-bat/pico-zxspectrum #24: Regarding Audio output

badrianiulian, 2022-06-12:
I think you should change the circuits for the audio pwm pin and discard the digital pin.
Look at the audio [circuit](https://othermod.com/raspberry-pi-zero-audio-circuit/) for Raspberry Pi.
By doing some math with a [RC High Pass Filter calculator](http://www.learningaboutelectronics.com/Articles/High-pass-filter-calculator.php) they allow frequencies over ~106 Hz and with a [RC Low Pass Filter calculator](http://www.learningaboutelectronics.com/Articles/Low-pass-filter-calculator.php) they allow frequencies under ~17871 Hz.
Humans can detect sounds in a frequency range from about 20 Hz to 20000 Hz.
The BAV99 are used for ESD protection (electrostatic discharges).
I've been following your project and been waiting for components to build it.

fruit-bat, 2022-06-12:
Ok, so the  filter components are most likely non-optimal. They were chosen by guessing and picking from the pile I had lying about. Clearly it's missing the decoupling cap and load resister... The suggested filter may sound better and I will give it a go if I can find the bits 🙂

...but that's not why I split the audio on to 2 pins. The AY8912 generates a signal that is encoded with pwm. Previously, the AY8912 signal would then have a value added to take into account the beeper level (1/0).  Now for pure tones on the beeper this works fine, but some spectrum programs used pwm on the beeper to created polyphony (Equinox). Now we are mixing mixing 2 high frequency signals (and losing phase from the beeper signal); the effect is messy, with audible beating and scratching sounds. 
Adding an extra pin to carry the beeper state means pwm implemented in spectrum programs sounds better (probably not perfect though).
In the mean time I have added a compile time switch so you can revert to the single pin output if you wish. 
Hope you have fun building your own version and please let me know how you get on.

badrianiulian, 2022-06-12:
OK... I now understand what you did... the mechanics behind the circuitry of the spectrum is still an unknown to me since I didn't deep dive into the matter so that's my bad. I'm just fond of the memories of messing with the old z80 box I had when I was a kid.
The single pin output will be a regression so I think you shouldn't add that compile time switch. Good (and even great) fidelity is a must as I understand and I fully support.
You basically mixed two audio signals: pwm on GP20 with analog audio signal (`getAnalogueAudio()`) and digital on GP21 with digital audio signal (`getBuzzer()`).
So the RC Band Pass filter can go on the analog side and then comes in the 15k resistor with the buzzer.
But as I understand from the schematic of the AY8912, it had 3 analog outputs... ? So in theory, stereo sound and buzzer could be possible by adding another GPIO pin? If that were an option then each channel would have an RC Band Pass Filter with two 15k resistors from the buzzer pin.
If my marbles are off then please knock them back...
Thank you for doing this project! My pico's are just begging to be used :)) and I can hardly wait to introduce the kids to the old BASIC... with no additional bloatware from the current OSs.

fruit-bat, 2022-06-13:
Yes, quite right, the AY8912 can output left and right audio. So the sound could be brought out to 3 pins.

badrianiulian, 2022-06-13:
As I can see in the `ZxSpectrum.h` code, there are the 3 (A B C) analog audio channels witch are mixed together.
Why not use 3 GPIO pins to output the audio and mix them like [this in the top left corner](https://hw.speccy.cz/pics/ay-interface.png) ?
Are there any limitation to the Pi Pico that I'm not aware of? If there aren't, there would be a little drop on processing power with audio mixed outside of the board... right?

fruit-bat, 2022-06-13:
I think the Pico can do lots of pwm signals. In terms of performance, I'm not sure what the effect would be it will require setting the period on 3 channels; hopefully this is very quick. I do like the idea sending A,B,C & Buzzer to different pins. As a side effect, each pin will have a lower dynamic range and hence the pwm window can be shorter (which is good). Might add this as an option, particularly if you fancy having a play with mixer circuits.

badrianiulian, 2022-06-15:
Been busy the last two days... I'm still amazed about the elegance of short C syntax.
Finally saw the modification... still doing the same thing but with a different approach witch separates the ABC channels for future mods.
Between the HDMI version and all the other 5 experimental versions, having them side to side, are there any pins left unused?

fruit-bat, 2022-06-15:
Each version has its own pin assignments. The boards by bobricius (Picomputer) do not have any spare pins as lots are used in they key matrix; so they will stick with 1 pin sound. The HDMI version has a few spare pins and if required, I could add new targets for vga/LCD that don't have a key matrix. As an aside, the experimental  versions are working reasonably well.

badrianiulian, 2022-06-17:
I noticed the new changes with 4 pin audio: 20=A, 21=BUZ, 26=B and 27=C.
I'm still waiting for those components but in the meanwhile I can't find the `happy` or `excited` buttons for those commits.
I can hardly wait to test all this setup.

fruit-bat, 2022-06-17:
I've linked the pull request so you can see how it's coming along. I have given it a try with the separate A,B,C channels, mixing back to mono; it sounds quite nice connected to some pc speakers. Glad you are looking forward to trying it.

fruit-bat, 2022-06-18:
There are now 1,2 & 4 pin audio variants of the HDMI build. The documentation is lagging behind but I will sort it at some point. Let me know how you get on with the analogue filters. If you are happy to share your successful designs I would like to add them to the README.

fruit-bat, 2022-06-23:
I've made some improvements to the audio emulation, in particular the noise generation and envelopes.  Also, I've added the filter I tried it with, which albeit mono, sounded really good. I think it's ready to have a play with.

badrianiulian, 2022-08-04:
I finally found some time this week to assemble the items I ordered. I worked with an old PS/2 HP Keyboard witch was hacked a while back with a Pi Zero W (had an usb hub and an PS/2-USB adapter).
For the pico I used ZxSpectrumBreadboardHdmi4PinAudio.uf2 from [https://github.com/fruit-bat/pico-zxspectrum/commit/2b3ed9ab9f93a68bfb1f1ec387931061801b6bd0](https://github.com/fruit-bat/pico-zxspectrum/commit/2b3ed9ab9f93a68bfb1f1ec387931061801b6bd0)
I had some fun with a test PCB and some recovered SMD resistors (8 x 220 & 4 x 10K).
For the SD card module and for the HDMI connector I used some kynar wire. 
For the audio part I bought a Mono Power Amplifier Board HXJ8002.
On each of the 4 audio pins I added 10K resistors and later I added another 47K resistor before going into the amplifier since it was too loud without it (white wire in the picture is without this 47K resistor).
I also used a 8 ohm 1,5W speaker recovered from an old laptop.
With the 47K resistor, sound decibels are ~65db witch is reasonable and also protects the speaker in the long run.
Maybe in the near future I'll make just one small case witch contains only the pico with a pcb and the speaker (also some volume control pot), that can be used with an external printed keycaps keyboard like this beauties: [https://github.com/zxbooka/zx-wasd](https://github.com/zxbooka/zx-wasd)

[image]
[image]
.

badrianiulian, 2022-08-09:
I'm away for a few days but as soon as I can, I'll add a schematic for the setup. As I said earlier, I will try and build a DIY PCB in KiCad for a small enclosure that would fit the Pico with wren's DVI output, the SD Card module, the mono audio amplifier and an 4 port usb module... stuff anyone with a little patience can order and solder.
For a stereo amplifier, I found (I have the pdf documentation somewere at home) that there were two ways of connecting to obtain stereo output... ABC (AB and BC) and ACB (AC and CB). As I can see, the common can be either B or C. For the purpose of removing a hardware switch between the two channels, the switching could be achieved by using the F3 mute key by adding another state maybe? I don't actually know games or apps for the ZX Spectrum that use this functionality... If you know some, please do tell. Maybe mono is all we really need...

badrianiulian, 2022-08-19:
Haven't included the USB hub and the keyboard (with the PS2 adapter) but this is a first draft:
[image]

fruit-bat, 2022-08-20:
That's great, and will really help if others fancy building it :-)

In the DAC version of the emulator I mix A+B, B+C and it sounds fine. 

This is supposedly circuit diagrams for the 128k...

https://www.8bit-wiki.de/fileadmin/8bit-wiki/Sinclair/ZX_Spectrum/_zx128/manual/ZX%20Spectrum%20128%2B%20Schematics.pdf

and it is mono... I think anyhow... I can see the mixer but not the audio output.

fruit-bat, 2022-08-20:
And on the +2 they even drop the mixing resistors...

https://zxnet.co.uk/spectrum/schematics/Z70830.pdf

badrianiulian, 2022-08-20:
In my understanding, the ABC/ACB mixing helps only with some programs and interfaces that were developed to actually use these chip's(AY-3-8912) features.
Found this link a while back that explains it a little better: [http://benophetinternet.nl/hobby/sales/Stereo audio v2014 Manual.pdf](http://benophetinternet.nl/hobby/sales/Stereo%20audio%20v2014%20Manual.pdf)