fruit-bat/pico-zxspectrum #37: Support for original ZX Spectrum Keyboard?

bajinchi, 2022-07-22:
Hi Guys!

LOVE THIS PROJECT!

I´m wondering if it´s a way to use the original keyboard of a ZX Spectrum directly wo the Raspberry Pico for more enjoy of the project!?

I´ve one to make tests if needed.

Keep up going the good work!

Regards

fruit-bat, 2022-07-23:
Quite possibly. Is it a 48k or 128k keyboard?

bajinchi, 2022-07-24:
It´s a 48K Keyboard

fruit-bat, 2022-07-25:
Ok, so this is mainly for my benefit, with a question mixed in...

The 48k keyboard is an 8 by 5 matrix, which needs 13 GPIO pins...

http://www.breakintoprogram.co.uk/hardware/computers/zx-spectrum/keyboard

So the question is what mix of features can be supported with 28 pins along side:
- SD Card 4 pins
- HDMI 8 pins
- VGA 5 pins
- PS/2 keyboard 2 pins
- Audio 1 - 4 pins (DAC 3 pins, should really use at least 2 pins for PWM audio) 
- Serial debug 2 pins

...but do you have preferences out of the above list?

Also, it's possible that a second Pico could be used to scan the keyboard.

emilen72, 2022-07-27:
Hi,
I'm interested too in the use of real keyboard, mine is ZX+

My prefs is to get some real stuff (keyboard, audio, deck i/o, joystick and cvbs video) plus SD card

Great job guy... best regards

bajinchi, 2022-07-27:
> Ok, so this is mainly for my benefit, with a question mixed in...
>
> The 48k keyboard is an 8 by 5 matrix, which needs 13 GPIO pins... [...]

Good question...

Must have (IMO)

SD Card
Audio
HDMI

At least

fruit-bat, 2022-07-27:
> Good question...
>
> Must have (IMO)
>
> SD Card Audio HDMI
>
> At least

Well, I think that will squeeze in. I'll have a think about it but I have promised some work on another board so it may take a while to make the new drivers.  It's worth noting here the tricky bit is mapping all the 'normal' keys onto the Spectrum keyboard so the menu can be used; the menu uses USB HID key values and will require its own matrix scanning routines. If you intend to operate the menus from a second usb keyboard it is a much simpler problem... although maybe a messier solution.

fruit-bat, 2022-07-27:
> Hi, I'm interested too in the use of real keyboard, mine is ZX+
> 
> My prefs is to get some real stuff (keyboard, audio, deck i/o, joystick and cvbs video) plus SD card
> 
> Great job guy... best regards

Correct me if I am wrong but I think the 48k & 128k keyboards use the same connectors (and the extra keys are made by connecting multiple points on the matrix)... so the same software should be good for both.

CVBS is interesting and I have seen some grey-scale examples on the pico. Not seen a color demo as yet.  I will raise this as a separate issue as it is of academic interest!

fruit-bat, 2022-07-31:
[image]

Waiting on some hookup wires now.

bajinchi, 2022-08-01:
> [image]
> 
> Waiting on some hookup wires now.

Oh so nice!

fruit-bat, 2022-08-13:
I have a circuit, now just(!) need some software :-
[image]
)

bajinchi, 2022-08-14:
> 

Ohhhh yessssssssss! The good is coming!

fruit-bat, 2022-08-14:
There is some code on a branch, including an executable:
https://github.com/fruit-bat/pico-zxspectrum/tree/feature/zxkeyboard
https://github.com/fruit-bat/pico-zxspectrum/blob/feature/zxkeyboard/uf2/ZxSpectrumBreadboardHdmiKbd1PinAudio.uf2

It all works fine, as far as I can tell, but as yet there is no way to open the emulator menu from the Spectrum keyboard... and all of the GPIO pins are used up. This needs a little thought but it's fine to play with as it is and you can always attach a USB keyboard to access the menus.

lortoffi, 2022-08-16:
> There is some code on a branch, including an executable: https://github.com/fruit-bat/pico-zxspectrum/tree/feature/zxkeyboard [...]

Why not use a sn74hc164 or similar shift register to drive the keyboard row lines? That should free some pin

fruit-bat, 2022-08-16:
Yup, could do. I will get some and have a play. Will need diodes on each on the row lines as well.

fruit-bat, 2022-08-22:
Shift register has arrived....

[image]

...needs some software now.

fruit-bat, 2022-08-26:
I've pushed some changes and a new uf2 to the main branch. For the example mappings see:

https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/ZxSpectrumKeyMatrix.cpp

It now supports an 8x8 key matrix.

The keyscan routines could do with a PIO program rather than being 'bit-banged', which I will try and get round to at some point.

There is one pin free, which I could use to 2-pin PWM audio?
The UART pins are also free which is nice for debug but could be used for something else.

Let me know your thoughts.