fruit-bat/pico-zxspectrum #39: CVBS output

fruit-bat, 2022-07-27:
I don't know how to do this but it is interesting!

badrianiulian, 2022-07-27:
Maybe have a peek here: [https://github.com/breakintoprogram/pico-mposite](https://github.com/breakintoprogram/pico-mposite), here [https://github.com/pa-art/test_ntsc](https://github.com/pa-art/test_ntsc), or here [https://github.com/obstruse/pico-composite8](https://github.com/obstruse/pico-composite8).
There are many variables to take into account though, like PAL vs NTSC (different resolution and frame rate) and don't forget about Color vs. Black & White.

fruit-bat, 2022-07-28:
> Maybe have a peek here: https://github.com/breakintoprogram/pico-mposite, here https://github.com/pa-art/test_ntsc, or here https://github.com/obstruse/pico-composite8. There are many variables to [...]

Many thanks, I will have a look at them 😀

svofski, 2024-12-08:
Hi!

I've had some success adding a CVBS-like video mode using standard scanline lib. It's not ideal and probably deviates from the standard in more ways than it should, but my little intercom CRT monitor picks it up.

[image]

It can probably be added as a variant of ZxSpectrum4PinAudioVga1111Ps2 (I call it ZxSpectrum4PinAudioVga1111Ps2_CVBSx50Hz) in my experiment.

fruit-bat, 2024-12-08:
Very sweet. Please share your changes and I will try to incorporate them .

svofski, 2024-12-11:
I don't know what is the preferred way to contribute, but here's a PR, hope it comes in handy https://github.com/fruit-bat/pico-zxspectrum/pull/195

fruit-bat, 2024-12-11:
> I don't know what is the preferred way to contribute, but here's a PR, hope it comes in handy #195

Many thanks, I will try and sort out the broken audio and then see what I can do here :-)