fruit-bat/pico-zxspectrum #42: SD card problem

sergbrest, 2022-07-29:
Hello, I made a project on the breadboard.
VGA connected according to the RGBY 1111 scheme. 
Keyboard via USB.
It works!

Emulator does not see memory card
I used data from table Wiring->General->MicroSD 0
But doesn't work.
The SD card is formatted with FAT32.
I created folders
zxspectrum/snapshots
zxspectrum/quicksaves
zxspectrum/tapes

Firmware: ZxSpectrumPicomputerVga1111Zx.uf2
Where are the pins to connect the SD card?
In which file can I see the connection of peripherals?

fruit-bat, 2022-07-29:
It's in here:

https://github.com/fruit-bat/pico-zxspectrum/blob/fb15a62fc2892ce83fe78799c208c01dc746f891/src/picomputer/CMakeLists.txt

(I need to document all of the pinouts... At the start of this project I only expected there to be one!)

sergbrest, 2022-07-30:
I connected the card according to the pinout above.
On firmware ZxSpectrumPicomputerVga1111Zx.uf2  there is no access to the card.
There are no pulses on the SPI bus.

On firmware ZxSpectrumPicomputerVgaZx.uf2 there is an appeal. The signals are coming. But Piko does not see the memory card.

I wanted to connect to the debug port. But the port is also assigned to the vga pin.
I want to ask you to assemble the finished firmware for my case.
Vga1111
SD_Card
USB_keyboard
Audio Pin
AY-3-8912 A,B & C channels.

fruit-bat, 2022-07-30:
Yes, those configurations are to match the various Picomputers. I will do you a target that is better for bread-boarding. Bear with me!

fruit-bat, 2022-07-30:
I will still look at a separate target for you, but I have just tried the VGA 1111 build and it is reading from my SD card; so I don't think it's a bug in the software. 

Could be SD card compatibility... I'm not sure all of them still support SPI mode. Maybe try a different card if you have one lying about. 

Oh, and those number are GPIO numbers not PIN numbers as the name may lead you to believe.

sergbrest, 2022-07-30:
I tried 3 different cards. Does not work.
And with the assembly of VGA 1111 the LED on the PICO board does not blink
[image]
.

Here is my connection to the card.

sergbrest, 2022-07-30:
What am I doing wrong?

The LED does not blink when accessing the SD card.
ZxSpectrumPicomputerVga1111Zx.uf2
https://youtube.com/shorts/9dRyPVo6UZ0 

LED blinks when accessing the SD card.
ZxSpectrumPicomputerVgaZx.uf2
https://www.youtube.com/shorts/AeF-mSUYrqM

fruit-bat, 2022-07-30:
Wiring looks the same as mine... but I don't have a driver chip... are you powering the SD board with 3.3v or 5v (I am assuming you are powering it as the led is on)? Is the SD board driving the IO lines at 3.3v or 5v? Do you have link to the spec of the SD card breakout?

fruit-bat, 2022-07-30:
Most of the new builds do not flash the led on the pico, since changing to the Pimoroni library.

sergbrest, 2022-07-30:
Sd card 
https://www.arduino-tech.com/robotdyn-micro-sd-card-high-speed-module-for-3-3v-5v-logic-for-microsd-mmc-card/
The card is powered by 3.3 volts. Constantly.

memory card for 8 and 16 gigabytes. Maybe I need to find a 2 gigabyte card?
Maybe the card should be formatted for a certain cluster size?

I'll try another card module without a driver.

fruit-bat, 2022-07-30:
Try disconnecting the card-detect line.

sergbrest, 2022-07-30:
disconnected, did not help

fruit-bat, 2022-07-30:
Might be the card. I have just tried a 32gb Samsung card and it does not work. I have a bunch of Transcend 2gb cards formatted as FAT16 and they are fine.

fruit-bat, 2022-07-31:
There is a new target for you here but you will have to move some of your wires about...

https://github.com/fruit-bat/pico-zxspectrum/blob/vga_target/uf2/ZxSpectrum4PinAudioVga1111Ps2.uf2

It's stuck on a branch (vga_target) for now as I am trying to reduce the amount of repeated code in the various main.cpp(s) which I seem to be accumulating.

Serial port is on GP0, GP1. There is some debug that should print out as the board starts up.

SD card is:
  SDCARD_PIN_SPI0_CS=5
  SDCARD_PIN_SPI0_SCK=2
  SDCARD_PIN_SPI0_MOSI=3
  SDCARD_PIN_SPI0_MISO=4

Audio is:
  BZR_PIN=21
  AY8912_A_PIN=20
  AY8912_B_PIN=26
  AY8912_C_PIN=27

PS/2 keyboard GP6, GP7 for data and clock.

sergbrest, 2022-08-02:
Soldered a new circuit.
Unfortunately, the memory card did not work.

Mounting SD card 0... Core 1 running...
 error: A hard error occurred in the low level disk I/O layer (1)
HID device address = 1, instance = 0 is mounted
HID Report Description 
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 03 75 01 05 08 19 01 29 03 91 02 95 05 75 01 91 01 95 06 75 08 26 FF 00 05 07 19 00 29 91 81 00 C0 
HID Interface Protocol = Keyboard
TUP_DCD_ENDPOINT_MAX 16
HID device address = 1, instance = 1 is mounted
HID Report Description 
05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C 09 01 A1 01 85 02 15 00 25 01 09 E9 09 EA 09 E2 09 CD 19 B5 29 B8 75 01 95 08 81 02 0A 8A 01 0A 21 02 0A 2A 02 1A 23 02 2A 27 02 81 02 0A 83 01 0A 96 01 0A 92 01 0A 9E 01 0A 94 01 0A 06 02 09 B2 09 B4 81 02 C0 
HID Interface Protocol = None
HID has 2 reports 
HID report usage_page=1, usage=128, has_report_id=1 dev_addr=1 instance=1
HID report usage_page=12, usage=1, has_report_id=1 dev_addr=1 instance=1
Mounting SD card 0...  error: A hard error occurred in the low level disk I/O layer (1)
Failed to mount SD card
Mounting SD card 0...  error: A hard error occurred in the low level disk I/O layer (1)
Failed to mount SD card
Mounting SD card 0...  error: A hard error occurred in the low level disk I/O layer (1)
Failed to mount SD card

fruit-bat, 2022-08-02:
Wow, that was quick work. Maybe try a different SD card, it looks like the pico is trying to talk to the card.