fruit-bat/pico-zxspectrum #46: Audio input suggestion

badrianiulian, 2022-08-09:
Since I figure that the ultimate goal is to recreate all functionality of the original ZX Spectrum, I am thinking that by using some modified hardware (or some new one even) from this [microphone library](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico), this goal can be achieved.
By adding a 100uF for isolation purposes as the guidelines for the analog module say maybe we can reach the goal.
I may be able (and willing) to test some hardware... I'm no audiophile but I got the tools to make what I usually need so that the analog "Line In" function can be available for the old cassette players out there... maybe even a save to ".tap" directly to the SD Card at the end... for comparison purposes since I still remember how error prone was this process in the old days.
I can try some different ways with the hardware part (will research more this part so we don't fry the pico) but since all this work is your baby, only you know if the analog part of the library can be adjusted to work with all the different layouts and if one input pin is available.

fruit-bat, 2022-08-12:
This may be of interest :
http://www.zxdesign.info/cassette.shtml

Might be fun to be able to load from a real audio source 🙂. 

The software side will be tricky.  There are moments when the z80 is paused to read the keyboard etc. It catches up afterwards but any audio input will need to be kept in sync.... I think it will require the CPU speed moderation to be rewritten and based on the depth of an audio queue.

badrianiulian, 2022-08-13:
From the above, I extracted this:
[image]
As for the loading part, how about adding an Cassette mode - F2 key is free - that sets the CPU to 3,5MHz and monitors only the F2 key... just a thought...

fruit-bat, 2022-08-13:
I'm not 100% sure but I think Vcc in their design is 5v... So we need to adjust R42 & R41 so that the gate bias is correct for 3.3v. Also, the author says his audio signal goes from -2.3 to +2.6 which seems high to me, I would expect more like -1.1 to +1.1v. The threshold voltage on the pico is about 1.8 volts, rather than 3.7 for the 74hc gate.

fruit-bat, 2022-08-13:
Hmm, also this circuit has no gain, which we may want a little of if I am right about the voltage range on the input pin... May need to give this a little thought. If you can take some measurements of actual voltage ranges into a 1k load that would really help. E.g a pc audio jack, a phone, an old tape player.

fruit-bat, 2022-08-13:
...Maybe the higher voltage is to able to drive headphones?

fruit-bat, 2022-08-13:
On the pico...

"Normally the voltage threshold is about 1.8V, but it isn't guaranteed; it can be anywhere between the maximum input low and minimum input high, that is, between about 0.8 and 2.0V."

fruit-bat, 2022-08-13:
Could just try it with 39k for R42. Please correct my junior school maths if I have got it wrong.

Centre voltage...
(2.0v - 0.8v) / 2 + 0.8v = 1.4v

Add in the BE voltage...
1.4v + 0.6v = 2v

So...
2 = 3.3  R42 / (R42 + 27)
=> R42 ~ 40kohm

And pick one that is easy to buy...
39kohms!

badrianiulian, 2022-08-13:
Been on the road all day and will be the same for the next two days... we had a planned family trip. As soon as I'll be home, I can take some measurements with actual components and different audio signals.

fruit-bat, 2022-08-13:
I hope you have a nice time with your family.

badrianiulian, 2022-08-17:
Thank you! I had a nice time with the family ^_^
As for the tests, today I measurements and here they are:
Using volume to max from PC and playing sound from Fuse (game menu - `ADayInTheLife 1985` - really loud sounds)
PC headphone source:
1k resistor = maximum of 290mV AC and 7mV DC
output voltage ~= 1,48V-1,5V DC
TV line out source (PC HDMI audio through TV):
1k resistor = maximum of 650mV AC and 3mV DC
output voltage ~= 1,53-1,55V DC
[image]
As for actual components, for the 27K I used a 22K+4,7K (~26,3K measured), for the 39K I used a 33K+3,9K+3,9K (~39,4K measured) and for the 56K I used a 47K+10K (~54K measured). As for the transistor, I used a 2N2222A since it was already in my components stack and it's an equivalent for BC548.
For the VCC (3,3V) I used a USB-TTL adapter. 
[image]

fruit-bat, 2022-08-17:
Thanks for those measurements. It looks like the voltage swing is not really big enough 🙁. It's a shame the adc doesn't work, it would be ideal for this. Maybe we need an amplifier (rather than an emitter-follower) or a Schmitt trigger? Don't want it to get too complicated but I've seen some 2 transistor designs for a Schmitt trigger.

badrianiulian, 2022-08-18:
At first glance, an amplifier may be too much (on the other hand it's late and I might not have all the marbles in place right now).
I did found some explained details about a Schmitt trigger [here](https://www.wellpcb.com/schmitt-trigger.html) and also [here](http://rubensm.com/transistor-implementation-of-schmitt-trigger/).
Now regarding the details, we could learn from some actual builds, as [this](https://www.reddit.com/r/AskElectronics/comments/q7xad2/3v_transistor_shcmitt_trigger_using_2n2222/) was another find.
What should the actual voltage swing need to be? 
Also as I seem to understand from an [alternative design](http://solarbotics.net/bftgu/tutorials_schmitt.html) there might be also a need to invert or non-invert the signal... (?)

fruit-bat, 2022-08-18:
Well, if the 0.8 to 2.0 volts thing is correct, at least 1.2v.I
I don't think it matters if it is inverting or not. Thanks for the links I will read through them when I get a mo.

fruit-bat, 2022-08-18:
I've done some more reading and found the rp2040 inputs have their own hysteresis...

https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__gpio.html#ga1f52eecc12d4348ef8bd1e17bdb4b107
 
I probably need to find out more about that.

badrianiulian, 2022-08-18:
I found an example with hysteresis experimentation on the [raspberry pi forums](https://forums.raspberrypi.com/viewtopic.php?t=316366).
If it takes too much of the CPU though, the hardware way is the way to go.

fruit-bat, 2022-08-19:
Been reading this...
https://en.m.wikipedia.org/wiki/Line_level
... which suggests 10k (rather than the 1k) load resistance. 
Perhaps that would give a bit more of a voltage swing.

badrianiulian, 2022-08-19:
As I should have expected, when using the 10K, with the PC headphone source (it was the only test I did), the AC voltage is the only difference encountered:
```
10k resistor = maximum of 65mV AC and 7mV DC
output voltage ~= 1,5V DC
```
Also the same AC voltage seems to be present at the output line (same as before witch I failed to mention above)... so this is definitely not the way to go.
I'll try to find components for the Schmitt trigger schematic... something equivalent to what the 74C/HC14 does.

fruit-bat, 2022-08-19:
They are easy to make with an op amp. I've ordered some that will run at 3v to have a play with. We can compare notes then 🙂. I'll post the circuit when I get a mo.

fruit-bat, 2022-08-19:
This site is good...

https://www.random-science-tools.com/electronics/inverting-schmitt-trigger-calculator.htm

fruit-bat, 2022-08-19:
Something like this...

[image]

The pot is there to adjust the threshold, which we could fix if we find a good value. I'm not sure about the lm224n, just need something that will work at 3.3v.

fruit-bat, 2022-08-20:
...but it does turn out the op-amps I have ordered are going to take a long time to get to me!

badrianiulian, 2022-08-20:
I dug and found some schematics from different clones of the ZX:
CIP-01
[image]
HC-2000
[image]
After that I changed the testing procedure:
I set volume to max from PC and had playing sound from PlayTZX inside DosBox (same game - `ADayInTheLife 1985`):
With the first schematic (transistors) the test had the same result as before.
For the op-amp schematic, I used a βM358N instead of a LM224N and also changed the 10k+100kVR to 100k+100kVR(with the actual value set at about 150k).
Results:
- PC headphone source:
```
output voltage in idle mode ~= 1.71V DC and 0V AC
output voltage in play mode ~= 1.06-1.08V DC and 1.15-1.27V AC (1.23V mostly)
```
- TV line out source (PC HDMI audio through TV and TV volume set to at least 30):
```
output voltage in idle mode ~= 1.71V DC and 0V AC
output voltage in play mode ~= 1.01-1.08V DC and 1.17-1.23V AC (1.20V mostly) | I'm not sure but I think I noticed DC voltage going down when TV volume goes up 
```

fruit-bat, 2022-08-26:
Hi, sorry for the slow reply, 'real-life' has been distracting me!

The 1st circuit looks like a non-inverting comparator, without hysteresis. I like the way the bias is arranged; the single (rather than dual) voltage divider means you don't have worry about tolerance on the resistor values.

The 2nd circuit looks like a high gain inverting amp... but I'm not sure what R30, R70, D9 & D10 are up to! Do you know what VMED gets connected to?

badrianiulian, 2022-08-28:
I know what you mean about 'real life'. I've been pinned to the bed for the last two days due to a nasty cold... better now.
I changed the above labels for the schematics since I mislabeled them. I only just noticed by searching through the schematic for VMED witch goes through at least three pages [(pages 2,3 and 4)](https://rhc14.grey-panther.net/doc/technical/schematics/schema_cip_01.pdf).
There are other schematics on that [site](https://rhc14.grey-panther.net/doc/technical/schematics/index.shtml) that could prove to be useful.

fruit-bat, 2022-08-31:
Thanks for finding those... I will get back to this at some point!

fruit-bat, 2022-09-03:
I've added a little test program which reads GP11 at 44Khz and sends 1s and 0s to the USB uart. Not as good as a dual-channel scope but it's something!
```
cp uf2/ear_in_test.uf2 /media/pi/RPI-RP2/
tio /dev/ttyACM0 
```

fruit-bat, 2022-09-03:
See https://github.com/fruit-bat/pico-zxspectrum/pull/54/files

There is even a small chance this may work! It samples GP11 once a microsecond using the PIO and feeds it into the emulator. 
Should copy with 32x32 microsecond delay in instruction emulation. 

I'ts going to take a bit of work to fix all of the targets so only ZxSpectrumBreadboardHdmi4PinAudio is working at the moment.
Let me know if you want a binary to play with.

badrianiulian, 2022-09-04:
Been a bit busy digging the net for anything Spectrum for a while now (I lost at least four hours tonight just to find [this](https://spectrumforeveryone.com/technical/zx-modules-software/) in the end). Also I've been checking some wav dumps of some tapes since I'll surely need them to test the audio input. I will have to put together another Pico and test it with the breadboard since it's easier in the long run (I ordered a while back one piece of Wren's module and a micro sd card module).
For now I'm off to bed (huge headache) but a binary and some instructions to test in a couple of days would be nice. As soon as I can test it (using the components with βM358N), I'll be back with feedback.
By the way... when I tested the op amp circuit, I used 3.3V. Since the 3.3V has to be stable for the SD Card, maybe I should test the circuit beforehand using 5V from VSYS just like I used for the [HXJ8002 amplifier circuit](https://github.com/fruit-bat/pico-zxspectrum/issues/24#issuecomment-1220602733)?

fruit-bat, 2022-09-04:
Hope your headache is better. Stick to 3.3v on the opamp. I think 5v will damage the pico.

fruit-bat, 2022-09-04:
I've just loaded Frogger from a WAV file 🙂
The circuit I used is a mess but it proves that the software sort of works!!!

fruit-bat, 2022-09-05:
Inspired by the HC-2000 diagram you posted above, I think this may be interesting....
[image]
It's sort of a schmitt trigger (+/- 50mv ?) but with the nice single voltage divider.  What do you think?

Smaller capacitor might work too... maybe 1uF

fruit-bat, 2022-09-05:
Test binary available:
https://github.com/fruit-bat/pico-zxspectrum/blob/feature/audio-in/uf2/ZxSpectrumBreadboardHdmi4PinAudio.uf2

badrianiulian, 2022-09-05:
I've been divided in many areas the last couple of days... I'll try to get my tools to work tomorrow (maybe I can squeeze in some build time on this project). The schematic looks good but the ultimate tests will be the wav files. For now tapes are a no go until I can find a working tape player... I do think I recall stumbling on a WalkMan in the basement at some point in the last couple of years.

badrianiulian, 2022-09-07:
Hello!
I finally tested the new setup and here are the results:
[image]
[image]
[image]

I used the last schematic with the βM358N and changed the 50K resistor to 100K (higher value was better in the earlier test).
I tried with 1uF capacitor but had some problems with one of the wav files - it had lots of static noises and gave an R Type error at some point. I used an 4.7uF after but I ended up with the 10uF like in the schematic and the static noises were still there but lowered and the game was loaded.
As I said in the earlier tests, sound volume must be above 50 for optimal results (from PC signal).

[image]

I think I must open a new issue though since I encountered a problem: with some games, the (default) keys were not working. They work fine in Fuse though.
Both games that had this issue were structured like this:
```
Main menu game 1 and 2 = press any key to start witch worked just fine.
In-game 1 = 'q,a,o,p,m' keys were not working
In-game 2 = 'left shift,z,c' and '8,9,0' keys were not working while only 'q' key was working ('q' returns to Main menu)
```

fruit-bat, 2022-09-07:
Fantastic news it actually works 😀

If you think the pc volume needs to be too high for a signal to get to the  Spectrum you could raise the feedback resistor to 150k (~ +/-30 mv). Don't want to make it too sensitive as the output will start flapping about when there is no input signal.

If you can give me enough information to reproduce the keyboard problem I will try and fix it. Separate issues would probably be best.

Many thanks for the testing and feedback; lovely to see some pictures of it in action.

fruit-bat, 2022-09-07:
On the off chance it fixes the keyboard problem I have 'improved' the port decoding and uploaded a new binary.
https://github.com/fruit-bat/pico-zxspectrum/blob/feature/audio-in/uf2/ZxSpectrumBreadboardHdmi4PinAudio.uf2

badrianiulian, 2022-09-08:
I tested the new binary but I had the same result.
I have a feeling that it's not a port decoding issue since in other games those keys work as they should. The setup is through an usb hub but I really do not think that this could be an issue. I don't have an PS2 port attached though... maybe that would help to test the setup.

These are the games that have those issues: [Distrugator 1](https://zxart.ee/eng/software/game/arcade/shoot-em-up/distrugator/) and [Terroindronzon](https://zxart.ee/eng/software/game/arcade/shoot-em-up/terroindronzon/).
There is a small save button on each page witch downloads a zipped tzx file.
In order to load them as tap files, I converted them using [Win_TZX2TAPv0.21.zip](https://worldofspectrum.net/pub/sinclair/tools/pc/Win_TZX2TAPv0.21.zip).
I used the original WAV dumps to load them with the audio input but the tap loading has the same result (as it should).

Maybe debugging these could help in finding the bug.

badrianiulian, 2022-09-09:
So I decided to do some more digging today.
I removed the USB hub and changed at least three different keyboards.
Using some quick saves that had the keys working some builds ago, I tested with an USB-OTG adapter directly connected and the in-game keys do not work anymore.
So it may not be about an actual game that it's not working and you may be right about the port decoding issue.

fruit-bat, 2022-09-09:
Thanks for that, and to add to the mystery, game 1 works if I force bit 6 low on a read of port 254 (i.e. 0xbf for no keys pressed).

fruit-bat, 2022-09-10:
Hmm, I think there may be more than one issue in all of this. 

There is a new uf2 with some code that should sort out the two games with stuck keys... but only as long as the ear input is 'quiet'. Hopefully the hysteresis on the op-amp circuit should mean it is(?).

The tap loader should override the ear input so I'm not sure what is causing that problem. I need to investigate that a little more.

fruit-bat, 2022-09-10:
My op amps have finally turned up :-)
So now I have to admit which values I guessed incorrectly!

Using my phone to supply the audio I needed a much larger feedback resistor:
[image]

The 1uf capacitor worked fine for me (but I know you were having problems).

With standard spectrum tapes this seems to work very reliably... with turbo loaders no so. May need to look in more detail at memory cycle timing... I really never expected the emulator to be doing timing critical stuff!

badrianiulian, 2022-09-10:
I was just about to reply after intensive tests using the last build (I was in front of the screen wen you posted the build and got notified instantly) and then I saw your last post about the resistor.
The keyboard problem is gone now with those games.
I replaced the 100K with 1M but haven't gotten at the point of testing the audio loading from the wavs since I was already testing some other issue by hitting my head against the wall and expecting different results.
I restarted testing... for now just like I did in the last couple of hours by loading a tap from the SD Card. Before and now, by loading the same game over and over again I got weird results that I think have to do with me messing with the CPU speed.
So here's what I noticed:
1. F12/128k mode resets automatically to 4.0MHz (it should be 3.54690 MHz, but that's another issue)
2. F11/48k mode resets to 3.5MHz.
3. F12/128K mode resets to 4MHz and while in this mode, from the menu you can select 48k mode but the CPU frequency setting remains at 4MHz.

Hitting that tap file over and over again,  here is what i found:
- F12/128k mode (actual 128k) with 4MHz, tap loads fine.
- F12/128k mode (48k mode selected from menu) with 4MHz, tap loads fine.
- F12/128k mode (48k mode selected from menu) with me tampering with the CPU to 3.5MHz, tap loads fine.
- F11/48k mode with 3.5MHz, tap loads fine.
- F12/128k mode (actual 128k) with me tampering with the CPU to 3.5MHz, tap last block outputs an error and the screen gets stuck there.
- F11/48k mode with me tampering with the CPU to 4MHz, tap last block outputs the same error with the screen getting stuck.

At some point (before I saw your last post about the resistor), I had a quick save in the 11th slot that when pressing Alt/F11 showed up only to reset to Basic... it may be related to the above tamperings, to a bad save to the SD Card, to a connection to the F11/48k reset (I doubt it) or to cosmic rays :). It hasn't happened since I've overwritten the quick save with another state.

Also, about what I said earlier about a menu setting about GPIO11 On/Off switch, I found an even more elegant solution:
- when a tape is inserted from SD Card, the tape slot is occupied with that tap file: this should set GPIO11 audio in tape loader to Off mode (stops listening from GPIO11)
- when a tape is ejected and the tape slot is free: this should set GPIO11 audio in tape loader to On mode (starts listening from GPIO11)

Another thing:
By pressing F1, the system menu shows up but the fields are outdated if a quick save was just loaded. Only by pressing up/down keys and moving through the options, the quick save states are refreshed in the menu options.

I'll get back about the wav loading using different setups (TV/PC/Phone) tomorrow... it's 02.30 and I burned the midnight oil enough for now. (I do enjoy these tests though)

fruit-bat, 2022-09-11:
Many thanks for all the detail above. I will go through it in detail shortly. The guidance as to what is an isn't working is very helpful :-)

The default CPU speed settings make much less sense now as loading from tape needs 3.5Mhz (loading .tap from the SD cards *should* be unaffected as it counts Z80 T states). The reason the 128k defaults to 4Mhz is the VGA 60Hz refresh rate... it probably makes more sense now to default everything to 3.5Mhz and offer an option to change it. The original idea was the 4Mhz CPU would compensate for the faster screen refresh rate.

badrianiulian, 2022-09-11:
Got a chance to test things today:
Working setup is with the 10uF capacitor and 1MΩ for the feedback resistor.
I did found in the schematic of the HC-85 a 2MΩ feedback resistor:
[image]

As for the actual signal when using the wav files I have, the audio cable needs to be high quality and the connections have to be very tight on each component of the schematic or else the signal will be disturbed by static at some point.
I think that the 10uF capacitor helps with the wav files since they seem to be dumped from the tape directly and I recall that a direct copy of a tape had issues back in the days (degradation built-in design of tapes for copy prevention) and that's why copying software was used to transfer programs in memory, only to record them on another tape (clean signal).
I checked only with the PC signal since I need a mic/headphones splitter for the phone (I do recall I made at least two of those at some point and one is surely at work). As soon as I'll find one, I'll test again using the phone.

Loading from a tzx file (with playtzx) always works when tested. I even tested a [nasty piece of tzx](https://worldofspectrum.net/item/0009446/) (with built in pulse signals witch prevents converting the tzx to tap) that worked when loaded with playtzx.

All the tests were done in 48k mode (with 3.5MHz).

fruit-bat, 2022-09-11:
This seems to be much more stable, although I don't really understand why (sorry)...

[image]

fruit-bat, 2022-09-11:
New binary available which always defaults to 3.5Mhz and refreshes the menu when F1 is pressed.

badrianiulian, 2022-09-12:
Started testing that tap file with the new binary and all loading works as it should now:
- 48k reset with no modification to CPU speed (=3.5MHz) => loading works all the way
- 128k reset with no modification to CPU speed (=3.5MHz) => loading works all the way
- 128k reset with no modification to CPU speed (=3.5MHz) and in-menu selection of 48k => loading works all the way
- 48k reset with modification to CPU speed (set manually to 4.0MHz) => loading works all the way
- 128k reset with modification to CPU speed (set manually to 4.0MHz) => loading works all the way
- 128k reset with modification to CPU speed (set manually to 4.0MHz) and in-menu selection of 48k => loading works all the way

Also tested quick save slots: they always load using 3.5MHz regardless of what their save state witch is OK. If anyone needs faster speed they can change it from the F1 menu.

Still haven't found the handsfree audio splitter to test loading tapes using phone signal... Searched at work for now but I'll have to dig deeper into my stash at home.
Meanwhile tested with a laptop using the wav files and the most stubborn wav (degraded) loaded only by setting volume to 75% from system sound settings.
I left the the 10uF capacitor and 1MΩ for the feedback resistor in place since they have their [role](https://www.electronics-tutorials.ws/opamp/opamp_2.html) in the circuit. 
The ultimate test will definitely be loading using the phone. I did found some PlayTZX equivalent app for android and noticed that it's setup as default to turn volume up to 75% as an optimal setting.

badrianiulian, 2022-09-12:
Success:
The splitter I have is actually for the headset side so I can insert mic and headphones to a PC/Laptop(older models) so I ended up buying one splitter (had the parts to make one but do not have the actual time to build it - I do my work usually during the night when the kids are asleep).

I tested playing some tzx file with default app settings (`PlayZX` app on android phone) witch worked perfectly. The app does output some noise right when finishing a segment and entering a pause but that hasn't influenced tape loading.

Then I went on to test the wav files. At first I got an R-Type error and rechecked my phone settings... volume was too low.
I changed the volume level to 12 out of 15 (80%) and after that tested with three different wav files.
From then on everything went fine and no errors were encountered.

fruit-bat, 2022-09-12:
Many thanks for the update, things look like they are heading in the right direction. 

I have spent some time this evening messing about with the input circuit... trying to get a nice balance between enough gain and not too much noise. I have a version that is better but need to have a bit more of a play with it before I bother you with more circuits! The new version has two stages an amp and then a Schmitt-trigger. I'll post a diagram if gets good enough!

Thanks again for the testing and feedback.

badrianiulian, 2022-09-13:
Well, as I said earlier, I enjoy these tests so it's not a bother.
I listened to the normal speaker output of `PlayZX` android app and there is a sort of zoom-out-vroom when entering a pause, the same noise that I heard through the speaker of the pico.
Then I listened to the normal speaker output of `ZX Tape Player` android app and there seems to be a clear sound when entering a pause.
I tried another test loading with `ZX Tape Player` and this time that zoom-out-vroom isn't there anymore. The difference now was that the phone sound level had to be 100% for games to load.
In my opinion, this schematic works but if things can get better, then why not. :)
I remember back in the days that I had to play with sound levels for some games to load and it looks like it's the same thing now.

badrianiulian, 2022-09-13:
Maybe this helps:
[image]
It's not exactly in the range (0.8v...1.8v) but it's a start?

fruit-bat, 2022-09-13:
Wow, your circuit looks cool! Have you tried it out for real yet?

This is my latest attempt, which is working quite well. You can get away with about half volume on a phone and turbo loaders seem fine. It's also nice and 'quiet' even with a dangling lead. Trying to keep amplifiers quiet on a breadboard next to a digital circuit is tricky!

[image]

Let's see if you run out of 10k resistors!

badrianiulian, 2022-09-13:
>  Let's see if you run out of 10k resistors!

:)) Last time I went to the electronics store I got about 20 pieces... I knew this would come in handy soon.
Haven't tried the circuit... but the βM358N already has two op-amps in it and I don't think I really got everything right when I tested the circuit... I think the op-amp settings in the simulator are not quite correctly entered. I tested your diagram and things are not as expected... If I wanna learn more, I'll have to check both of the circuits.

fruit-bat, 2022-09-13:
haha - yes analogue electronics is black-magic! You can do lots of maths... then simulate it... then completely redesign it once you try if for real :-)

fruit-bat, 2022-09-14:
I've been looking at the 2 transistor Schmitt trigger your have used... I may be wrong here (and do tell me if I am) but...
Although we may get nice hysteresis from it we may struggle to get the voltage swing needed because of the resistor ladders.
I *don't think* there is the same problem with the op-amp version(?).

fruit-bat, 2022-09-14:
It would be interesting to compare both of our pre-amps for gain and how much noise they pick up.

badrianiulian, 2022-09-14:
I tried loading them schematics into pspice and maybe I'm doing it wrong but with your schematic, the output level is about 1.6v (can't seem to figure out how to make it swing the voltage yet). But the readings from the transistor schematic are about 0.878v.
What's the voltage swing again? I need a high and a low so I can play with them resistors... maybe I can come to a working voltage that fits the needs.

fruit-bat, 2022-09-15:
Tonights attempt at a single stage amplifier. Really hard to stop it picking up all sort of noise but it works down to about 2/3 volume on my phone.
[image]
I had to twist a 0v line around the signal line to the pico(!)

badrianiulian, 2022-09-15:
I've given up on the transistor schematics... they don't work since they're not calibrated good enough... they're supposed to be around the same voltage the op-amp gives out... about 1.75v. Getting at that value requires a lot more knowledge than I have. I removed them apples from the post above since they don't do any good.

About your first schematic:

> [image]

I tested it (physically) and could not load any wav or files. I tried with all levels of volume from 10 to 15 and couldn't even get the game name to show.
Instead I used a mix of schematics witch worked with wav files and tapes played with `ZX Tape Player` on volume levels from 11 to 15:
[image]

Grounding the transport wire may be the best idea yet... just don't connect it on both ends because it creates a nasty ground loop witch generates a lot of noise.
In my opinion, two op-amp setup seems to work better than the single one and that is easily seen when loading files with `ZX Tape Player`: with the first setup (one op-amp) volume level had to be at 100% and with this setup it works even at 73% (level 11 out of 15)
With this setup, the noises heard are only the ones the phone seems to generate when entering those pauses.

Besides... I think I can squeeze some smd dual op-amps like LM2904/LM258/LM358 with smd rc components on a small pcb... just like a small module that can be closely connected to GPIO11.

BTW, the above schematic was made with [this](http://falstad.com/circuit/offline/). It's not like the other one I posted before but it supports inserting a wav file... did not find a way to see voltage variations yet... I'll have to play more with it.

fruit-bat, 2022-09-16:
The simulator looks interesting, I will have a play with it. Thanks for the link. 

Yes, I agree, the two stage version is the way to go. I think I learnt something from the 1 stage version though... Shielding the signal to the pico matters, and the 100k resistor on the input makes the circuit a lot quieter with a dangling audio lead.

When I get time I will have a go with the version you posted above.

[49 more replies]