fruit-bat/pico-zxspectrum #50: Retroflag classic usb gamepad not working

sdbrother0, 2022-08-24:
"Retroflag classic usb gamepad" Not detected in F1 screen.

sdbrother0, 2022-08-24:
[image]

fruit-bat, 2022-08-24:
To stand a chance of fixing this I need the game-pad's HID report descriptor.
On Linux you can do something like:
```
sudo usbhid-dump -e descriptor
001:123:000:DESCRIPTOR         1661369192.405711
 05 01 09 04 A1 01 A1 02 75 08 95 05 15 00 26 FF
 00 35 00 46 FF 00 09 32 09 35 09 30 09 31 09 00
 81 02 75 04 95 01 25 07 46 3B 01 65 14 09 39 81
 42 65 00 75 01 95 0C 25 01 45 01 05 09 19 01 29
 0C 81 02 06 00 FF 75 01 95 08 25 01 45 01 09 01
 81 02 C0 A1 02 75 08 95 04 46 FF 00 26 FF 00 09
 02 91 02 C0 C0
```
[That command is in the usbutils package]

You may end up with a bunch of them so run it first without the game-pad connected then again with it plugged in so you can spot the new one.

Also, you can copy-paste the the lines _after_ the one containing 'DESCRIPTOR' into...

https://eleccelerator.com/usbdescreqparser/

... and click 'Parse as USB HID report descriptor' to check you have the correct descriptor.

sdbrother0, 2022-08-25:
Hallo! 
Thanks for the answer!

Strange, but the gamepad is defined as "Microsoft X-Box 360 pad" (by dmesg):

sd@sd-ThinkPad-T480:~$ sudo dmesg
[  863.032608] usb 1-2: new full-speed USB device number 11 using xhci_hcd
[  863.189537] usb 1-2: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.14
[  863.189551] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  863.189557] usb 1-2: Product: Controller
[  863.189562] usb 1-2: Manufacturer: Controller
[  863.189567] usb 1-2: SerialNumber: Controller
[  863.193075] input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/input/input23

Unfortunately usbhid-dump reports "No matching HID interfaces"

sd@sd-ThinkPad-T480:~$ sudo usbhid-dump -e descriptor
No matching HID interfaces

Does this mean it won't work?
Thanks!

sdbrother0, 2022-08-25:
But I have another device of the same look (no name).
It is defined,
file:///home/sd/Pictures/Screenshots/Screenshot%20from%202022-08-25%2009-15-13.png
but the buttons on the latest version are stuck

usbhid-dump report:

001:014:000:DEC         1661407656.424612
 05 01 09 04 A1 01 A1 02 75 08 95 02 15 00 26 FF
 00 35 00 46 FF 00 09 30 09 31 81 02 95 03 81 01
 75 01 95 04 15 00 25 01 35 00 45 01 81 01 65 00
 75 01 95 0A 25 01 45 01 05 09 19 01 29 0A 81 02
 06 00 FF 75 01 95 0A 25 01 45 01 09 01 81 02 C0
 A1 02 75 08 95 07 46 FF 00 26 FF 00 09 02 91 02
 C0 C0

 0x01, 0x14,        // Unknown (bTag: 0x00, bType: 0x00)
0x00,              // Unknown (bTag: 0x00, bType: 0x00)
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x04,        // Usage (Joystick)
0xA1, 0x01,        // Collection (Application)
0xA1, 0x02,        //   Collection (Logical)
0x75, 0x08,        //     Report Size (8)
0x95, 0x02,        //     Report Count (2)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x35, 0x00,        //     Physical Minimum (0)
0x46, 0xFF, 0x00,  //     Physical Maximum (255)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x03,        //     Report Count (3)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x01,        //     Report Size (1)
0x95, 0x04,        //     Report Count (4)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x35, 0x00,        //     Physical Minimum (0)
0x45, 0x01,        //     Physical Maximum (1)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x65, 0x00,        //     Unit (None)
0x75, 0x01,        //     Report Size (1)
0x95, 0x0A,        //     Report Count (10)
0x25, 0x01,        //     Logical Maximum (1)
0x45, 0x01,        //     Physical Maximum (1)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x0A,        //     Usage Maximum (0x0A)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x06, 0x00, 0xFF,  //     Usage Page (Vendor Defined 0xFF00)
0x75, 0x01,        //     Report Size (1)
0x95, 0x0A,        //     Report Count (10)
0x25, 0x01,        //     Logical Maximum (1)
0x45, 0x01,        //     Physical Maximum (1)
0x09, 0x01,        //     Usage (0x01)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xA1, 0x02,        //   Collection (Logical)
0x75, 0x08,        //     Report Size (8)
0x95, 0x07,        //     Report Count (7)
0x46, 0xFF, 0x00,  //     Physical Maximum (255)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x09, 0x02,        //     Usage (0x02)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xC0,              // End Collection

// 101 bytes

sdbrother0, 2022-08-25:
[image]

fruit-bat, 2022-08-27:
> Hallo! Thanks for the answer!
> 
> Strange, but the gamepad is defined as "Microsoft X-Box 360 pad" (by dmesg):
> 
> sd@sd-ThinkPad-T480:~$ sudo dmesg [ 863.032608] usb 1-2: new full-speed USB device number 11 using xhci_hcd [ 863.189537] usb 1-2: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.14 [ 863.189551] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3 [ 863.189557] usb 1-2: Product: Controller [ 863.189562] usb 1-2: Manufacturer: Controller [ 863.189567] usb 1-2: SerialNumber: Controller [ 863.193075] input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/input/input23
> 
> Unfortunately usbhid-dump reports "No matching HID interfaces"
> 
> sd@sd-ThinkPad-T480:~$ sudo usbhid-dump -e descriptor No matching HID interfaces
> 
> This is device: https://www.retroflag.com/Classic-USB-Controller-J.html Does this mean it won't work? Thanks!

For the time being the code needs a HID report to work with a joystick. I'm wondering about supporting some joysticks, that do not supply a description, by including pre-built HID reports for them... but it's just wondering for now!

fruit-bat, 2022-08-27:
> But I have another device of the same look (no name). It is defined, [image]
> 
> but the buttons on the latest version are stuck
> 
> usbhid-dump report:
> 
> 001:014:000:DEC 1661407656.424612 05 01 09 04 A1 01 A1 02 75 08 95 02 15 00 26 FF 00 35 00 46 FF 00 09 30 09 31 81 02 95 03 81 01 75 01 95 04 15 00 25 01 35 00 45 01 81 01 65 00 75 01 95 0A 25 01 45 01 05 09 19 01 29 0A 81 02 06 00 FF 75 01 95 0A 25 01 45 01 09 01 81 02 C0 A1 02 75 08 95 07 46 FF 00 26 FF 00 09 02 91 02 C0 C0
> 
> 0x01, 0x14, // Unknown (bTag: 0x00, bType: 0x00) 0x00, // Unknown (bTag: 0x00, bType: 0x00) 0x05, 0x01, // Usage Page (Generic Desktop Ctrls) 0x09, 0x04, // Usage (Joystick) 0xA1, 0x01, // Collection (Application) 0xA1, 0x02, // Collection (Logical) 0x75, 0x08, // Report Size (8) 0x95, 0x02, // Report Count (2) 0x15, 0x00, // Logical Minimum (0) 0x26, 0xFF, 0x00, // Logical Maximum (255) 0x35, 0x00, // Physical Minimum (0) 0x46, 0xFF, 0x00, // Physical Maximum (255) 0x09, 0x30, // Usage (X) 0x09, 0x31, // Usage (Y) 0x81, 0x02, // Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position) 0x95, 0x03, // Report Count (3) 0x81, 0x01, // Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position) 0x75, 0x01, // Report Size (1) 0x95, 0x04, // Report Count (4) 0x15, 0x00, // Logical Minimum (0) 0x25, 0x01, // Logical Maximum (1) 0x35, 0x00, // Physical Minimum (0) 0x45, 0x01, // Physical Maximum (1) 0x81, 0x01, // Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position) 0x65, 0x00, // Unit (None) 0x75, 0x01, // Report Size (1) 0x95, 0x0A, // Report Count (10) 0x25, 0x01, // Logical Maximum (1) 0x45, 0x01, // Physical Maximum (1) 0x05, 0x09, // Usage Page (Button) 0x19, 0x01, // Usage Minimum (0x01) 0x29, 0x0A, // Usage Maximum (0x0A) 0x81, 0x02, // Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position) 0x06, 0x00, 0xFF, // Usage Page (Vendor Defined 0xFF00) 0x75, 0x01, // Report Size (1) 0x95, 0x0A, // Report Count (10) 0x25, 0x01, // Logical Maximum (1) 0x45, 0x01, // Physical Maximum (1) 0x09, 0x01, // Usage (0x01) 0x81, 0x02, // Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position) 0xC0, // End Collection 0xA1, 0x02, // Collection (Logical) 0x75, 0x08, // Report Size (8) 0x95, 0x07, // Report Count (7) 0x46, 0xFF, 0x00, // Physical Maximum (255) 0x26, 0xFF, 0x00, // Logical Maximum (255) 0x09, 0x02, // Usage (0x02) 0x91, 0x02, // Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile) 0xC0, // End Collection 0xC0, // End Collection
> 
> // 101 bytes

Which buttons are stuck (joystick, keyboard)?
Which 'uf2' file are you using?

fruit-bat, 2022-08-27:
I will have a look at that HID report and see if there is anything I can fix to make it work :-)

fruit-bat, 2022-08-27:
I've fixed an issue relating to the joystick for which you supplied a HID report...  there are some new uf2 files available. 

Let me know if this helps.

sdbrother0, 2022-08-28:
This is great! With the latest version of uf2, the joystick works completely! Thanks a lot!. 
Hope the device like https://www.retroflag.com/Classic-USB-Controller-J.html will work in the future.
Can be closed :-)

fruit-bat, 2022-08-28:
Glad that helped and thanks for letting me know. I'll raise Xbox compatible controller support as a separate issue.

https://github.com/fruit-bat/pico-zxspectrum/issues/52

closed 2022-08-28 by fruit-bat