fruit-bat/pico-zxspectrum #53: SD Card reading stuck on lots of files

alex-taras, 2022-08-30:
Hi, noticed the SD Card loading gets stuck if I have a lot of files on the SD Card (280). Is it possible to add support for subfolders so we can have big collections in on a single SD Card (tried with subfolders and it doesn't display anything)? around 50 files works fine (talking about the tapes folder)

PS. Thanks for this amazing piece of work !

fruit-bat, 2022-08-31:
Sounds like a nice idea. I will need to re-think the interaction between the folders and menus.
At the moment the folders are read once after reset/power on .

Ripper912, 2022-10-24:
Hi any update on this? The Sub-Folder would it be a great ideal for having a lager collection/s. At least in the Quickload section or Tápe Sub-Folder.
Thanks for your great programing skills as well support via this awsome pice of software.

Regards

fruit-bat, 2022-10-26:
Hi, no progress yet. Currently adding some support for tzx format,  which turns out to be quite involved. Thanks for the nudge, I will have a look at this after tzx. No promises though, I really don't have a lot of ram to work with 🙂

fruit-bat, 2022-12-14:
I've merged some changes that should allow more files in the tapes and snapshots folders.
The changes also deliver directory information to the menus to allow these to be presented and acted on in a later set of changes. 

(1) and (2) from the above comment are now complete. When I get a mo, I will look at adding directories to the menus.

fruit-bat, 2022-12-24:
There are binaries on the feature/fexp branch. Not well tested but out for Christmas!
These give basic folder navigation and move the quicksaves into a sub-folder of the snapshots.
Let me know how you get on with it :-)

alex-taras, 2022-12-25:
Oh, thanks, just in time for Christmas ! Will give it a try and let you know, Happy Holidays !

closed 2022-12-25 by fruit-bat

alex-taras, 2022-12-26:
tested this a bit, everything seems fine ! great improvment !