fruit-bat/pico-zxspectrum #55: Can not read .tap files from SD card.

lfmm-brz, 2022-09-06:
I've hooked up the ZxSpectrum4PinAudioVga1111Ps2 on a perforated breadboard, it is working very well, except the fact that I can not read any .tap filefrom the SD card. I can do load .z80 files and save/load the snapshots. The Keyboard F5/F6/F7 keys are good, the .tap files are good (tested on an emulator). 
Steps I am doing:
1 - Press F1,
2 - Choose .tap file,
3 - Motor on,
4 - Esc to go back to ZX screen,
5 - Press F5.
I've tried other ways with no success.

I created the folders as per instructions:
zxspectrum/snapshots	Put your .z80 snapshot files in here.
zxspectrum/quicksaves	Folder for quick saves.
zxspectrum/tapes	    Folder for .tap tape files.

Using F5/F6/F7 as per instructions:
F5	play current tape
F6	play previous tape
F7	play next tape

Any clue?

Thanks a lot.

fruit-bat, 2022-09-06:
I've just tried my version of the VGA board and it seems to be working... but some things to note:
1 - Performing a reset of the Spectrum on the F1 menu stops the tape.
2 - If you start a tape from the menu you do not need to press F5
3 - There is a few seconds of delay before the tape header is played

The F5, F6, F7 are a little bit separate from the menu, and in fact were written before the menus existed. The just loop through tape files found in the tapes folder. Still, I am surprised they are not doing anything. I will have a glance at the code and see if it is doing something silly.

Try:
1 - Reset the spectrum
2 - Select a tape in the F1 menu
3 - Exit the menu
4 - Tell the spectrum to load from tape

lfmm-brz, 2022-09-06:
I've tried,

1 - Reset the spectrum: Main menu / Option 6 / Reset 48k ZX Spectrum / Esc
2 - Select a tape in the F1 menu: Main menu / Option 2 / Choose a tape / Esc
3 - Exit the menu: Esc
4 - Tell the spectrum to load from tape: LOAD ""

The screen remains blank.

If I reset to the 128K ZX Spectrum and issue steps 2, 3 and 4, the screen remains with LOAD "" on the left top corner.

Thanks a lot for your support.

fruit-bat, 2022-09-06:
There was a bug like this but it was a while back now. Are you using the latest firmware (there should be a uf2 from about 6 days ago)?

lfmm-brz, 2022-09-06:
No,
It is from August 20.

lfmm-brz, 2022-09-06:
I will download the new one and revert to you.

lfmm-brz, 2022-09-06:
It is working.
Thank you very much "fruit-bat"!!!

closed 2022-09-06 by lfmm-brz