fruit-bat/pico-zxspectrum #56: Nice to have: Pimoroni VGA Demo Base support

alex-taras, 2022-09-11:
Any chance to support https://shop.pimoroni.com/products/pimoroni-pico-vga-demo-base?variant=32369520672851 since there's already VGA support for some other implementations? They have schematics and a lot of demos for it. Cheers and keep up the good work !

fruit-bat, 2022-09-12:
Sure, I might have a look at some point. The code I was at the moment is very much 8bit (or less) for the RGB so it's a reasonable amount to change, but it would be nice!

fruit-bat, 2022-09-23:
Should be there now. Might not be perfect. Let me know how it works for you :-)

alex-taras, 2022-09-24:
Hey there, everything works on my VGA board. The only minor issue is some colors ar a bit off. The menu is green and stuff is a bit yellowish. I've added some photos of my screen, it's a bit more yellowish in reality, teh phone corrected it a bit for some reason (Dizzy is almost yellow instead of white).
[image]
[image]
[image]
[image]

fruit-bat, 2022-09-24:
I just picked a random colour for the menu, I thought the green was a bit more relaxed! Can make it white if you prefer.
I'm not seeing the 'yellowing' on my version. I do think the difference between bright/normal is a bit much but the colours don't seem off. Could be a problem with your lead, or maybe there are different variants of the VGA board(?). Can you send a picture of the back of your board where it lists the connections.

alex-taras, 2022-09-24:
Oh that explains the menu change. As for the rest it can absolutely be a problem on my end (monitor, lead, cable...). Attached an image of my board for reference but please don;t spent any more time on this if it works on your end. It's already more than i could have hoped for and the colors don't bother me in my setup. All i need from you now is a paypal so i can buy you a few beers :) THANKS !
[image]

closed 2022-09-24 by alex-taras

fruit-bat, 2022-09-24:
There is a new release in which I have fixed some problems with the audio output and brightened up standard (non-bright) colours. Pinout looks the same as mine, let me know if you can't figure out what is wrong and I will investigate a little further, quite possible I have made a mistake.

Thanks for the paypal offer, much appreciated but this is just for fun! Glad you are enjoying it!

alex-taras, 2022-09-25:
Tested it a bit, still on the yellowish side. But again it may be my monitor's viewing angle, possibly something wrong with it. If i look at it from a really high position everything seems acurate. It's weird that i didnt see this in my regular day to day use but then again i just use it as a side monitor, in vertical mode. Please don't waste time on this one any more unless anyone else reports the some at some point. There are other nice things worth implementing in that time :). Thanks again !