fruit-bat/pico-zxspectrum #60: Kempston joystick issiue on PicoZX VGA Board with Pico W

Ripper912, 2022-10-10:
Hi,

I build myself a PicoZX with VGA  boards from Bobricius:
https://www.tindie.com/products/bobricius/picozx-zx-spectrum-128k-vga-emulator-on-rpi-pico/
with Fruit-bat onboard. However I used Pico W instead a regullar Pico and have a non-working 9pin- Kempston joystick on may games. Strange behaviour as around 20% .z80 are working flawlessly, 20% .z80s are freezing and the rest around 60% of .z80 are compleatly non responsive at all ;-(

PS:Joysticks are Competion Pro & w Quickshots(xtested on ZX Spectrum +2A and Clones like Didaktik Gama/M tested & fully working)

Thanks for Amy help or support

Ripper912, 2022-10-10:
apparently a long/er 9 pin cable could cause exact such issues. will shorten the cables and test if works

fruit-bat, 2022-10-11:
Sorry to hear that you are having lots of trouble with z80 compatibility. The emulation only supports the 48k and 128k models - and then only an approximation. The z80 format can contain information for other models, which will just not work. Though I am surprised at the failure rate... most work fine for me, rather than the other way round.

I will check my latest build on the bobricius board in case I have broken it!

fruit-bat, 2022-10-13:
I've loaded up the latest build and it seems to be working ok for me. 

When you say 'freezing' can you still access the menus or has it locked up completely. I ask as maybe the PicoW is less happy running at 200Mhz than the Pico.

Ripper912, 2022-10-17:
Hi thanks for testing. I had time to play with the board once again and have to say I propabaly found the error. I have powered the board from the MicroUSB of the RB Pico however on Bobricius VGA board there are actualy 2xaddiational Vias [next to audio Jack](+5V&Ground).Once powered via the Vias All USB devicess (incl.keyboards& joystick 🕹️)are now fully working. Also the compatiblity with the Atari 9pin connctors is now Higher(but still not 100%- Menu works Also by freeze)-propably shorter cable 20-30cm max should fix this as well according the creator - Peter.thanks for the support & help -keep going 👍