fruit-bat/pico-zxspectrum #61: Pico W ?

Ripper912, 2022-10-10:
Hi ,

Is/would it possible to use PicoW instead od standard one? The Wi-fi options could be quite handy ;-D
Dont you think?

Thanks & regards for any feedback

fruit-bat, 2022-10-11:
Hello,

I don't have a PicoW at the moment so I cannot say for sure... but having had a look at the pinout it *should* work just fine.
Sorry I know this isn't a very helpful answer but I'm afraid you are trailblazing the PicoW adoption at the moment.

andrewlavens, 2022-10-27:
Obviously YMMV but I've given the Pico W a try with the Pimoroni DV Demo Base. I was able to run the pre-made uf2 and sound, video and USB keyboard were fine. I can't try other devices as I don't have them.

(And this project is fantastic, by the way!)

siffland, 2023-04-03:
If you happen to land a Pi Pico W, the SDK kit looks like you can connect bluetooth devices, hence a bluetooth keyboard.  How cool would that be a ZX with a bluetooth keyboard or controller, or both.

ArnoldUK, 2023-04-03:
It is possible to use bluetooth controllers with the standard Pico through the USB connector using a OTG cable.
Simply plug the bluetooth dongle into the Pico's micro usb socket and supply power through VSYS. Works for me with my mini bluetooth keyboard and mouse.

fruit-bat, 2023-04-05:
I do have a Pico W now but it's still in its package. When I get chance I will see how involved using a bluetooth keyboard is. Also, believe it or not I am running out of cpu cycles for all this stuff...  a lot of which is take up by trying to make the audio acceptable.

ArnoldUK, 2023-04-05:
The audio is good enough for me and I wouldn't expect perfect audio signals to sound good through a tiny 8 ohm speaker. I think the file and menu system could be improved and I'm working on adding a graphical GUI at the moment but my time is limited.