fruit-bat/pico-zxspectrum #62: Keyboard

jaromaz, 2022-10-21:
I'm using a Pimoroni Pico VGA and everything works fine, except the USB keyboard. Only the basic keys work correctly on it, but it is impossible to get special characters through which it is impossible to program in Basic. :( Is there any way to get the correct key mapping of a standard USB keyboard? (Great project btw :) ).

fruit-bat, 2022-10-22:
Give me an example of a specific key & symbol you are trying to get to. 
This will give me something to test works for me... and we can go from there.

jaromaz, 2022-10-24:
My mistake - the project simply emulates the ZX Spectrum keyboard layout and it works fine, but a configuration option to use the standard keyboard layout would be useful too. Can you add such an option? Then it would be comfortable to program in Basic on a regular keyboard.

fruit-bat, 2022-10-26:
The emulator maps a keypress on the usb keyboard to one or more simultaneous keypresses on the emulator. 

In addition, there is support in one of the builds for a key matrix that matches the Spectrum 's original layout. 

If you have keys that are not doing what you expect please suggest a mapping for them that follows the rule above (or suggest a new rule 🙂) . Keyboards are not all the same and I appreciate that people may get different behaviour.

jaromaz, 2022-10-27:
I used the .uf2 file available on the project GH homepage for the Pimoroni Pico VGA. It is enough to implement additional support for the **Generic 105-key US** keyboard layout. Even in Europe we mainly use this layout :) Ideally, somewhere in the emulator there would be an option to choose between these two keyboard layouts (US Keyboard / ZX Spectrum).