fruit-bat/pico-zxspectrum #68: build failure

pi-gram, 2023-02-05:
Setup environment and can build, compile and make other applications - so the environment is okay.

Followed the readme instructions as defined; 

`cmake -DPICO_COPY_TO_RAM=0 -DPICO_BOARD=vgaboard ..
`
```
Using PICO_SDK_PATH from environment ('/opt/pico-sdk')
PICO_SDK_PATH is /opt/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
Using PICO_EXTRAS_PATH from environment ('/home/tony/dev/pico/c-code/pico-extras')
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is Release
PICO target board is vgaboard.
Using board configuration from /opt/pico-sdk/src/boards/include/boards/vgaboard.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
TinyUSB available at /opt/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
cyw43-driver available at /opt/pico-sdk/lib/cyw43-driver
lwIP available at /opt/pico-sdk/lib/lwip
lwIP available at /home/tony/dev/pico/c-code/pico-extras/lib/lwip/src/core/tcp.c; TCP/IP support is available.
Zeta v0.1
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/GNUInstallDirs.cmake:239 (message):
  Unable to determine default CMAKE_INSTALL_LIBDIR directory because no
  target architecture is known.  Please enable at least one language before
  including GNUInstallDirs.
Call Stack (most recent call first):
  /home/tony/dev/pico/c-code/zxspectrum/Zeta/CMakeLists.txt:17 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

Z80 v0.2
-- Found Zeta: /home/tony/dev/pico/c-code/zxspectrum/Zeta (found version "v0.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tony/dev/pico/c-code/zxspectrum/pico-zxspectrum/build

```

however when I get to this point the build fails:
`make -j4 ZxSpectrumPicoVga
`
```
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
/home/tony/dev/pico/c-code/zxspectrum/pico-zxspectrum/src/ZxSpectrumMenu.cpp: In member function 'void ZxSpectrumMenu::nextSnap(int)':
/home/tony/dev/pico/c-code/zxspectrum/pico-zxspectrum/src/ZxSpectrumMenu.cpp:452:39: error: type-specifier invalid in lambda
  452 |   _chooseSnap.next([](FILINFO* finfo) bool {
      |                                       ^~~~
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
[ 37%] Building ASM object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/rp2_common/hardware_irq/irq_handler_chain.S.obj
[ 37%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/opt/pico-sdk/src/common/pico_sync/sem.c.obj
make[3]: *** [bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/build.make:622: bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/__/ZxSpectrumMenu.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:3128: bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:3135: bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/rule] Error 2
make: *** [Makefile:663: ZxSpectrumPicoVga] Error 2

```

Looking at the code in ZxSpectrumMenu.cpp it is just stepping through the snapshot images, so if I comment it out it gets further but then fails here on the same type of declaration:

```
[ 25%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/home/tony/dev/pico/c-code/pico-extras/src/common/pico_scanvideo/vga_modes.c.obj
/home/tony/dev/pico/c-code/zxspectrum/pico-dvi-menu/src/PicoExplorer.cpp: In constructor 'PicoExplorer::PicoExplorer(SdCardFatFsSpi*, FatFsFilePath*, int32_t, int32_t, int32_t, int32_t, int32_t)':
/home/tony/dev/pico/c-code/zxspectrum/pico-dvi-menu/src/PicoExplorer.cpp:21:40: error: type-specifier invalid in lambda
   21 |   _cache.filter([] (const char *fname) bool {
      |                                        ^~~~
[ 25%] Building C object bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/home/tony/dev/pico/c-code/pico-extras/src/common/pico_util_buffer/buffer.c.obj
make[3]: *** [bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/build.make:1899: bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/home/tony/dev/pico/c-code/zxspectrum/pico-dvi-menu/src/PicoExplorer.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:3128: bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:3135: bin/picovga/CMakeFiles/ZxSpectrumPicoVga.dir/rule] Error 2
make: *** [Makefile:663: ZxSpectrumPicoVga] Error 2
```

Wild guess from me - but is this a C compiler version problem? have you used a new way of defining a function with a bool return or something relating to the []'s? or are you needing a different C ompiler version?  any ideas how to over come this would be great.

The pre-built .uf2 file works great on my picodemovga, but I need to modify it a bit that's why I'm looking at build/compile myself.

ArnoldUK, 2023-02-05:
I assume that you are clearing out your build folder after each new build and have run make clean before make -j4 ?
Looks like a compiler issue with those error functions in the log. What build platform are you using ?

pi-gram, 2023-02-05:
yep, doing an rm -r * every time in the build folder - am I missing a an extra library or something?
I'm just using UBuntu 22.04 with compiler versions as shown in cmake output above

fruit-bat, 2023-02-05:
I've removed the bool type-specifier in both cases (it can be inferred anyhow). 
You will need to pull:

- fruit-bat/pico-zxspectrum
- fruit-bat/pico-dvi-menu

There were some outstanding changes on pico-dvi-menu which I have also merged back to main; hope this does not cause any additional problems.

Please, let me know if this helps

pi-gram, 2023-02-06:
nailed it.

[100%] Linking CXX executable ZxSpectrumPicoVga.elf
[100%] Built target ZxSpectrumPicoVga

works a treat now.  thank you!

closed 2023-02-06 by pi-gram