fruit-bat/pico-zxspectrum #69: Composite Video output from VGA signals

ArnoldUK, 2023-02-06:
I've been working on getting a composite video out signal by using an AD724 and the circuit taken from the datasheet, also used [here ](https://www.cpcwiki.eu/index.php?title=RGB_SVideo&mobileaction=toggle_view_mobile) for the Amstrad CPC.

Problem is that although I can get a colour image on my monitor, there is some kind of sync issue as the image is not stable and appears to roll around. I was wanting to find out if anyone has managed to get a stable composite video
out signal ? I think the issue is to do with the CSYNC signal or timing from the Pi to the SYNC signal on the AD724.

fruit-bat, 2023-02-07:
The current code as it is outputs VGA 640x480 at 60hz. Pixels are repeated and horizontal lines are repeated. I think some extra lines that are not displayed are sent too but I would need to look through the code.

Old PAL screens wanted 50hz interlaced frames. Each interlaced frame being half the total vertical resolution of  (or something like that). Maybe there are newer standards I don't know about.

To make this work I think you need to drop the frequency to 50hz and remove the line doubling... And maybe change the number of hidden lines. The sync looks complicated and I don't know how much is actually needed to get frames to lock. 

http://blog.retroleum.co.uk/electronics-articles/pal-tv-timing-and-voltages/

ArnoldUK, 2023-03-08:
Yeah, I think you are right, a bit more involved than I thought. I've given up on the idea for now as it is a huge step backwards in terms of video quality. The VGA output is superb and very stable from my obsverations.

closed 2023-03-09 by ArnoldUK