fruit-bat/pico-zxspectrum #74: Pimoroni DV

larypie, 2023-04-12:
Hi, I got the Pimoroni DV board thinking that I could drop the precompiled UF2 on and the emulator would be ready to go. But nothing happens, no video output, no LEDs flash, nothing.

Do I need to do anything else? Is the emulator in the UF2 or do I need to put it onto an SD card?

I'm sure I'm not the only untechnical (would be) user. Some instructions would be a great help.

ArnoldUK, 2023-04-12:
Could be several reasons and without much info on what board version or UF2 you ar eusing I can't be specific.
Are you using the correct UF2 firmware for the Pimoroni Board ? what UF2 are you using ?
Have you checked your wiring and connections and jumpers are set correctly on the board ?
You may also have an older or updated board that is not pin-pin compatible.
The SD card is only required for loading files and not used for the firmware loading.

fruit-bat, 2023-04-12:
Looks like you need an SD card in for it to boot. Probably just set up with empty folders will do. It is *supposed to* boot without the SD card in. I will try to figure out why it get stuck.

larypie, 2023-04-12:
Thanks for the quick reply. But I am more confused than ever now.

The UF2 is ZxSpectrumPicoDv.uf2
The board is the Pimorini picoDV. 
The Pico is the bog standard, non wifi one.

What wiring/jumpers should I set or check? What versions (of what) should work.

I used the flash nuke UF2 to clear the Pico and the dropped the ZxSpectrumPicoDv.uf2 on.

I also put micro python on the Pico to check it was working and tested the HDMI and power cable on a Raspi Pi to ensure they were good.

The TV reports no video signal. I will try it on some other TVs tomorrow when I have the place to myself.

fruit-bat, 2023-04-12:
@larrypie all sounds fine to me (well, apart from the no signal bit). I take it you don't have an SD card in with the folders set up?

larypie, 2023-04-12:
No, I havent put one in. Was waiting for some to come from fleabay. But these are 32 Gb. I have some 2GB cards somewhere and will try one of these instead when I find them.

Meanwhile I'll check the link.

fruit-bat, 2023-04-12:
Sure, I will have a look at why it gets stuck. There are quite a few targets now and its easy for problems to creep in.

larypie, 2023-04-12:
Panic over, it needs an SD card in before it boots. I am impressed with how quickly the Speccy screen comes up.

Next question, how to add a keyboard. Can I use an OTG cable and hub or do I connect to the board via a serial to USB dongle? 

I'll look through the info on the github site, but if you can point me in the right direction it will be quicker :)

fruit-bat, 2023-04-12:
OTG cable. Not sure if you need a hub - depends on what state tinyusb was in when I built the binary. If you have a hub you can connect joysticks too :-)

larypie, 2023-04-13:
The OTG cable works perfectly. Not tried a hub yet as I dont have access to a joystick.

Thanks for creating the compiled UF2s. The Pico DV one is fantastic. Do you have any plans to do other 8 bit compiuters like the Atari 800, CBM64/Vic20 or BBC?

fruit-bat, 2023-04-13:
yes, would really like to do all of those... but they take a lot of time to do well on a micro controller. The C64 would be good to do just to have a hack at the SID chip. The AY-8912 was difficult to do for the Specy and keep suitably efficient.

larypie, 2023-04-14:
I will keep watching your github page for future developments. The Spectrum version runs really well. I haven't tried audio and I dont have a USB joystick... yet :)

fruit-bat, 2023-04-15:
This is the first one I had a go at for the Pi Pico and has a pre-built binary for the board you are using.

https://github.com/fruit-bat/pico-sorcerer-2

There are a bunch of games hidden in there if you investigate!

larypie, 2023-04-16:
Thanks for the link, the UF2 and info is downloaded. I had heard of the Sorceror way back in the 80s, probably from reading C&VG. It was a surprise to see that it was actually from 1978 and contemporary with the Pet and TRS80.

I will probably buy another PicoDV board next month. At the moment I have the MMBasic VGA and MCB2 boards to play with. I'm also trying to get my remote tank changed over from PicoW to Raspi Pi control.

Too many toys...

fruit-bat, 2023-04-29:
Just an update on the original problem... Some of the targets use the Pimoroni SD card library as the pin assignments don't match up with the hardware SPI. The Pimoroni PIO SPI driver does not seem to time out if no SD card is present; I may be using it incorrectly or it may be broken (needs further investigation).

larypie, 2023-04-30:
Having an SD card inserted to start with isnt a problem. The emulator works brilliantly and I enjoy using it. 

I did find the USB joystick I had didnt work on a hub. Could have been the joystick or  hub. I'll try to find replacements and test again.

grevie2, 2023-12-31:
I'm having similar problems to larypie. I put the same uf2 file on the pico but am not getting any signal from the hdmi output. I have a micro sd card in the slot. I only have a 16gb card at present and think that might be too big. I noticed larypie said they had a 2gb card, but I don't know if the 2gb card is what got things working for him/her.

I've tried formatting the 16gb card with just about every option in windows but am guessing that's not where the issue is.

fruit-bat, 2023-12-31:
Hi, it should just work with the uf2 and boot without an sd card inserted. I'll check the build still works on my board and get back to you. We should be able to get it working. Happy new year!

larypie, 2023-12-31:
@grevie2  Hi, my Pimoroni wouldnt boot without the card, it was perfect with the card in. I only used the 2Gb memory card as that was all I could find and would probably hold everything speccy that I wanted.

I also had a Pimoroni VGA and it worked on that too.

What is your card formatted as? I'm guessing 4Gb or under would be plain vanilla FAT. If I can find a card larger than 4Gb I'll test tomorrow and let you know how it went.

grevie2, 2023-12-31:
Hi @larypie, This is my card:

card size: 16GB
file system: fat32 
allocation unit size: 4096 bytes
format options: quick format

The Pico seems to be okay as I was able to run an Arduino hello world serial program, but still get nothing when I put ZxSpectrumPicoDv.uf2 back on. After installing the uf2 I connected the micro B cable to to the USB power connector on the Pico DV board. I think I am doing everything correctly.

larypie, 2024-01-01:
@grevie2
Hi, I tested with a 32Gb FAT card, a 64GB Linux card and the original 2Gb card.

With no card the Pico wouldnt boot. 
Inserting the 1Gb or 32Gb it booted to the Spectrum menu immediately. Both FAT 32
It wouldnt boot with the 64Gb Pi Zero card.

I should have looked at the 32Gb before I started. I think it was blank. Now it has some folders written when the Pico boots.... The first is called  zxspectrum which contain two folders - snapshots & tapes. The snapshots folder also has one more called quicksaves.

Maybe your card is write protected, or cannot be written to for some reason?

fruit-bat, 2024-01-01:
This one starts for me without SD card....

[ZxSpectrumPicoDv.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13805252/ZxSpectrumPicoDv.uf2.zip)

Note that you need to unzip it to get the uf2 for the pico

larypie, 2024-01-01:
Are there any functional differences other than the sd-card/no card start?

larypie, 2024-01-01:
Fantastic, the joystick was one thing that gave me problems. I'll re-flash my pico.

grevie2, 2024-01-01:
Thanks fruit-bat. That uf2 you posted above worked fine for me! I managed to get to the spectrum menu screen (the one with the light blue selection bar). I did not need an SD card with that uf2 either. I haven't tried a keyboard yet as I don't have the required bits to hand but it's a big step forward for me, and I feel very happy now. Thanks again and thanks to larypie for helping too. :-)