fruit-bat/pico-zxspectrum #75: Where I can find the USB pins?

Chandler-Kluser, 2023-05-03:
Hello, I am assembling the Breadboard Version of pico-zxspectrum and I saw it is compatible with USB and PS2 keyboards.

I do not have any PS2 keyboards with me, but I looked for the D+ and D- pins to connect to my USB keyboard at pinout figure but I did not find any dedicated pin.

Could someone help me finding that? Thanks in advance!

fruit-bat, 2023-05-04:
You can use an OTG adapter like this 
https://www.amazon.co.uk/CentBest-Splitter-Enhancer-Charging-Converter/dp/B072Z9QHYR/

Some have a built in hub, which I will post an example link if I can find one.

Or, if you are good at soldering, there are a couple of small pads on the back of the Pi Pico near the USB connector. If you add your own host connector, only use either the Pi Pico usb connector or yours, not both at the same time.

Chandler-Kluser, 2023-05-04:
Thanks, I will try that :)

Chandler-Kluser, 2023-05-14:
I have made a small prototype with and HDMI PMOD Adapter + TF Adapter + USB Type A Connector:

[image]
[image]

Then I connected:

| USB CONNECTOR | RPI PICO |
|:-------------:|:--------:|
|     VBUS      |   VBUS   |
|      D+       |   TP3    |
|      D-       |   TP2    |
|     GND       |   GND    |

as @ArnoldUK mentioned. The HDMI pins were connected too.

The Board gave video output, as expected:

[image]

But when I attached my USB keyboard, it did not work...

[image]

I also switched the D+ and D- just in case of I exchanged the pins by myself for accident, but it did not work either...

Are you sure the breadboard version does really support USB keyboards? Thanks in advance for your support!

ArnoldUK, 2023-05-14:
If you are using your own USB OTG wiring to the D+ and D- pads then you cannot use the micro-usb socket as the D+ and D- in the micro-usb socket conflict. You should remove that micro-usb cable labeled USLION and power the pi directly with 5V to VSYS.

Chandler-Kluser, 2023-05-14:
Thanks a lot!!

Wiring 5V power at VBUS and GND worked flawlessly!!!

I thought that connecting a Power Supply at OTG connector would not face issues but you are right. Thanks a lot!!

closed 2023-05-14 by Chandler-Kluser

ArnoldUK, 2023-05-14:
Yeah, glad to hear that worked. Like a said, some power adapters pull those data line low, others leave them floating.