fruit-bat/pico-zxspectrum #83: Kempston Joystick

digitalmexer, 2023-06-06:
I use the ingenious emulator PICOZX from Bobritius. Everything works very well but there is a problem with the Kempston interface:
When loading programs with the load function for TZX files, the joystick works fine with a snapshot not. The USB joystick always works.
Unfortunately I'm not a programmer so I can't find any hints about the bug in the sources.

ArnoldUK, 2023-06-06:
I don't think the snapshots save the emulator joystick settings. reconfigure the joystick after loading the snapshot.

digitalmexer, 2023-06-07:
You're right, I don't think so either. 
But I have found out something else:
The error is not in the snapshots, but how they are activated. If a snapshot is copied to the quicksave folder and renamed to "Slot 1.Z80", the Kempston interface works fine. If the same snapshot is loaded from another directory, the joystick does not work.

ArnoldUK, 2023-06-07:
I'm not quite sure how the snapshots are saved or which folder but the snapshots are simply a snapshot of the spectrum memory and AFAIK do not restore the emulator settings when it was saved. Maybe you could request a feature to add that.

fruit-bat, 2023-06-07:
> You're right, I don't think so either. But I have found out something else: The error is not in the snapshots, but how they are activated. If a snapshot is copied to the quicksave folder and renamed [...]

This is curious. Are you referring to the using the buttons on the keyboard as the Kempston joystick? Also, are you aware that shift-Kempston switches the buttons from acting as cursor keys to a Kempston joystick (I only ask as I have not documented this well)? When the keyboard buttons are acting as a joystick the pico led should be on.

digitalmexer, 2023-06-08:
Switching between the two functions of the joystick switch works. The idea of displaying the switching state via the PICO LED is ingenious. I am sure that I am operating the emulator correctly.
I do not believe in a hardware error. 
It would be nice if other users of the PICOZX hardware from Bobritius would comment on the problem. 
It seems that the cause of the effect is due to differences in the startup routines of the Raspberry program parts.

adamenkov, 2023-06-08:
I'd like to comment on this, but I can't even find a keyboard that would work with PICOZX.
I have more complaints about the software:
- it doesn't start if there's no microSD inserted, have fun guessing why,
- it hangs if you press replay or previous/next replay and there are no replays,
- it doesn't memorize the volume level,
- it only has one quick save, unless you use an external keyboard, and I haven't yet found a compatible keyboard,
- I naively expected TR-DOS, at least in BASIC, otherwise you can't save your work (except in the only quicksave).
I would make Issues from this, I just don't know what's the proper way to submit bugs/suggestions.

fruit-bat, 2023-06-10:
> I'd like to comment on this, but I can't even find a keyboard that would work with PICOZX. I have more complaints about the software:
>
> * it doesn't start if there's no microSD inserted, have fun [...]

Ok, so first thing to note is that I just wrote this for fun. I do not receive funding, it's published so folks can have a bit of amusement with a micro-controller. Bobricius kindly posted some of his boards to me so that I could add support for them, which I had a lot of fun doing :-)

> I'd like to comment on this, but I can't even find a keyboard that would work with PICOZX. I have more complaints about the software:

I have not come across a USB keyboard that does not work with my builds. You most likely have a problem with the leads you are using. Some USB leads only route power. Of course I do not know which keyboards you have tried, or which version of the firmware, or which of the now many targets you are using.

> * it doesn't start if there's no microSD inserted, have fun guessing why,

The latest version of the firmware should have the problem with the startup fixed.

> * it hangs if you press replay or previous/next replay and there are no replays,

Please raise an issue or supply a code fix.

> * it only has one quick save, unless you use an external keyboard, and I haven't yet found a compatible keyboard,

Key-mapping with a limited number of keys is surprisingly difficult; there are some technical limitations in keeping the code small and quick. If you have a *well thought out* suggestion please raise an issue.

> * I naively expected TR-DOS, at least in BASIC, otherwise you can't save your work (except in the only quicksave).

Please raise an issue or supply a code fix.

>   I would make Issues from this, I just don't know what's the proper way to submit bugs/suggestions.

I am happy for you to raise issues even if you are intending to supply code modifications. If I know you are making changes before they are pull-requested then I can arrange a suitable branch that we can collaborate on.

The key to raising an issue is detail. I only get short blocks of time to play with this code so describing the problem and possible solutions concisely means I might get round to fixing them. Try to keep issues to *one* topic. It is useful for me to know:
* The hardware you are using
* The name of the firmware (these have been referred to as 'target's in the Readme)
* The version of the firmware you are using
* Which display is being used (if there is more than one supported by the hardware)

adamenkov, 2023-06-10:
Understood and thanks a lot for your detailed explanations!

digitalmexer, 2023-06-13:
The problem with the Kempston joystick is in the firmware for the PICOZX. The uf2 file of the latest release "ZxSpectrumPicomputerVGA222Zx.uf" brings the described error. The firmware for the PICOZX-LCD "ZxSpectrumPicomputerZxLcd.uf2" works without errors with the same device.

fruit-bat, 2023-06-13:
I've had play on the PICOZX and it works fine for me. Please give exact steps from power on to reproduce.

digitalmexer, 2023-06-13:
1st step: I open the Github page https://github.com/fruit-bat/pico-zxspectrum and load the firmware: "ZxSpectrumPicomputerVga222Zx.uf2".
2nd step: i load the firmware into the PICOZX and start the emulator.
3rd step: I open the PICOZX main menu, change to the Snapshot subdirectory and load the file "Exolon.z80" from the snapshot directory of the SD card, the program starts.
Step 4: I switch on the Kempston joystick function with [CapsShift + Next] (the LED on the Raspberry pico module lights up) and set the control for Kempston joystick in the game menu [5].
Step 5: I start the game by pressing [1]. With the cursor keys no action is possible, they do not work as joystick.
6th step: I open the Github page again: https://github.com/fruit-bat/pico-zxspectrum and load the firmware: "ZxSpectrumPicomputerZxLcd.uf2".
7th step: I repeat step 2 (in VGA mode) step 3 and 4.
8th step: I start the game with key [1]. Everything works as expected.

- With a USB joystick everything always works.
- Snapshot files from the "quicksave" folder always work.
- Programs loaded with the tape player always work too.

It is very strange that this behavior has only been noticed by me so far. But I can't find a fault in my hardware. Maybe the Raspberry Pico has problems?

Matthias

digitalmexer, 2023-06-13:
It is not a problem with this particular game, other programs behave exactly the same way. I have tried several.

ArnoldUK, 2023-06-13:
Have you built the hardware from scratch yourself or was it pre-built ? I onyl ask because maybe you have a an issue with your hardware board. There cannot be a problem with the raspberry pi pico becasue you say it works when you use the `ZxSpectrumPicomputerZxLcd.uf2` firmware. So I suspect the firmware that you are tryin to use is not fully supported or designed for your hardware.

digitalmexer, 2023-06-13:
The hardware is based on the Bobritius board set: 
 https://www.tindie.com/products/bobricius/picozx-zx-spectrum-128k-vga-emulator-on-rpi-pico/
You can't really go wrong there. There are only a few components to assemble and they all seem to work well.

ArnoldUK, 2023-06-13:
There's likely a fault with the PCB or your solder connections then.

digitalmexer, 2023-06-13:
Yes, that may be, but it is very unlikely. It does not really matter, I have a firmware that works with my board without problems, that's enough for me.

ArnoldUK, 2023-06-13:
The firmware that works is designed for your board, other firmware will not have the same pins mapped out for the PICO so I guess that is where the issue is. There are so many different boards, makes it difficult to support them all with a single firmware.

fruit-bat, 2023-06-14:
> 1st step: I open the Github page https://github.com/fruit-bat/pico-zxspectrum and load the firmware: "ZxSpectrumPicomputerVga222Zx.uf2". 2nd step: i load the firmware into the PICOZX and start the [...]

Can reproduce now. Thanks for the detail. Will look into it.

fruit-bat, 2023-06-14:
[digitalmexer](https://github.com/digitalmexer) I've pushed a new ZxSpectrumPicomputerVga222Zx.uf2

Please let me know if this sorts the issue for you. Thanks for bearing with me on this one :-)

digitalmexer, 2023-06-14:
Thank you for the good and fast work:
Magic - everything works. Since I have hardly any idea about programming, I am always very impressed about the feats of software developers.
By the way, the cursor keys in the main menu also work when Kempston mode is activated is very pleasant.
I am curious how this project will develop further. 
I like it extraordinarily well.

closed 2023-06-14 by digitalmexer