fruit-bat/pico-zxspectrum #84: Firmware update manual needed

adamenkov, 2023-06-10:
I have PICOZX and I'd like to be able to update my firmware once in a while.
Could you please add a section to the README.md file on how to do it?
Or if it's already explained somewhere, please just refer me to that manual.

fruit-bat, 2023-06-10:
I've expanded on the notes in the section headed 'Firmware targets'. Let me know how you get on.

adamenkov, 2023-06-11:
Thank you for this.  By the way, where is this BOOTSEL button?  I couldn't find it in the README.md.

fruit-bat, 2023-06-12:
I've added a pic and some more notes

adamenkov, 2023-06-12:
I appreciate all your help and the additions to the README.md file.

ArnoldUK, 2023-06-12:
> Thank you, but for me it's `picozx`, not just `pico`. Guess which webpage Google refers me to. :)

But you asked where the BOOTSEL button was on the pico and you have a raspberry pi pico not a picozx so why search for a picozx ?

fruit-bat, 2023-06-18:
> I appreciate all your help and the additions to the README.md file.

@adamenkov Did you manage to upgrade your firmware successfully?

adamenkov, 2023-06-18:
> @adamenkov Did you manage to upgrade your firmware successfully?

I found the BOOTSEL button but I couldn't connect the PICOZX board to either PC or Mac.  I turned off the power, took the battery off, pressed the BOOTSEL button, connected the USB to my computer, released the BOOTSEL button - like in your manual and some YouTube tutorials.  Tried this with a number of variations, nothing worked.

Thanks anyway, and let me close the issue.

ArnoldUK, 2023-06-18:
How did you get the firmware on the PICO in the first place ?

adamenkov, 2023-06-18:
Purchased pre-flashed.

ArnoldUK, 2023-06-18:
In that case the instructions to update the firmware should have been provided with the board.

adamenkov, 2023-06-18:
"the PICOZX Lcd board has the Pico facing inwards and you will need to reach a finger inside to press the bootsel button."

I finally managed to connect the Pico to my laptop.  Mostly it was my fault - I used the cable that was only good for power supply, not data transfer.  But the critical piece of the puzzle was that the PICOZX had to be... ON, i.e. no battery (I guess), but the Power On/Off switch should be set to On, otherwise it doesn't work.  Also, you don't need to keep pressing the Bootsel while connecting the cable - as Bobricius told me it's enough to just press Reset shortly while keeping the Bootsel button down.  Also, there's no way you can press the Bootsel with your finger on PICOZX (the one with display) - unless you remove the cover, I for once used a ceramic knife. :)

ArnoldUK, 2023-06-19:
I guess the pico is soldered to the board then and not removable :(