fruit-bat/pico-zxspectrum #85: PICOZX TFT

digitalmexer, 2023-06-13:
I accidentally bought HC28010-D TFT displays. Unlike the HC28010-ISP displays, these display the image inversely. Is it possible to adjust this in the software without much effort? Maybe I am not the only one who wants to use such "inverse displays". 

Matthias

fruit-bat, 2023-06-15:
'inverse' in what sense? 
Is this a colour thing or rotation/flip thing?

digitalmexer, 2023-06-15:
It works perfectly ! 
Thank you very much.  
Yes, I meant the representation of the colors, I should have been more specific. 
I think it would be a good idea to enable the color switching in each firmware version. Some programs are more recognizable on the small display in the inverse representation. It also expands the possible uses of the available hardware.
Couldn't a config file from the SD card be used for such settings? I'm sure there will be other functions added later

I am very happy about this firmware extension, now I can connect a slightly larger display to Bobricius hardware. It becomes more and more beautiful.

digitalmexer, 2023-06-15:
The display in the main menu looks good. Whether black on white or white on black makes no difference. In my opinion, nothing needs to be changed there.

fruit-bat, 2023-06-17:
I've tidied up the code and added a new target ([ZxSpectrumPicomputerZxInverseLcd.uf2 ](https://github.com/fruit-bat/pico-zxspectrum/blob/main/uf2/ZxSpectrumPicomputerZxInverseLcd.uf2) ) to the 'uf2' folder.

I agree that some way to persist configuration would be nice. Not sure the SD card is the best place. Possibly I could write it to the built in memory on the RP2040. I don't know how to do this at the moment. Will do some reading when I get chance :-)

FYI (and so I remember) it looks like the LCD controller has a command to negate the colours but I couldn't make it work. The final code modifications change the way the RGB colour tables are built for the spectrum and menus.

fruit-bat, 2023-06-17:
+ please post a pic of the working screen, so I can see it looking good :-)

digitalmexer, 2023-06-17:
Thanks again for this useful extension, I have downloaded the new firmware version from Github and tested it. Everything works fine.
I think changing the RGB code tables was the only way because the ST7789 with RGB interface does not support inverse display only with i80 interface.  
![ST7789VW_InterfaceMode](https://github.com/fruit-bat/pico-zxspectrum/assets/135758851/4e6bb73b-
5fee-4c38-bda6-533f64cc7f1e)

However, I have only read this and have not tried it myself.
Below I post some images in inverted and non-inverted color mode. The display also works with a ST7789V chip in SPI mode and has a 3 inch screen.
[image]

![Exolon_inv_377](https://github.com/fruit-bat/pico-zxspectrum/assets/135758851/09e76c87-7e41-4676-9ce9-fb5f

[image]
a8ce
[image]
3be3)

digitalmexer, 2023-06-17:
I have other thoughts about extending the software:
1. is it possible to install the firmware from the SD card of the emulator? 
2. is it possible to make the switch from LCD to VGA without reset. So without having to reload the emulated application program?

fruit-bat, 2023-06-18:
> I have other thoughts about extending the software:
> 
> 1. is it possible to install the firmware from the SD card of the emulator?

Possibly but I really don't know how to do it. I will raise an issue.

> 2. is it possible to make the switch from LCD to VGA without reset. So without having to reload the emulated application program?

Possibly, the issue here is shutting down the VGA driver. I 'adapted' a VGA library from another project and I think it needs some careful (and possibly extensive) work to close it down in a tidy fashion. I will raise an issue.

closed 2023-11-01 by digitalmexer