fruit-bat/pico-zxspectrum #90: It hangs when in the menu you try to refresh Snaps or Tapes without a micro SD inserted

adamenkov, 2023-06-19:


ArnoldUK, 2023-06-19:
Put an sd card in or dont press refresh when a card is not inserted ?

adamenkov, 2023-06-19:
That would be too easy. :)

A message like "Insert micro SD card" would suffice for me.

ArnoldUK, 2023-06-20:
How about a post-it note with "don't forget the sd-card" stuck to your device. Now I know I sound sarcastic but asking for such a pointless feature is asking for a pointless reply. There are so many things that are required that would throw errors. For example, not plugging in the power would cause the emulator to not boot, or unplugging the video cable would show a blank screen :)

adamenkov, 2023-06-20:
That’s funny :) but please bear with me for a minute.  I’m not trying to find work for you.  The issue is serious enough for me that I made an Issue about it.  My situation is that I have a PICOZX, for which I haven’t found a compatible external keyboard yet.  This means that if I want to have more than one snap (as a way to save a collection of BASIC programs, for example), I need to rename the default snap.  At the moment, when renaming files, I can only add characters to snap file names (to recall, I don’t have an external keyboard), so to do proper renaming I need to take the micro SD out, plug it in another computer, do the renaming, maybe move some snaps to a different directory.  It is possible that I forget to plug the micro SD back in the PICOZX, and when I do, the next quick save attempt halts everything.  I can’t see if the micro SD is in, the slot is below the keyboard.

I’m just asking to make the thing a little bit more friendly.  It’s not a high-priority request, more like a nice-to-have feature.

ArnoldUK, 2023-06-20:
Well, the file rename without an external keyboard requires more work and yes would solve your problem. But, I would not like to have a message pop up when a card is not inserted as I use the emulator without an sd card in some of my projects and the constant message popping up would annoy me. I get your point though.

adamenkov, 2023-06-20:
No need for a popping up.  For example, instead of the list of files, it could just show "(no micro SD)".  The hang only happens when I press 5 (refresh) without a micro SD.  So the message could only show up in this situation.

ArnoldUK, 2023-06-20:
Well, the emulator did not start if a card was not inserted but that was fixed. I still think the emulator menu should be disabled if a card is not inserted. Would be much less prone to errors when the emulator is used without a card.

adamenkov, 2023-06-20:
Just disable the menu?  Half measures!  How about disable everything?  Electric shock as penalty for forgetfulness?  Self-destruct? :)

Seriously, the menu is still useful to adjust the volume or reset in a different (48K/128K) mode, even without the micro SD.  All I want is just that my PICOZX doesn't hang.  If there's a short message "(no micro SD)", that would be just super awesome. :)

ArnoldUK, 2023-06-20:
I'll go with the self-destruct, gives the user a second chance :)

Oh, yeah I forgot about the 48K/128K reset mode, that is needed. There is only so far a programmer can protect the end user from their sillyness, not pointing fingers :) But let's that a message is added when the card isn't detected, then more checks will be required to check the card is correctly formatted or not corrupted. That's why I mentioned to disable the menu system, or at least all card required menu items.

adamenkov, 2023-06-20:
Maybe just a "(can't read micro SD)"? :) - to catch all possible scenarios: no micro SD, micro SD not formatted properly etc.

It's really a good idea to be friendly to the user when possible.  One day it might be you! :)

fruit-bat, 2023-06-20:
It hangs less now v0.35.  There still seems to be an issue if you refresh repeatedly.

ArnoldUK, 2023-06-20:
> 

Sure, but only a couple of programmers contribute to this repo and don't get much of a reward from the users so I guess focus is on more making it stable. Hey, I use Windows so I'm the butt end of non user friendly ui's

fruit-bat, 2023-06-21:
The UI needs to operate in a tiny amount of memory (RAM). Just getting a list of files to work with is challenging. Hence all the scanning folders behaviour.

fruit-bat, 2023-06-21:
Is it any more stable now? For some reason repeatedly attempting to open a non existent card causes problems for me.

adamenkov, 2023-06-21:
> The UI needs to operate in a tiny amount of memory (RAM).

Ah ok, I see.

fruit-bat, 2023-06-24:
Thanks @ArnoldUK, this matches what I have found. The idea of the repeated attempts to mount the SD card are there to try and cope with a card being removed or inserted (in a limited way).  I  think the issue lies within the _sdCard->mount() call, which seems to result in memory corruption. Not figured out why so far.

ArnoldUK, 2023-06-24:
Yes, I agree and I struggled to pin point the exact code so ended up commenting out all calls to detect or mount the card. Sorry, I've not submitted any pull requests but I'm struggling with github. I will try my best to upload my code addditions and updates to my github page. I'm working on another project at the moment so don't expect anything quick.

Chandler-Kluser, 2023-08-28:
I am facing issues of mounting the SD card, too.

It is used to crash, but now I am booting without it connected, so it has been solved.

But now it scans the SD card but do not find anything neither creates any directory inside it, can someone help me?

Thanks in advance!

ArnoldUK, 2023-08-28:
I've found If you have too many files in the tapes or snapshots folders then ths can cause the file caching to take some time to rebuild or often freeze. Also some long filenames with odd special characters can cause problems. I suggest adding around 10 tape/snapshot files and try again.

Chandler-Kluser, 2023-08-28:
I have placed only one .z80 snapshot and it didn't work

ArnoldUK, 2023-08-28:
Try another smaller SD-card and make sure to format as FAT not FAT32. The firmware works fine as I'm using it myself with and without an SD-card.

Chandler-Kluser, 2023-08-29:
you are right, these modules have 5V lines + level shifters and I did not observe it.

But now I wired VCC into 5V directly and it did not work either, can I replace the module for an adapter like this one?

[image]

ArnoldUK, 2023-08-29:
Yes, but look at the schematics of all the circuits to check how all the pins. A few simple dropper resistors also work fine to level shift the voltage.

Chandler-Kluser, 2023-08-30:
it worked with a cheap SD to microSD adapter!

[image]

[image]

ArnoldUK, 2023-08-31:
Yeah, JetPac Excellent! I see you have used directly soldered wires. That's a much more reliable setup than jumper wires, I've often wasted hours of problem solving because of cheap poorly connected jumper wires.