fruit-bat/pico-zxspectrum #96: Pimoroni Pico VGA Demo Base and a keyboard?

DualBrain, 2023-09-13:
How does one go about connecting a keyboard to this board for use with this project? USB? PS/2?

It looks like people are doing it but I'm not finding anywhere in the issues and/or readme.md (main info) on the specifics other than a statement that a keyboard (USB/PS2) is supported.

Thanks.

ArnoldUK, 2023-09-13:
There are plenty of example circuits with pinouts on the main page showing how to connect a PS/2 keyboard and which pins to use. It does however require some elctronics knowledge and soldering skills. This project is for building your own circuits unless you use a ready made board from either MURMULATOR or Bobricius.

fruit-bat, 2023-09-16:
Thanks for helping out Arnold!

Just to add a bit of USB information... To use a USB keyboard you will need either a USB hub or an OTG cable. This should be plugged into the Pico and power connected to the extra micro-usb connector on the board.

All my USB keyboards just work fine but there may be some compatibility issues. Just go with a cheap, simple keyboard. 

I have added some links below as examples,obviously you can source your cables from wherever you like :-)

Here is a simple OTG cable
https://www.amazon.co.uk/UGREEN-Adapter-Android-Adaptor-LenovoTablet-black/dp/B00LN3LQKQ

This OTG lead has an extra power input (but you don't need it on the Pimoroni board)
https://www.amazon.co.uk/adaptare-cable-Samsung-Galaxy-Micro-USB-Anschluss/dp/B00EJP5XR2/

And this is a hub, so you can connect a keyboard and joystick
https://www.amazon.co.uk/Mavis-Laven-Extension-Compatible-Smartphone-default/dp/B07R3X9RCH

DualBrain, 2023-09-16:
Looks like I purchased two of the first one you linked back in 2017... time to rummage around. ;-)

Thanks for the USB info, I'll certainly give that a go before (if?) taking the PS/2 route.

ArnoldUK, 2023-09-16:
Oh yes, I forgot the USB OTG cable route, that also works too and does not require the complicated wiring and soldering. Also with a hub a USB joystick and keyboard can used together.

sonietje, 2024-07-31:
Hi,
I managed to connect a USB keyboard to the pico but none of the shift keys are working. When I press shift ; I get nothing while I expect : 
I tried with different USB keyboards but no luck. 
Thanks for looking into this.