fruit-bat/pico-zxspectrum #97: Audio Question

BrettRogersUK, 2023-10-19:
HI,

First off, I must take my hat off to you for producing such a wonderful project!

I have a question about the audio side of things. I'm just making a breadboard like your original prototype as can be seen on the readme. I will get round to getting a PCB done eventually. However, I have purchased the speaker listed in the components section of this github (https://shop.pimoroni.com/products/adafruit-stemma-speaker-plug-and-play-audio-amplifier) but am unsure as to how to wire it up. It only has 3 terminals (GND, +ve and Sig).

How would I go about wiring this up so that I can get audio output from the Pi Pico? Would it need to be done with the 1 pin audio setup?

Many Thanks and I look forward to trying this out on my brand new Pi Pico. 

Kindest Regards,

Brett

fruit-bat, 2023-10-20:
Hi Brett, glad you are having fun with the project. 

The 4 pin audio is the best quality. For this, you will need a few components to make filter, shown here:

https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/Pico%204%20pin%20PWM%20audio%20filter%20mono.png

Feed the 'audio out' to the signal line on the little amplifier. Power the amp with 5v from the USB.

fruit-bat, 2024-04-03:
Brett, you can now get audio over HDMI.