fruit-bat/pico-zxspectrum #98: Keyboard issue

BrettRogersUK, 2023-10-19:
Hi

I've built this on a breadboard with all connections as per the wiring diagram for ZxSpectrumBreadboardHdmiNPinAudio section.

I've got my power being supplied by USB cable to the VSYS and GND connections via a Micro USB breakout board with just the VCC and GND connections attached and I've got a small Micro USB to USB-A (Female) connector on the micro usb connector attached to the Pico. I get a wonderful steady picture with the 48k/128k/tape loader screen up but thats about as far as I can go.

I'm using an old Dell PS/2 keyboard via a PS/2 to USB connector but I'm not getting anything when pressing on the keys for example pressing F1 is supposed to bring up the menu but I don't get anything.

I've also tried with a wireless keyboard with a dongle that plugs in to the USB end but that doesn't work either. I'm getting 4.97v on the VSYS pin and 3.3v on the 3.3v output pin. This is the PS/2 to USB adapter I'm using (https://amzn.eu/d/4XPk16J).

I wonder if anyone can help me out.

Many Thanks,

Brett

fruit-bat, 2023-10-20:
PS/2 to usb adapters might look quite different to the Pico from a 'normal' USB keyboard.  You would be better off wiring it in directly:

https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/Pico%20PS2%20interface.png

Since making that diagram I have read various articles saying the RP2040 pins are 5v safe so you can probably get away without the zener diodes.  Also, they are only there in case the keyboard has pull up resistors, which I have never actually found.

Not sure why the wireless keyboard does not work. I use one on mine and it works OK. Mine is a logitech k400.

I don't think det is used anymore but I will check and get back to you.

BrettRogersUK, 2023-10-21:
Hi, just thought I'd give an update. I got the OTG cable and plugged it in with power going to the optional extra power jack and I thought I'd try my PS/2 TO USB adapter again. SUCCESS! I also tried my small Rii Wireless keyboard too with a dongle and that worked as well. 

I think the problem was that there wasn't any power getting to the keyboard for some reason with my previous set up. 

Would it help if the 5v supply line was on vbus instead of vsys?  

Got to get the sound circuit built now so I can get some sound of it. 

Thanks again for the work you've put into this project with the help of others.

Regards,

Brett

BrettRogersUK, 2023-10-21:
When using a PS/2 OR USB keyboard what are the key mappings for keys like BREAK for example?

fruit-bat, 2023-10-22:
Glad the keyboard is now working.

Key mappings for usb keyboards are in here:
https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/ZxSpectrumHidKeyboard.cpp

Is your sd card working (I don't think you need the card detect line connecte)?

BrettRogersUK, 2023-10-24:
> Glad the keyboard is now working.
> 
> Key mappings for usb keyboards are in here: https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/ZxSpectrumHidKeyboard.cpp
> 
> Is your sd card working (I don't think you need the card detect line connecte)?

Hi, sorry for the late reply. Yes the SD card is working flawlessly. Can't fault it. I've still got to build the audio filter circuit to test the sound out. 

What recommendations for games do you recommend to really test the sound. I'm guessing 48k would have just been beeps and the 128k would have used the AY sound chip.

Cheers

Brett

fruit-bat, 2023-10-25:
Good sound tracks on:

- Batman
- Bedlam
- Auf Wiedersehen Monty
- Destiny Mission