fruit-bat/pico-zxspectrum #99: PIMORONI PicoVision support

michal-konopinski, 2023-10-22:
Hi, is it possible to add support to this hardware?

Rijast, 2023-10-22:
I also have one of these and was wondering if it is supported? thanks!

fruit-bat, 2023-10-25:
Not supported at the moment. It is probably a reasonable amount of work to port due to the dual RP2040s. 
...I do think it would be a nice platform though; offloading the keyboard/usb handling would make life easier for the emulator, in particular the sound generation.

Rijast, 2023-10-25:
Thanks, great to consider for the future. I am going to use the Pimoroni Pico DV Demo Base - awsome project btw!

darrencuk, 2024-01-18:
This looks an awesome project. I also have a PicoVison and would love it if you could support that too!