fruit-bat/pico-zxspectrum #100: PICOZX with membrane keyboard better menu operation

digitalmexer, 2023-10-23:
I have installed a PICOZX-VGA board from Bobritius in an original 48 K case with foil keyboard. 
The firmware: "ZxSpectrumPicomputerVgaAuK.uf2" works fine. 
Unfortunately the operation of the menu is very limited. 
The use of a standard joystick is also not possible, because the down button of the joystick is used for the menu call.

Is it possible to add the navigation in the menu so that it can also be operated exclusively with the key codes of the Spectrum keyboard matrix?
Cursor navigation:
left :   CAPS-SHIFT + 5
right : CAPS-SHIFT + 8
up:     CAPS-SHIFT + 7
down: CAPS-SHIFT + 6
Activating the selection with ENTER already works.
The menu could be called via CAPS-SHIFT + SPACE.
Calling up the menu via the joystick down key must be deactivated.

The joystick should be activated automatically according to menu item "6".

This means that the functions of the "special keys" (ROW1 - COL1 to COL8) can be retained and can be used for extensions.

On the board design only the wiring of the old menu key is changed to use the proposed functions.

Matthias

fruit-bat, 2023-10-24:
I think I can do that.  It's a little while since I have looked at the code so bear with me.

fruit-bat, 2023-10-25:
I don't have any hardware that matches your setup... so I've had a guess at the changes needed for the keyboard. I will post it here for you to try. If you are happy to test the binaries and report back I think we can make some progress. 

Note that attachment will need to be unzipped.

I don't even know if this will boot so let's start from basics... let me know if it works at all. 

The key mappings as suggested have been added, so let me know if any/all work. I have also added CAPS-SHIFT + 1 to exit the menu. I've not used the same as to enter the menu in case it flips back and forth - we could experiment with that later.

[ZxSpectrumPicomputerVgaAuk.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13169985/ZxSpectrumPicomputerVgaAuk.uf2.zip)

bobricius, 2023-10-25:
Hi, By my quick test
- menu is displayed but it need very quick press because disappear, .... need delay after menu is displayed
- menu go directly to chose file to load submenu or to last activated menu
- if menu is activated with menu button go to main menu

which combination is ESC? or back one level menu ?

Thanks

digitalmexer, 2023-10-25:
Thank you for the quick action. It already looks very good. :)
The additional key assignments for navigation work flawlessly. Calling up the menu also works flawlessly.
Exiting the menu with SHIFT+1 also works.

Unfortunately, switching back and forth between menu and program with SHIFT+SPACE is difficult.

SPACE key and the ENTER key have the same effect in the menu: the menu entry is activated. In the menu there is no difference between SPACE alone and SHIFT+SPACE. Therefore one jumps from the program to the menu and immediately back to the program when SHIFT + SPACE is pressed.
If you press SHIFT+SPACE very briefly, you can sometimes return to the menu.

Calling the menu with the old method (ROW1 - COL6) does not work again. But the call with ROW1-COL5 is still possible.

@bobricius
go back one level with CAPS-SHIFT + 5 = cursor right

digitalmexer, 2023-10-25:
If the menu item is executed only with ENTER, this should solve some problems. The SPACE key must not also trigger this action.

fruit-bat, 2023-10-26:
Thanks for the detailed feedback. Give this a go:

[ZxSpectrumPicomputerVgaAuk.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13182960/ZxSpectrumPicomputerVgaAuk.uf2.zip)

digitalmexer, 2023-10-27:
thank you very much, I have tested the new version:

there are still problems with the function of the SPACE button when selecting snapshot and tape files.

In detail:
Calling the menu with CAPS+ENTER and 
Return to the program with CAPS+1 works.

Main menu: **No problems**
CAPS+5 (left): no action - OK
CAPS+6 (down): Cursor navigation down - OK
CAPS+7 (up): Cursor navigation up - OK
CAPS+8 (right): Menu item is activated - OK
SPACE : no reaction - OK
ENTER: Menu item is activated - OK
_____________________
Menu 1 (Snapshot): **Misfunction**
CAPS+5 (left): back to main menu - OK
CAPS+6 (down): Cursor navigation down (loop)- OK
CAPS+7 (up): Cursor navigation up (loop)- OK
CAPS+8 (right): Menu item is activated - OK
**SPACE : Menu item is activated - not OK**
ENTER: Menu item is activated - OK
The reactions in the subfolders behave the same way
_____________________
Menu 2 (Tape player): **No problems**
CAPS+5 (left): back to main menu - OK
CAPS+6 (down): Cursor navigation down - OK
CAPS+7 (up): Cursor navigation up - OK
CAPS+8 (right): Menu item is activated - OK
SPACE : Menu item is not activated - OK
ENTER: Menu item is activated - OK
_____________________
First submenu 2 (Tape player - Tapes): **Misfunction**
CAPS+5 (left): back to tape player menu - OK
CAPS+6 (down): Cursor navigation down (loop) - OK
CAPS+7 (up): Cursor navigation up (loop) - OK
CAPS+8 (right): Menu item is activated - OK
**SPACE : Menu item is activated - not OK**
ENTER: Menu item is activated - OK
_____________________

The main menu items 3 to 8 : **No problems**

fruit-bat, 2023-10-27:
Once again, thanks for the detail, give this a try:

[ZxSpectrumPicomputerVgaAuk.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13190269/ZxSpectrumPicomputerVgaAuk.uf2.zip)

digitalmexer, 2023-10-27:
Perfect, The navigation and activation in the menus works as it should. 

With this and a USB joystick you can get along well.

Nevertheless, I would wish that the old functions of the special keys and the joystick connector (as with the ZXPICO LCD) were enabled again.
I have the connections for the joystick routed to a 6 pin din socket. This fits exactly into the recess for the modulator. The special keys I also want to accommodate their function I found very convenient.

The reactivation of the joystick will perhaps not please Bobritius because his menu button no longer works as before.

digitalmexer, 2023-10-27:
[image]

[image]

bobricius, 2023-10-27:
Hi boys, 
I really appreciate that control via rubber buttons.
I'm trying to create a better motherboard, specially for original case, without cover board, LCD ....
It's on the way from the factory
PCB based on the AUK modified scheme.
**Please check whether I made a mistake in the connection.**
Features, not yet tested:
-TH components, absolute minimum
- The PCB should have the same dimensions as the original ZX
- keyboard is similar in size to the ZX and could probably fit into the faceplate with rubber buttons, of course it certainly won't fit instead of foil
- in the future, however, it will be possible to cut off the keyboard
- it will be possible to use foil connectors without changes and use a rubber keyboard
- all components are now from the upper side (vga, jojstick, usb)
-VGA resistors can be used both TH and SMD
**connector alignment:**
- reset to the hole for the modulator
- headphone output to the audio output hole
- audio input to SD card
- the expansion connector should fit USB, VGA and a joystick
- USB-B power connector (as on a printer)
- the other buttons have a place on the PCB, but probably whoever wants to install them in the original package will not want to drill holes, but they will be used in such a bare version on the table
- there is a large piezo buzzer for sound output, it does not need any amplifier

[image]
[image]
[image]

digitalmexer, 2023-10-27:
Hi Bobricius,

the new board looks very good.
Good that we could improve the menu navigation, we will need that for this board version.
When using the buttons on the board, long plunger buttons would have to be used because now the VGA and joystick connector are on the other side of the board.
Have you measured the length of the original membrane keyboard connectors - that could be tight.

fruit-bat, 2023-10-27:
Another lovely looking board!

I think this has the kemptson joystick mode enabled by default...

[ZxSpectrumPicomputerVgaAuk.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13193409/ZxSpectrumPicomputerVgaAuk.uf2.zip)

bobricius, 2023-10-27:
I didn't measure the lengths of the membrane cables, I don't know if there are different versions, for example, mine is 8 cm, the connectors on the board are approximately in the same place as on the old version. in the case of assembly in the original box, the arrows and menu on the motherboard probably won't be usable, I think they are at the interface of the rubber keyboard and the flat cover, but since you have already solved the control very elegantly and when the joystick works, I won't have to use it, just in case standalone use. the sample is already ready and will leave Cina.
I didn't put the connector for the LCD at the moment, I think there are already many LCD versions...
etc ...

[image]

digitalmexer, 2023-10-28:
@fruit-bat 
Thank you, the Kempston joystick is indeed active.
Unfortunately, the direction assignment is reversed and the old menu call with the joystick down button also disturbs. 
I suspect it is due to the faulty circuit diagram, there the pico pin assignment does not match the circuit board.

[image]
This is shown by the schematic (wrong)

[image]
This corresponds to the PCB (correct)

The test showed the following effect:
Joystick down -  effect: menu call
Joystick left    -  effect: movement up
Joystick right  -  effect: movement down
Joystick up     -  effect: movement right

This corresponds exactly to the swapping of the COL numbers

The change in menu item 6 to Sinclair joystick switches the Kempston off but not the Sinclair on.

digitalmexer, 2023-10-28:
@bobricius 
I wasn't really thinking: the cable length of the ribbon cable will definitely fit. In my construction, the cable is connected from the bottom because there was not enough space for the connectors at the top. This meant that I had to extend the cables - this is not necessary with your new design.

I like your tiny PICOZX in V-design very much. Is it possible to order the boards for it at Tindie?

bobricius, 2023-10-28:
@digitalmexer 
next week I get little fixed board, I send you email with details

fruit-bat, 2023-10-28:
@digitalmexer 
I'm super tired so not sure I got this right. I've adjusted the joystick pins, let me know if it helps.

@digitalmexer  &  @bobricius
Peter, I think this put the layout at odds with your design, but I am a little confused over where the original mistake was.... so maybe it doesn't? Possibly also causes issues for Arnold's board. May have to split into 2 build targets if this is the case.

[ZxSpectrumPicomputerVgaAuk.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13196743/ZxSpectrumPicomputerVgaAuk.uf2.zip)

bobricius, 2023-10-28:
I will wait for PCB boards then we can see more, I draw this new board by Arnold schematic, I think Arnold forgot change signals on schematic on 9pin joystick connector.

on current release PCB PICOZX LCD I just add foil connectors for keyboard I not count with use internal arrows o DSUB9 because they use different pins.

I probably made it unnecessarily complicated, I didn't want to create another HW variant, I thought I'd only use Arnold's firmware, old board with the fact that no one will use arrows and dsub9

digitalmexer, 2023-10-29:
First of all, I would like to express my respect for the ingenious programmer who has so successfully programmed the desired adjustments here. Without available test hardware, his work is a feat more difficult than playing chess blind.

I am sorry if I have caused any confusion. 

The circuit of all previous boards (ZXPICO-VGA, ZXPICO-LCD and the board with the adaptation for foil keyboard ) is always the same as the new design of Bobricius.

The drawing of the schematic has always been incorrect for the joystick and therefore corrected. 
The firmware for the previous versions of the PICOZX boards has always worked well. 

The small errors that I have noticed concern all versions and do not need to be clarified here now.

The extension of the menu operation with the Spectrum cursor keys works well and CAN be made usable for all previous PCB versions ADDITIONALLY - but this does not necessarily have to be. 
But it is necessary for the new PCB version.
I think that the firmware for the LCD/VGA version of the PICOZX must be extended only by the operability with the spectrum cursor keys and the changed menu call to reach all problems, maximum possible usability and compatibility.
The change for the Membrane keyboard- firmware with the menu call button between ROW1 and COL5 is a bad solution because it deviates from the previous design and is not compatible. This must be removed from the firmware. With the new board design, nobody needs this anymore anyway.

Maybe it is a good approach to build on the working software of the LCD/VGA version and add there only the extension of the menu call and the navigation with the ZX cursor keys. This program change is already successfully tested.

digitalmexer, 2023-10-29:
@fruit-bat 

the new firmware version because of the joystick problem did not bring any improvement:

The test showed the following effect:
Joystick down - effect: movement up
Joystick left - effect: movement down
Joystick right - effect: movement left
Joystick fire - effect: no effect

fruit-bat, 2023-11-01:
> @fruit-bat
> 
> the new firmware version because of the joystick problem did not bring any improvement:
> 
> The test showed the following effect: Joystick down - effect: movement up Joystick left - effect: movement down Joystick right - effect: movement left Joystick up - effect: fire Joystick fire - effect: no effect

Yep, though I was not quite awake at the time....

[ZxSpectrumPicomputerVgaAuk.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13226746/ZxSpectrumPicomputerVgaAuk.uf2.zip)

hopefully, this one is at least a bit better!

If you and Peter can end up using the same key layouts long term that would be great but happy to see if we can get your existing hardware working (even if we don't end up giving it an official 'target').

digitalmexer, 2023-11-02:
Thank you very much, the joystick now does what it should.

digitalmexer, 2023-11-08:
I apologize for any inconvenience.
Unfortunately, I got some things mixed up in the version by Bobricius and ArnoldUK. 
I've only now realized that.
@fruitbat:
Thank you for your patience and the customized firmware especially for me.
I think the second last software version should be the best fit for the ArnoldUK circuit. 
I will adapt my construction to his circuit diagram and report back.

digitalmexer, 2023-11-09:
My hardware now corresponds exactly to the circuit diagram from ArnoldUK.

I'm not sure if Arnold would prefer to customize the software themselves.
The fourth firmware version of fruitbat (dated 10/27/2023 11:25 PM GMT+2) in this ISSUES matches the Arnold hardware almost perfectly.
Only the fire button of the DB9 joystick does not work in the Spectrum programs.
The USB joystick works perfectly. The menu and 48K/128K toggle buttons also work as they should.
Menu navigation with the ZX cursor keys is also possible without any problems.
This means that all the features planned by ArnoldUK have been implemented.

fruit-bat, 2023-11-19:
@digitalmexer  I'm not sure why the fire button is not working; I've had a read through the code and can't see the problem. I'm making a target for @bobricius  new board so I may end up working out what is wrong as I do that. I think @bobricius is going to send me a board which will make sorting these issues out a lot easier! 

See https://github.com/fruit-bat/pico-zxspectrum/issues/103

closed 2023-11-23 by digitalmexer