fruit-bat/pico-zxspectrum #101: PicoMiteVGA-To Enable STEM Project

waterfallm, 2023-10-28:
I'm looking at the PicoMiteVGA board to make a STEM project with a museum i volenteer at.  Would be get to have a config that we coulpd switch between to allow a bit of fun after a session doing some Basic programming.

waterfallm, 2023-11-19:
yeap i have the circuit here.

[14A Circuit diagram.pdf](https://github.com/fruit-bat/pico-zxspectrum/files/13404512/14A.Circuit.diagram.pdf)

waterfallm, 2023-11-19:
i'll get the toolk chain sorted and satrt to play with the baord def and build

fruit-bat, 2023-11-19:
I've had a look at the circuit diagram and most things are there, including a ps/2 keyboard and sound channels 😀. 

The spectrum has a set of bright colours that will end up being extra green. 

Might need some extra code to read the joystick pins.

fruit-bat, 2023-11-19:
I've added an empty branch for changes to go into. If you work in this branch then it's easier to collaborate without disturbing the main code base. I'll have a think about which is the best existing target to start working from.

fruit-bat, 2023-11-21:
Very early attempt at changes here:

https://github.com/fruit-bat/pico-zxspectrum/pull/105/files

Binary here (you need to unzip it):-

[ZxSpectrumPicoMiteVGA.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13431052/ZxSpectrumPicoMiteVGA.uf2.zip)

Obviously, virtually no chance it will work but give it a go :-)

The new target is 'ZxSpectrumPicoMiteVGA'

waterfallm, 2023-11-22:
Great start to help me understand how to get going. 

As you predicted no joy :) no VGA output.

How can i go about debuging?

fruit-bat, 2023-11-22:
In theory, the serial port will be talking on gp0 & gp1. Those get brought out to one of the db9 connectors. You would need a 3.3v ttl serial to usb adapter to monitor it.

If you have a scope, might be worth seeing if any of the vga lines are active.

I think I sent the c-sync signal to v-sync instead of h-sync by mistake. I will swap that over when I get a moment and put another build up.

waterfallm, 2023-11-22:
Thank for all your help so far.. Once i get the toolchain setup properly i can actually start to itteriate through the config changes myself.

Do you have a git_action that scripts out the build process?

waterfallm, 2023-11-22:
so the only debug output i'm getting is

"Core 1 Running"
[image]

fruit-bat, 2023-11-22:
Well, that's actually rather promising... it says 'panic' when something bad has happened!

Here is a build with c-sync sent to the h-sync line:-

[ZxSpectrumPicoMiteVGA.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13444512/ZxSpectrumPicoMiteVGA.uf2.zip)

No git actions I'm afraid. If you are a wiz at setting them up, advice would be appreciated; I just develop and build on a Pi 4.

waterfallm, 2023-11-23:
Yeap I think a git_action can be my gift to the project.

waterfallm, 2023-11-23:
and we have progress. Best i finish soldering up the audio,sd card and keyboard :)

[image]

fruit-bat, 2023-11-23:
> Yeap I think a git_action can be my gift to the project.

That would be very much appreciated 🙂

fruit-bat, 2023-11-23:
> and we have progress. Best i finish soldering up the audio,sd card and keyboard :)
> 
> [image]

The RGBY is in the wrong order and sorting it needs a little thought... but in the meantime at least you can see what is going on. 

If you have an otg adapter you should be able to connect a USB keyboard and/or joysticks (the ps/2 keyboard should work too).

fruit-bat, 2023-11-23:
This has the VGA colours output BGYR, which may be a bit better; afraid it will never be perfect without a different VGA circuit. If this is terrible, we could try BYGR instead.

[ZxSpectrumPicoMiteVGA.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13453568/ZxSpectrumPicoMiteVGA.uf2.zip)

waterfallm, 2023-11-23:
> If you have an otg adapter you should be able to connect a USB keyboard and/or joysticks (the ps/2 keyboard should work too).
I'm a couple of transistors from getting PS2 keyboard running and a 2r2 resistor for SD Card but they arrive tomorrow so should be up and running then.

RGYR gives me much better in the menu but magenta background. I've got 10 boards now to make up for the STEM elelment and the colors are ok for that the main need is to get MM Basic running with some fun maybe getting the Spectrum up and running.