fruit-bat/pico-zxspectrum #102: LED

Suggy50, 2023-10-29:
Not an issue as such but is there any chance you can make the LED on the pico light up when power is supplied to the PICO?

fruit-bat, 2023-11-01:
On which target (On some of them it is used to indicate if the board is in cursor/joystick mode)?

Suggy50, 2023-11-01:
The board designed by BOBRICIUS, PEMI.TECHNOLOGY
Currently using 
ZxSpectrumPicomputerVga222Zx.uf2
and
ZxSpectrumPicomputerZxLcd.uf2
Firmware

fruit-bat, 2023-11-02:
In the attached builds, the LED is  on for cursor mode (the default), and so lights up when the board is powered on. Let me know how you get on with them.

@bobricius are you happy with this? I could leave the LED as on for joystick mode and change the power on default?

[ZxSpectrumPicomputer.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13242971/ZxSpectrumPicomputer.uf2.zip)

bobricius, 2023-11-02:
Thanks, very nice, I think it can be very useful for new incoming AndyUK rubber keyboard PCB version because pico is again located to top side and led is better visible.

Suggy50, 2023-11-04:
@fruit-bat fantastic, green light on the pico. If I have trouble in the future I know power is getting to the board and the issue is else were

Suggy50, 2023-11-04:
@bobricius Very nice board but perhaps a small amplifier circuit for volume, im hard off earing and find id difficult with the volume on some games

Suggy50, 2023-11-12:
@bobricius Do you have a part number for the speaker socket? I got my board ready built from Stephen Stuttard, and been hard off hearing the piezo speaker is too hard for me to here, all the caps, resistors, BSS123 & BSS84 are on order and I just need the socket to complete it

bobricius, 2023-11-12:
> @bobricius Do you have a part number for the speaker socket? I got my board ready built from Stephen Stuttard, and been hard off hearing the piezo speaker is too hard for me to here, all the caps, resistors, BSS123 & BSS84 are on order and I just need the socket to complete it

I lost model, checking now, it disappear from my distributor, you can use any speaker

Suggy50, 2023-11-12:
> I lost model, checking now, it disappear from my distributor, you can use any speaker

Thanks for getting back to me
Its the speaker socket im after the one that goes here
[image]

bobricius, 2023-11-14:
@Suggy50 I am use this https://www.tme.eu/sk/details/1503-20/konektory-jack/lumberg/1503-20/

Suggy50, 2023-11-16:
> @Suggy50 I am use this https://www.tme.eu/sk/details/1503-20/konektory-jack/lumberg/1503-20/

Thanks Peter, have ordered some