fruit-bat/pico-zxspectrum #103: Kemston joystick not working in AUK firmware

bobricius, 2023-11-12:
[image]
[image]
[image]

Hi Phil,

Thanks for this amazing project,
I redraw AUK board version to fit in to original ZX spectrum, onboard button matrix have same wiring as original membrane keyboard, also switches spacing correspond to rubber buttons.
On board are also connectors for foil keyboard.

Dsub9 is connected to onboard arrows.

Arrows works but not emulate Kempston joystick. Joystick works on USB
Also I prefer modify shortcut 48K to Previous game, 128K to Next game. Shit + Menu to QSAVE1, Shift+Prev/ Next 
Cursor/Kempston joystick

Maybe you can glue New Rubber Navigation version + LED mode

I am ready to send assembled board.

Thanks very much.

Peter

fruit-bat, 2023-11-19:
Peter,

I've made a start on the changes for your new board....

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13404474/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

...I don't know it if works but it's something to try :-)

Glad you are still enjoying the project and making boards for it. I have had a lot of fun working on them.

Many thanks,

Phil

bobricius, 2023-11-19:
I think all works,
I send you board from new revision, fixed some dimensions details.
By my quick test.

prev/next game works
on/off menu button works
by default is kempston jojstick switchable to cursor
led indicator also works

digitalmexer, 2023-12-09:
I had suggested in Issue 100 to call up the menu with [CAPS-SHIFT] + [SPACE] - I don't think that was a good idea, because you can no longer interrupt a BASIC program with BREAK. Bobricius has made the menu button available on the back of the 48K case in his new board version, I'm sure that's a better solution than using the standard Spectrum buttons. What do you think?

fruit-bat, 2023-12-11:
> I had suggested in Issue 100 to call up the menu with [CAPS-SHIFT] + [SPACE] - I don't think that was a good idea, because you can no longer interrupt a BASIC program with BREAK. Bobricius has made [...]

Yup, best to use another key if you have one...

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13641287/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

digitalmexer, 2023-12-12:
Thank you I have tested it. 
The BASIC program abort with SHIFT + SPACE (Break) now works as it should. 
But however if the cursor joystick mode is activated, the menu is still called up with SHIFT+SPACE .
By the way:
The fire key does not work correctly with the cursor joystick (but this has always been the case) - the fire key brings up "ENTER", not "0"

fruit-bat, 2023-12-14:
This may sort:
> the cursor joystick mode is activated, the menu is still called up with SHIFT+SPACE 

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13672012/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

When the joystick is mimicking the cursor keys I don't think there is a well defined mapping for 'Fire'. The idea was that it would at least allow navigation of the 128k boot menu. Which key did you expect it to press?

digitalmexer, 2023-12-14:
I meant the cursor joystick mode, which can be selected in the game menus.
Source:
[(https://worldofspectrum.org/faq/reference/peripherals.htm)]

**Joystick Interfaces**
**Cursor**
A cursor joystick interfaces maps to keys 5 (left), 6 (down), 7 (up), 8 (right) and 0 (fire). Reading a cursor joystick thus requires a combination of bit 4 of port 0xf7fe and bits 0, 2, 3 and 4 of port 0xeffe. Common interfaces offering a cursor joystick option included those produced by Protek and AGF.

I therefore assumed that the key code for the "0" key is transmitted to the ZX when the fire button is pressed.

There are some ZX programs that support multiple players. In this case, selecting the cursor joystick mode on the DB9 port would be a useful addition to the USB joystick and keyboard.

fruit-bat, 2023-12-14:
Ok, does this seem better to you....

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13673162/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

fruit-bat, 2023-12-14:
Ok, does this seem better to you...

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13673182/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

digitalmexer, 2023-12-14:
Thank you very much - it works perfectly.
I think with this change the PICOZX behaves like the original ZX.

fruit-bat, 2023-12-24:
@bobricius here is the latest firmware (just in case you haven't spotted it on the other issue)...

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13761902/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

Merry Christmas!