fruit-bat/pico-zxspectrum #106: Sinclair Joystick Type

digitalmexer, 2023-12-09:
How can I select the Sinclair joystick type left/right?
I can only work with Sinclair-L so far.

fruit-bat, 2023-12-11:
It puts the second USB joystick  on the right. Maybe it needs a bit more control than that?

digitalmexer, 2023-12-12:
The possibilities offered by the current firmware for the pico-zxspectrum are ingenious. It already offers a wide range of functions. The desire to make the Sinclair joystick type L/R selectable is pure luxury. I still think it would be nice to be able to select both Sinclair types and the Kempston joystick via the joystick menu (menu item 6).
> 1. Kempston
    2. Sinclair L
    3. Sinclair R
Would this be possible with reasonable effort?

fruit-bat, 2023-12-14:
I will have a read through the code and remind myself how this works :-)

fruit-bat, 2023-12-18:
Please let me know if this helps...

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13704654/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

digitalmexer, 2023-12-18:
Well done, it looks very good. 
Unfortunately I only have one USB joystick available at the moment and couldn't test whether the left and right Sinclair joystick work at the same time. 
Everything works perfectly with just one USB joystick.

digitalmexer, 2023-12-23:
Now I was finally able to test the behavior of two Sinclair joysticks connected at the same time:
In principle, everything works. However, there is an anomaly when comparing the results of the IN commands in ZX-Basic with the INKEY commands.
The key code obtained with INKEY is always correct for both joysticks. 
But the IN command only gives the expected result for the first joystick, for the second there are no low bits in the data.
I have written a bad Basic program for the test. If you now also have functioning test hardware, you can try it out.

[KeyTest.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13759937/KeyTest.zip)

fruit-bat, 2023-12-24:
Many thanks for the test program, that made it so much easier so (hopefully) sort out.

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/13761898/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

Merry Christmas!

closed 2024-01-01 by fruit-bat