fruit-bat/pico-zxspectrum #107: Pre-build question: Can audio input work with Pimaroni?

EDKGames1, 2023-12-14:
I saw that tape input is supported on the page. I would like to use a Pimaroni video out board, but I want to be able to use tapes. Will it read/write through the audio jack on the VGA/HDMI interface board?
Thanks in advance.

fruit-bat, 2023-12-14:
Both the Pimoroni  VGA and HDMI builds output audio through the on board DAC. This is because the video library is for DVI not HDMI. 

I have seen an example of HDMI output on the Pico (https://github.com/shuichitakano/pico_lib), which is rather impressive.  With HDMI output, in theory, the audio can be sent to the monitor. 

There isn't audio input capability on either of the Pimoroni boards (https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf). If there was you would still need a pre-amp for it.  There are a couple of examples on the Readme of this repository.  

If there is demand for it maybe @bobricius  would consider adding it to a future PCB design? It's kind of fun but there isn't really a facility to 'save' at the moment.  For save  to work the 'buzzer' audio output would need to be less approximate!

EDKGames1, 2023-12-14:
Ok. Is there any way anyone else has added that in, along side the Pimoroni?

fruit-bat, 2024-04-02:
It's been a while,  but there has been some progress.  There is now firmware for the Demo DV that sends audio over HDMI. I think gpio0/gpio1 are free to use as an input (you still need a pre amp)... It would need a slightly different build, so let me know if you are still interested.

EDKGames1, 2024-04-17:
Thanks for the update after so long. I have a lot of projects so this one
is on the backburner right now, but if I get time and the urge I'll check
it out. Thanks Again.