fruit-bat/pico-zxspectrum #112: Problem loadung games

DMNSonic, 2024-01-23:
Hi, newbie here.

I built one using ZxSpectrumPicomputerVga222Zx firmware, just wire VGA amd SDcard amd using USB Keyboard, I didn't build the keyboard matrix.

I try to load some games and it just don't start up. It loads the game, amd some of them reach the option screen but when it starts the emulator just freezes or restart.

I tested Robocop, Knight Lore and Jet PAC, all of them load and freeze afterwards. I used tzx format.

Is something I did wrong? I need to build the keyboard matrix?

Thanks in advance

fruit-bat, 2024-01-23:
I don't think it matters if the keyboard is wired up. Maybe try resetting as a 48k Spectrum in the menus before loading from tape? Also,  probably best not to flip in and out of the menus while it's loading.

fruit-bat, 2024-01-23:
Make sure you have not accidentally connected any of the keyboard rows or columns to vcc/gnd, as the emulator will see this a pressed keys. Can you type normally into the Spectrum (this is different to the menus)?

DMNSonic, 2024-01-24:
Thanks Fruit-bat for your tips. It worked! I never played with a ZX before so I didn't understand about the 48k and 128k differences. I found how to write " on it, just use AltGr+P and loaded a bunch of games and played.

I found some situations to work on:
With my USB hub it doesn't work the keyboard + USB joystick, it's one or another. I'll check if it's a problem with my hub or Amp power.
Hybrid games (they are both 48k and 128k) if you try to load on 128k mode it just hang up. Exclusive games maybe work, tried Midnight Resistance and worked like a charm.

And just one more question, is it possible to emulate Kempston joystick on keyboard? Some key choices on games made me crazy to play.

And again thanks for the help! Amazing project you made!

fruit-bat, 2024-01-26:
Having both keyboard and joystick connected *should* work; let me know if it continues to be a problem.

The tzx format is quite involved and I may not have implemented all of it correctly/fully. If you can let me know the location of specific examples that do not work it *may* help me fix it.

What key mappings would you like for Kemptson joystick?

DMNSonic, 2024-01-27:
After a lot of tests and different PSUs I found the is the USB Hub. Tested two here and one doesn't work and the other causes this problem that I've mentioned before.

About the tzx format I tried and had luck with many others after that, so I believe that's a problem with the older files.

Z80 snapshots worked very well and I already made a list of games booting from it, it helps a lot to not wait loading tzx files. 

The key mappings in many games you can change the key, which is something very useful but some just use the strange AQOP combination that it's impossible to play some games with precision.

The Sinclair Joystick sometimes maps controls on arrow keys.

Maybe, in a future update, if possible of course, have the ability to map Kempston keys on Arrow Keys and use an unused key for action button. But it's just a suggestion, maybe it cannot be possible or not feasible to do.

And again thanks for this marvelous project. I finished building one integrated on a keyboard and played a lot!

DMNSonic, 2024-01-27:
Update: I'm gonna build a DB9 to use Atari-like joysticks but the Row and Columns combination just don't help that much.

I saw many games use "Cursor" keys to play, it uses 6, 7, 8, 9 and 0 to play but 6 and 7 is on 2 column combo and 8, 9 and 0 on the 3 column combo.

If I try to use the cursor keys on keyboard not all games accept and the 0 key I can't map to a pin on the DB9.

Reading the code I saw there's a way to activate Kempston mode, how to do it? And what keys it uses?

fruit-bat, 2024-01-31:
I think this is the one you are working to...

[image]

(Bobricius, I hope you don't mind me posting this)

fruit-bat, 2024-01-31:
There are already some mapping for a db9. Hopefully this helps.

DMNSonic, 2024-02-01:
Again many thanks for the help! I'll gonna make a DB9 based on this schematic because I had no luck with USB hubs.