fruit-bat/pico-zxspectrum #114: Help for Keyboard Matrix pinout for support full 48+ keyboard

emilen72, 2024-02-02:
Hi, I have a spare empty 48+ case (no pcb and no membrane) and want to fit it with rp2040 (etc etc) and a custom made pcb with a tactile switch for each of 58 key of the 48+.

I can't understand how the matrix for this configuration and if exist a ready to use uf2.

Many thanks for your help and a big thanks for this wonderful emulator.

fruit-bat, 2024-02-02:
Hi there, glad you are enjoying the project.

If you look in this thread https://github.com/fruit-bat/pico-zxspectrum/issues/112 there is a circuit diagram towards the end that matches the target below.

[ZxSpectrumPicomputerVgaAukBob.uf2](https://github.com/fruit-bat/pico-zxspectrum/blob/main/uf2/ZxSpectrumPicomputerVgaAukBob.uf2)

There are a bunch of other possible layouts, but this one matches that of the original spectrum.

fruit-bat, 2024-02-03:
I've updated the previous comment. Sorry for the miss direction.

fruit-bat, 2024-02-03:
I've started a page with the stuff I think you need. I will split out all of the firmware targets onto separate pages.... I didn't expect there to be so many versions when I started the documentation!

https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/pico_zx48_128.md

emilen72, 2024-02-03:
Many thanks for your reply… and sorry if ask again some stupid questions but I’m still a bit confused :-(

in the schematic the matrix is 7x7 but the zx+ and 128 have 56 keys so it need 7x8 at least or I’m wrong?
In the schematics there are some keys in row1 and row7 that no exist in real zx, zx+ and 128
It mean that not all keys activated by the original zx+ or 128 lower membrane (invvideo, truevideo, caps-lock, breack etc etc) are supported or not needed by emulator?
watching some videos I understand that the additional keys on row1 and row7 have special function required by the emulator and they need to be connected to dedicatd key (as on bobricius pcb) but they are mandatory or not? Or Can I omit them for cabling all original keyboard keys?

In other word can the emulator to be compiled to support a base configuration of  all original 56 keys, vga, audio out and cassette in? Even taking away other feature as SD and joystick?

fruit-bat, 2024-02-03:
Feel free to ask questions (sometimes I have more time to answer than others):

- I *think* the 128 and 48k Spectrum had the same keyboard connections but some buttons hit combinations of rows and columns.
- Row1 of this hardware layout is used to support a few 'extra' buttons and a DB9 joystick port
- The emulator *should* run with no extra components added to the Pico
- If you want to access the menus you will need to wire up the menu button (or we need to decide on a set of membrane presses, which is usually harder than you might imagine)

See https://zxrenew.co.uk/ZX-Spectrum-48k-128k-Toastrack-keyboard-membrane-p69999635

With regards to:
>   In other word can the emulator to be compiled to support a base configuration of all original 56 keys, vga, audio out and cassette in? Even taking away other feature as SD and joystick?

Yes, but it's not trivial. Each target has at least some custom code to make sure everything is in the right place. Happy to help out with this sort of thing but need to agree exactly what is going where first... and the Pico only has a certain number of pins and some of them are tied to specific functions.

fruit-bat, 2024-02-03:
If you go with this VGA circuit you get some extra pins (at the expense of some extra diodes)...

https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/Pico%20VGA%20RGBY1111.svg

DPRCZ, 2024-02-11:
I've just test zx spectrum plus keyboard with my board based on ZxSpectrumPicomputerVgaAuk/ZxSpectrumPicomputerVgaAukBob and keyboard works great.

emilen72, 2024-08-14:
very late reply...

I would like to assemble on breadboard for testing  and then make a new PCB with these characteristics:

- real 48k+ keyboard
- db9 joystick
- audio out
- audio in for load real cassette
- vga rgb1111
- sd card
- usb host

what firmware can I use? 

many thanks

fruit-bat, 2024-08-14:
There is not a configuration for that organisation at the moment... but it sounds possible. Hopefully, just by creating additions to the 'make' system.... but at the moment I am 'trying' to fix the VGA builds which have broken since the SDK update for the RP2350.

fruit-bat, 2024-08-17:
@emilen72 VGA builds are working again :-) I can try and make a build based on 
https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/pico_zx48_128.md but use a 1111 VGA encoding. Then I will see if we can add the audio in on a pin freed up from the 222 VGA, probably GPIO 6

Does that sound like what you are after?

emilen72, 2024-08-17:
good to hear your progress and yes that is what I need.

tnx

🙂

fruit-bat, 2024-08-17:
Here you go:

[ZxSpectrumPicomputerVga1111AukBob_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/16645647/ZxSpectrumPicomputerVga1111AukBob_720x576x50Hz.uf2.zip)

I've not added the audio in yet. Let me know if this works and we will go from there. VGA is on GPIO 0-4 with (CSYNC on CPIO 4).

emilen72, 2024-09-08:
hello, sorry for late reply .... firmware works... only issue is that menu color letter is green instead of white. see photos
[image]
[image]
[image]

emilen72, 2024-09-11:
Hi, many thanks new firmware is ok, see photos (the text is actually white)

[image]
[image]

others ideas

1)  Would it be possible to have a 15khz 720x576 50hz output? because I would like to have a SCART output or be able to generate a CVBS (as in original zx spectrum) via an encoder as AD724.

2) would it be possible to make a build for the wide used purple pico clone with 16MB on board? this would be enough memory for many user and would save the use of an SD and have more gpio pin than standard pico for other addiction...

I hope these ideas not hurt you :-)

fruit-bat, 2024-09-14:
A1. Probably possible, but tricky, as I no longer have anything to test against and there are not any pre-built timings in the VGA library.

A2. I imagine the code will just run on the clone, but using the FLASH memory as a file system has complications. The code for the emulator spills into the FLASH memory so I would need to be careful none of the file-system code had spilled over - and there may just not be enough space in RAM (not sure). Also, my previous attempts messed with the video timings and caused system-freezes. So, yes nice idea, but not easy to do particularly if you want to write to the file-system.

emilen72, 2024-09-17:
ok, thanks for your reply

I'm wait the add of audio input to firmware, many many thanks

fruit-bat, 2024-09-20:
Audio in on GPIO 6...

[ZxSpectrumPicomputerVga1111AukBob_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17076864/ZxSpectrumPicomputerVga1111AukBob_720x576x50Hz.uf2.zip)

Please let me know if it works (at all).

emilen72, 2024-09-21:
can load tape from "tape loader" menu entry or only from basic?

fruit-bat, 2024-09-21:
From basic is fine. The audio will need a pre-amp (won't work directly from line-in/headphones).

emilen72, 2024-09-21:
wave in the photo the is on gpis pin 6, I used now the mormulator audio in
pre-amp circuit, but with "taper loader" no sign of life. Then I try with
basic.

emilen72, 2024-09-21:
[image]
[image]
[image]

emilen72, 2024-09-21:
[image]
[image]
[image]
even with lm358n no sign of read on gpio6

fruit-bat, 2024-09-21:
Nice clean signal.... I take it runs from 0v to 2v, not +/- 1v? Does the border flash different colours at all?

Most likely a software issue... I will have a read through.

fruit-bat, 2024-09-21:
Give this one a go...

[ZxSpectrumPicomputerVga1111AukBob_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17084802/ZxSpectrumPicomputerVga1111AukBob_720x576x50Hz.uf2.zip)

emilen72, 2024-09-21:
with the latest firmware the game load :-) even the sound from speaker came out during load (with previous firmware no border color at all and no sound)
I'm happy :-) many many thanks
[image]
[image]
[image]

fruit-bat, 2024-09-21:
Very glad it worked!

If you are happy to post any circuit diagrams, hints/tips, I will add a page for your configuration along with the a download for the firmware.

fruit-bat, 2024-09-21:
Thanks for the pics; lovely to see it in action!

emilen72, 2024-09-21:
it work  even with murmulator pre-amp circuit and two simply BC547 transistor, only have to up a bit the sound level of source (iPod).
[image]

[image]

emilen72, 2024-09-21:
> Very glad it worked!
> 
> If you are happy to post any circuit diagrams, hints/tips, I will add a page for your configuration along with the a download for the firmware.

Sure, I'm happy, I will post the circuit even with a simply but useful amp for a real speaker 8ohm 1W...