fruit-bat/pico-zxspectrum #115: mirrored text on lcd

aadhyatmik, 2024-02-03:
Great project, kudos to you, when I tried to assemble on a zero pcb to test, I get mirrored text on lcd, using zxspectrumpicomputerzxlcd uf2 file gp2-gp6 for ili9341 lcd and gp19-gp13 for sdmicro. Someone please suggest a fix.
[image]

fruit-bat, 2024-02-03:
That will be because the driver is for a st7789. 

A tweak to the MADCTL (Memory Access Control 0x36) register will probably sort the rotation. Possibly 0x47 or 0x87.

Fixing the colours may be more involved... but interesting to try!

[ILI9341_DS_V1.10_20110415.pdf](https://github.com/fruit-bat/pico-zxspectrum/files/14152593/ILI9341_DS_V1.10_20110415.pdf)

See: https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/picomputer/picomputer_st7789/st7789_lcd.cpp
 
You in a position to build the firmware?

fruit-bat, 2024-02-03:
Looks like blue and red are swapped.  That what you think?

aadhyatmik, 2024-02-03:
Thanks for your reply, I have never built, however i have picosdk installed, will surely try, kindly guide me stepwise.

I think I need to replace tinyusb then tweak madctl value ( kindly suggest filename), then execute cmake command in build folder then execute make command.

fruit-bat, 2024-02-03:
Yup, I can have a go for you if you like... give me a few minutes....

aadhyatmik, 2024-02-03:
Unfortunately I am not good at recognising colours, it seems a washed out to me. Will see once it's working.

fruit-bat, 2024-02-03:
This is 0x10, so we know for the next try.

[ZxSpectrumPicomputerZxLcd.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14152725/ZxSpectrumPicomputerZxLcd.uf2.zip)

NB you need to unzip the uf2

aadhyatmik, 2024-02-03:
Unfortunately that makes it behave more erratic.

[image]

aadhyatmik, 2024-02-03:
X30 seems good, attached picture for reference, colours also look right to me ;-)

[image]

aadhyatmik, 2024-02-03:
Thank you very much, I need some more time to test usb keyboard etc. I am preparing this for my daughter, she love programming. 🙏

fruit-bat, 2024-02-03:
Colours look inverted (or wrong in some other way_ to me...

What does this one look like?

[ZxSpectrumPicomputerZxLcd.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14154200/ZxSpectrumPicomputerZxLcd.uf2.zip)

aadhyatmik, 2024-02-03:
X30 looked better to me, stripes on logo in front of this are blue cyan green yellow, previous one (X30) contains yellow red pink blue.

[image]

aadhyatmik, 2024-02-03:
Kindly tell me when you are done, will close it.

fruit-bat, 2024-02-03:
This swaps the red and blue components...

[ZxSpectrumPicomputerZxLcd.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14154360/ZxSpectrumPicomputerZxLcd.uf2.zip)

Hopefully,  that should do it :-)

aadhyatmik, 2024-02-04:
Sorry for delay, this one I think matches with the logo, colours are red, yellow, green and cyan.

[image]

aadhyatmik, 2024-02-04:
Here is the picture
[image]

aadhyatmik, 2024-02-04:
On testing keyboard it is working but I was not able to type " or : and few other keys , tried 2 different keyboards, one dell wired keyboard and one portronics hydra in wired configuration, it seems a layout issue may be. Seems need to read manual.

fruit-bat, 2024-02-04:
The keys on the USB keyboard map 'literally' to the spectrum keys. Alt-gr (on the right) is symbol shift.

aadhyatmik, 2024-02-04:
Thanks will try alt-gr tonight, never owned a spectrum or any 8 bit computer, so missed that. 🙏