fruit-bat/pico-zxspectrum #117: I have a PicoZXpi board that I am sure is based on this firmware and Ringo does not work correctly

midicdj1000, 2024-02-07:
HI,

I am currently testing a few SOC spectrum boards and have the PicoZXpi board from a uk seller and tried to find new games that pushed the spectrum hard and Ringo is one of those new games. It loads ok, but at the base 3.5 MHz speed it crashed once past the intro screen. Increasing the speed to 4 MHz or unregualted and the game will run, but looks totally wrong.
I am using VGA output so I guess the problem is related to timing.

[image]

fruit-bat, 2024-02-07:
Yes, quite possibly a timing issue. There isn't spare memory for a frame buffer so I just go with the refresh rate of the physical display. Games that rely on a precise number of cpu cycles per frame are going to suffer. 

Having said that, Ringo looks pretty crazy when it is working (picture from the Ringo website)...
[image]

midicdj1000, 2024-02-08:
The Pico board is the only one this game does not work on and even the espectrum was ok. The pico is my favorite setup overall so it would be nice to make it better.

fruit-bat, 2024-02-13:
I bought a copy of Ringo and have been having a play with it. I don't think I am going to be able to get the display working correctly. I *think* the devs have been super clever and chase the raster with the CPU updating colour attributes as the screen is drawn. Because VGA runs at 60Hz (not 50Hz) I would need a frame buffer to construct the image correctly and I don't have enough RAM left... plus it's difficult!

Shame, cos it looks like a fun game.

midicdj1000, 2024-02-14:
I am working on my compare video and I think the ESPectrum version also has some audio issues so the only one that worked 100% is the ZXZERO with fuse on the PI, but that has a 8+3 filesystem and it impossible to browse the games list.

The PIco is still my favorite just because it supports long filenames correctly and loads the games list so much quicker that the others.

midicdj1000, 2024-02-16:
not a review, but comparing features against the other projects.

fruit-bat, 2024-04-21:
Ringo should now be working on firmware from this site