fruit-bat/pico-zxspectrum #118: Pimoroni DV Screen Size & Refresh Rate

deesto1, 2024-02-09:
Hello there! I want to congratulate you on such fantastic work first of all! I'm in complete awe. I bought a Bobricius board recently and inspired by it, led me to try to make my own stuff but using pre-soldered/built and ready-to-go components such as this. And then this led me back here for the software/FW side of things - and it seems as though everything seems to be accounted for, which is amazing. I have two queries: One of which I think you are already looking into, the possibility to have an option to output audio over HDMI or over the DAC as it is currently. The more pressing question I have though, is whether it is possible to output at 1080p@60/50Hz, rather than 640x480@60Hz? I have no idea how to compile anything, but I'm wondering if it would be easy enough to do this, or if nothing else to ask you if it's even capable of achieving this? Many Thanks for your reply! If it's more convenient to chat about this elsewhere, then how/who?

fruit-bat, 2024-02-10:
Thank you for your kind words and I am glad you are enjoying the project. 

I think audio over HDMI would be a nice addition. There is even an example [here](https://github.com/fruit-bat/pico-infones-hid). This is a fork, so I don't deserve any credit for it. If I, or someone, could strip out and adapt the HDMI driver it would *probably* work. Does depend on how many extra CPU cycles it absorbs. I might add, adapting the code from the NES emulator looks really complicated!

Happy to be proved wrong about this but I don't thing 1080p DVI output is going to be possible. The data clock (TDMS) would need to run at 1,485.0 MHz. The RP2040 is already running at double speed to achieve the 250MHz for 640x480. Might be possible with a VGA output as the dot clock is around 148Mhz.

fruit-bat, 2024-04-02:
Audio over HDMI is now implemented. 

For firmware, see https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/ZxSpectrumPicoDv.md

Let me know if it works ok :-)