fruit-bat/pico-zxspectrum #119: Will the Waveshare RP2040-PiZero be compatible?

RattyDAVE, 2024-02-13:
Found here: https://www.waveshare.com/rp2040-pizero.htm

Looks like all it needs is sound output.

Any ideas if this will run?

Thanks!

fruit-bat, 2024-02-13:
Looks like it could...

https://files.waveshare.com/wiki/RP2040-PiZero/RP2040-PiZero-Schematic.pdf

...but I don't think it matches any existing target. Do you have one?

RattyDAVE, 2024-02-13:
Well I have one on order. Should be here tomorrow.

So looking at the specs at https://www.waveshare.com/wiki/RP2040-PiZero. It seems that the DVI is based on the modification of [Wren6991 PicoDVI](https://github.com/Wren6991/PicoDVI). So that's a positive.

SD card uses the the standard implementation too. 

I am not too sure on the USB (for the keyboard) 

So I was thinking to try the HDMI breadboard images and see what happens!

I am open to other ideas if you think there is a better image out there. Or if you want someone to test then I am happy to do that too!

RattyDAVE, 2024-02-14:
Sounds good to me. More than happy to test. Dont worry about it not working 1st time. I know how development works.

Also if you want to do a "debug program" that may be easier for you. To test all the pin outs etc. That is fine too.

Either way it has not arrived yet. Royal Mail says on the tracing site delivery 13/2/24.

RattyDAVE, 2024-02-15:
Its here! So far plugged in the demo is quite impressive!  

I have found a good diagram - http://cdn.static.spotpear.com/uploads/picture/learn/raspberry-pi/rpi-pico/rp2040-pizero/RP2040-PiZero.pdf

fruit-bat, 2024-02-17:
So, obviously not much chance of this working, but does it do anything?

(you will need to unzip the uf2 to use it)
[ZxSpectrumPiZero.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14320108/ZxSpectrumPiZero.uf2.zip)

PWM audio on GPIO 17

USB keyboard will need OTG adaptor connected to primary USB port

RattyDAVE, 2024-02-17:
It's ALIVE!!!!

[image]

I have tried with a direct from Primary USB C to USB C Keyboard and not working. This might be me though. Should it work this way or do I need an adapter?  Like - https://www.amazon.co.uk/OTG/dp/B09CTSHZR7

Anyway looking good!

fruit-bat, 2024-02-17:
Wow, that's a promising start!

Yes, you need an adapter.

RattyDAVE, 2024-02-17:
It boots quicker than my original Speccy.

Ill get an OTG cable on order!

RattyDAVE, 2024-02-18:
OTG arrived!

Keyboard works!
Put a buzzer (for testing) on 17 that works. (If I have give you a list of pins could you change the output pin?)

SD card test in progress. (I think it maybe me)

Looking very good so far.

fruit-bat, 2024-02-18:
Yes, no problem.

RattyDAVE, 2024-02-18:
I have having some issues with the SD card. So want to check it is not me.

I smallest SD card I have is 8GB.
I have tried a few ways and not having any luck.

1)
Using gdisk 
Deleted all partitions on the SD card.
Created the all partition as MSDOS.
Then tried and the /zxspectum directory was not created.

I then put a .tap and .z80 in the root  the sd card that was still blank.
in the F1 menu these were not seen in tapes and snapshot.

2)
Deleted all partitions on the SD card.
Created a 2GB partition on the SD card.
```sudo mkdosfs -F 16 /dev/sdb1``` where sdb1 is the sd card.

Then tried and the /zxspectum directory was not created.

I then put a .tap and .z80 in the root  the sd card that was still blank.
in the F1 menu these were not seen in tapes and snapshot.

Do you have any guidance?

Re the pins for the buzzer. Are any of the following free?  4, 18, 7, 8, 16, 21, 13, 6, 11, 22, 2
This is to do with the spacing for the GND pins.

Thanks btw!

fruit-bat, 2024-02-18:
It sounds correct. You could try making the folders manually.

| Folder | Contents |
| ------ | -------- |
| zxspectrum/snapshots | Put your .z80 snapshot files in here. |
| zxspectrum/snapshots/quicksaves | Folder for quick saves. |
| zxspectrum/tapes | Folder for .tap and .tzx tape files. |

The first time you visit the snapshots/tapes in the menus you may need to 'refresh', which is button 5 (5=REF).

Also, I will check I assigned the SD card pins correctly.

RattyDAVE, 2024-02-18:
i will make them manually and try again. 

I was also spamming the 5 key. (REF) and that did not do anything.

fruit-bat, 2024-02-18:
Can't see anything obviously wrong. I did have a PIO header for I2S audio included, which I have removed. Probably won't make any difference but let me know...

[ZxSpectrumPiZero.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14323042/ZxSpectrumPiZero.uf2.zip)

RattyDAVE, 2024-02-18:
I have tried the new version and no change.

RattyDAVE, 2024-02-18:
Update:

I have tried another Pico-PiZero and still the same issue.
Looked at https://github.com/fruit-bat/pico-zxspectrum/commit/f797a8763d5a1e8b02f1a027348364f252a7e481 ant the pins seem to be correct. (a different set of eyes sometimes helps!)

I am thinking maybe the way I have done the SD card. I am going to play to see what I can come up with.

fruit-bat, 2024-02-18:
I've just successfully tried a (although not on your version):
 SanDisk Ultra 32GB W95 FAT32 (LBA)
 Lerdisk 4GB W95 FAT32 (LBA)
(of course, the problem may be with my software not the SD card, apologies if that is the case)

RattyDAVE, 2024-02-18:
It was not you, it's me!

I tried a different SD card. Formatted at W95 vfat. Put in the Pico-PiZero, saved the config and then looked on the PC.
Directories created and .config

[image]

[image]

[image]

fruit-bat, 2024-02-18:
Nice! Thanks for the pictures :-)
I'll see if I have time  to look through your pin list for the audio. How many pins do you want to use for the audio out 1,2 or 4? See the main README for notes. Obviously will be limited by the number I find free... and I can't remember if they have to be consecutive.

RattyDAVE, 2024-02-18:
Ideally HDMI audio :-)

1 pin is fine for me. If it is going to cause issues then I can live with it as is.

I am impressed this worked as well as it did on the 1st attempt!

fruit-bat, 2024-02-18:
HDMI audio is really difficult. It's in my list of things to look at but it won't be quick.

fruit-bat, 2024-02-18:
I *think* the following are free:
GPIO4, GPIO8, GPIO13, GPIO11

Any preference for the audio pin out of that lot?

RattyDAVE, 2024-02-18:
For ease of plugging in and to make look tidy.
4 or 13, really have no preference..

fruit-bat, 2024-02-18:
GPIO4
[ZxSpectrumPiZero.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14324364/ZxSpectrumPiZero.uf2.zip)

Please let me know if this works, then I can merge the changes for this board :-)

RattyDAVE, 2024-02-18:
Tested on Manic Miner! (6031769 - That is a number I will never forget!)

Perfect! 

Everything works!

fruit-bat, 2024-02-18:
Nice! Thanks for the all the feedback.

closed 2024-02-18 by fruit-bat

RattyDAVE, 2024-02-18:
No problem it was good to get it working.

For the note it might be worth mentioning:

- Power can be provided by the secondary USB C port.
- A standard PC buzzer can be used for sound (low volume) the pins align with GPIO4 and GND.
- An adapter from Micro HDMI is needed to HDMI.