fruit-bat/pico-zxspectrum #120: Ability to save to .tap file

Panagis1980, 2024-02-15:
Hi there, I am using Bobricius planar in original case, my kid complains that it cannot save its work. Is it planned to create Tape Recorder meny entry like the Tape Player? 
Selecting file, "Motor On" signifying the record button pressed.

fruit-bat, 2024-02-15:
You can save your work with a 'quick save'. The snapshot can be renamed or copied from the menus. 

Saving to tape in a way that doesn't create really big files is a little involved.  I may have a go at some point.

Panagis1980, 2024-02-15:
> Unfortunately, it's not quite that simple. The timing of the audio output
> is approximate. To get the audio input to work I use a PIO processor to
> buffer and synchronise the signal. I imagine the same would be needed for a
> 'save' to work (I've lost track now and I not even sure there is one free).
> I might have a go sometime but the point is it's non-trivial so won't be a
> quick change.

Thank you. Final commend. How can someone save a snapshot without USB
keyboard and only the original one ? Meaning to press AltShift Fx
combination and AltFx to bring it back?
I won't bother you again.

fruit-bat, 2024-02-15:
It's not a bother. Can you tell me which firmware you are using or send a pic of your keyboard.

Panagis1980, 2024-02-15:
PicomputerAukBob.
When the Spectrum Plus is closed. I have only access to Reset and Menu.
I want my daughter to.be able to save her programs even with snapshot.
If we could make a weird combination to simulate the F keys or have a "Save
snapshot" in the menu it would be great

fruit-bat, 2024-02-15:
CAPS and the menu button should do it. Give that a try on something quick and let me know if it works.

Panagis1980, 2024-02-15:
Unfortunately not . on bobricius it goes in and out of the menu rapidly flashing the screen

 

Generaly it is the only thing missing and I believe, since menu can be navigated byt eh real keyboard, it would be easier for you to create menu entry for it instead of playing with keyboard scan codes

fruit-bat, 2024-02-15:
I agree it might be nicer to have a menu option. Also, not sure the menu 'back' button will be very easy with the case close. Maybe CAPS-Enter for back (Enter generally selects things).

Panagis1980, 2024-02-15:
Sure.

I any case, this will give full functionality of machines that have no usb device and only original keyboard and two buttons externally available. 

 

Internally, there are two buttons that are used as F11-F12, completely useless. 128K mode has 48 basic option that works like a charm. 

Cycling through the snapshots currently performed with F9-F10 is redundant. You can just select your image and go.

 

Same with F3. It can be changed from the menu without usb kb.

If you see that row 2 and above should not be tampered more, I can solder the original reset button (that has a cable) on the left side to get a third one externally, and use on from the internal buttons (Row1) along with menu and reset. Choosing from the below and use it as you see fit.

fruit-bat, 2024-02-16:
I've not added the menu option yet but have just been doing a little testing.  CAPS-Menu is working fine for me with this firmware. I do not get rapid switching between the menu and back. Also, CAPS-Enter works for 'back' in the menu system.

You will need to unzip the following to use it:

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14310282/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

I will still look at adding the menu option but thought this may help in the mean time.

Panagis1980, 2024-02-16:
It worked with this uf2!!!. Wow Reloading it can be done now by the menu so we are good!

 

It would be interesting if it could create a new file every time you press it(Slot 2, Slot 3 etc.) 64GB space is more than enough. Cleanup should be user’s responsibility

In any case I can rename it if it is something than needs to be preserved. 😉

 

Thanks again!

fruit-bat, 2024-02-18:
I've added a save option to the snapshots explorer (6=SAV). On the menu SYMBOL is a period (.) and SHIFT-0 is delete.
You can add your own sub folders to the snapshots directory, but you will need to do it on a PC.

Let me know how you get on with it... I think it is much nicer than quick-save followed by copy and rename.

[ZxSpectrumPicomputerVgaAukBob.uf2.zip](https://github.com/fruit-bat/pico-zxspectrum/files/14322843/ZxSpectrumPicomputerVgaAukBob.uf2.zip)

Panagis1980, 2024-02-18:
It worked like a charm.

 

I can save with the desired name and load on demand just with the buttons of the keyboard.

 

Thanks a million.

 

Panagis