fruit-bat/pico-zxspectrum #121: SD Card not detected

Clumpton, 2024-02-16:
Hi, 
Not really a bug but more a request for ideas to debug my build.
Using Bobricius PICOZX motherboard for ZX Spectrum original case.
Firmware latest ZxSpectrumPicomputerVgaAukBob.uf2 from your repo
2 GB µSD card, tried FAT, exFAT, FAT32
SD card is not detected at all, i.e. nothing appears in the tape menus.
I have checked that:
- no short-circuits between the pins of the µSDcard socket
- all socket pins connected to the correct Pico pins

Keyboard, joystick, VGA all working. Basic starts and runs so apparently I have put it together OK.
How can I check if the SD card is being seen by the firmware or not?
I don't have a debugger for the Pico, if I need one please could I have a recommendation?
I do have a small oscilloscope, what could I look for?
Thanks very much in advance

fruit-bat, 2024-02-16:
Try doing a refresh when on the tapes or snapshots menu. It's option 5, say 5=REF on the screen.

Clumpton, 2024-02-16:
> Try doing a refresh when on the tapes or snapshots menu. It's option 5, says 5=REF on the screen.
Thanks for the quick reply!
I see "Scanning folder. . . " flash very quickly. Nothing appears
If I press option 5 13 times in a row it hangs, showing "Scanning folder. . . "

fruit-bat, 2024-02-17:
Not saying these are the only formats that will work, but the ones I am using at the moment are:
2 GB FAT16
2 GB FAT32
4 GB W95 FAT32 (LBA)

Clumpton, 2024-02-17:
I've tried both FAT16 and 32. No dice.
I'd better check my soldering again.

JuniorFilho1001, 2025-02-24:
I also couldn't record on the SDcard. I'm using a 2Gb sdcard, formatted in Fat32 and it still doesn't record anything on it.

fruit-bat, 2025-02-26:
> I also couldn't record on the SDcard. I'm using a 2Gb sdcard, formatted in Fat32 and it still doesn't record anything on it.

It should create the zxspectrum folder on start up. Does this happen?
Have you already created the folder structure and put snapshots or tape files in it?