fruit-bat/pico-zxspectrum #123: Could be used USB joystick/gamepad in the emulator menu?

DPRCZ, 2024-02-20:
USB joystick/gamepad has got usually four direction buttons and four other for fire. As far I know the zx spectrum joystick had only one fire button. Could be used some of the others fire buttons for enter to menu? And moreover, could be used the USB joystick/gamepad for movement in emulator menu then?

fruit-bat, 2024-02-25:
Possibly, I'll have a look at how the code is set up. It will need some translation from joystick state changes to HID keyboard events, which isn't there at the moment.

fruit-bat, 2024-07-19:
I've pushed some changes, which should work now along with configuration in the menus.

Please, let me know if you find anything that needs tidying up.  I think ZxSpectrumHidMouseJoystick.* can be removed but have left them there for now as reference, in case my merging of mouse and joystick isn't quite correct.

DPRCZ, 2024-07-20:
Something missing in git "tinyusb?":
hid_app.c:(.text.tuh_hid_mount_cb+0x11c): undefined reference to `process_mouse_mount'
/usr/lib/gcc/arm-none-eabi/13.2.1/../../../arm-none-eabi/bin/ld: hid_app.c:(.text.tuh_hid_mount_cb+0x202): undefined reference to `process_mouse_mount'

fruit-bat, 2024-07-20:
I'll have a look... what were you trying to build when you got that message?

DPRCZ, 2024-07-20:
I did "git pull mousejoy" and than I tried to rebuild pico-zxspectrum project.

fruit-bat, 2024-07-20:
you need to pull pico-emu-utils and tinyusb

fruit-bat, 2024-07-20:
The examples are broken in TinyUSB at the moment - I'll fix them bit it's not looking simple at the mo.

DPRCZ, 2024-07-20:
Yep, as I already said. Nesu zadne vorezavatko.

fruit-bat, 2024-07-21:
Are you on the feature/mousejoy branch in pico-zxspectrum  ?

If not:

  git checkout feature/mouejoy

fruit-bat, 2024-07-21:
I've merged the mousejoy branch back to main now. So you should be fine just building from there. If you got round to changing to the feature/mousejoy branch you should switch back to main (git checkout main)

You will need to pull:
* pico-emu-utils
* tinyusb
* pico-zxspectrum

DPRCZ, 2024-07-21:
OK,  pico-zxspectrum is compiled. There is only one  issue: pico-zxspectrum/tree/main/src/picovga/main.cpp

static ZxSpectrum zxSpectrum(
  &keyboard1, 
  0, 
  &joystick,
  &mouse,
  0
);

Last 0 in constructor.

fruit-bat, 2024-07-21:
Ah, thanks for that. Should be fixed now.

DPRCZ, 2024-07-21:
What is new in this version? I thought that this version supports the USB joystick in setup menu (for movement and setting), but I probably wrong.

fruit-bat, 2024-07-21:
A feature javavi has been helping with. It allows a USB mouse to be used as a joystick, with settings in the menu.

DPRCZ, 2024-08-04:
Done.

closed 2024-08-04 by DPRCZ