fruit-bat/pico-zxspectrum #124: Cursor key joystick issues

midicdj1000, 2024-02-24:
HI, On my zxpicopi is has the option for cursor key joystick and it works in the pico menus and the 128k menu, but when you select cursor type joystick the fire is not working because it's mapped to enter and the cursor joystick fire is expecting 0

I am just using it wrong ?

fruit-bat, 2024-02-24:
I am sure you are using it fine 🙂.  When I started out, I just used the cursor mode to control the boot menu on the 128k spectrum. I think it makes more sense for it to use 0 instead of Enter...I will swap it over and post new firmware.

fruit-bat, 2024-02-25:
Please would you tell me which firmware you are using.

midicdj1000, 2024-03-04:
HI, I checked and the person that made this board changed the firmware in a few places to help with some old VGA issues so I don't think you can compile a new firmware for me. I am also not sure if he knows how to take your new source and combine with his. so I am a bit stuck.

fruit-bat, 2024-03-04:
Maybe ask the supplier to pull-request his changes... otherwise it's hard to help.