fruit-bat/pico-zxspectrum #126: Specials keys row 1 Auk Firmware

nano54120, 2024-03-18:
Hi, I am using ZxSpectrumPicomputerVgaAukBob.uf2 firmware and the 8 specials keys in row 1 are not working completely. I can't reboot to 48k or 128k. And how do I go back in the menu?

I tested all the wiring and everything is correct. Only the 48k and 128k keys does not work.

Thank you for yours answers.

fruit-bat, 2024-03-19:
Shift-ENTER to go back in the menu.

I think Bob asked me to change the 128k and 48k buttons to next/prev snapshot... though I am struggling to find the conversation.

Shift-Menu is quick-save.

If you are able to build your own firmware the mappings are in here: https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/picomputer/pzx_keyscan.cpp

nano54120, 2024-03-21:
Sorry but I won't be able to. But thank you very much. I'm not stuck in the menu anymore. And it doesn't matter for the 48k and 128k keys, I'll adapt. Are there other actions with the keyboard ?

keypink66, 2024-07-08:
Hi, I also use the ZxSpectrumPicomputerVgaAukBob.uf2 firmware but I'm stuck on the home screen of the ZX 128. The keys on the PCB don't work (only down), and the keys in the first row where cis are also the arrow keys (rubber keyboard of the Spectrum with new membrane), do not work.
Any suggestions on how I can navigate the menus? Thank you

fruit-bat, 2024-07-08:
Make sure you have recent firmware from this site. Do any other keys work on the membrane,  like Enter?

Plug in a USB keyboard and press F1... just to see if it works at all.

keypink66, 2024-07-09:
> Make sure you have recent firmware from this site. Do any other keys work on the membrane, like Enter?
> 
> Plug in a USB keyboard and press F1... just to see if it works at all.

Hello and thanks for the reply!
From a first test I did with the Spectrum keyboard, only the left keys seem to work. I already tried a USB keyboard but it didn't work, green light on for a moment but then nothing! Maybe because I powered the PBC with a USB cable connected to the PC? Could it be that the power supply is too low, considering that the PC's power supply is just 0.5 A? Tonight I'll try again with another keyboard and I'll update you.
In the meantime, can you tell me which updated firmware I can download?

fruit-bat, 2024-07-09:
Post a picture(s) of your board so I can recommend firmware

fruit-bat, 2024-07-10:
Ok, the is the page: https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/pico_zx48_128.md

And you should be able to use either of:
https://github.com/fruit-bat/pico-zxspectrum/raw/main/uf2/ZxSpectrumPicomputerVgaAukBob_720x576x50Hz.uf2
https://github.com/fruit-bat/pico-zxspectrum/raw/main/uf2/ZxSpectrumPicomputerVgaAukBob_640x480x60Hz.uf2

(As an aside, I'm not sure you have wired up D+ and D- for the USB host port; see near the end of the Pi Pico. Solder missing from the power pins on the on 'USB Power')

Powering from the PC should be fine. I run mine and 2 keyboard a mouse and 2 joysticks from a PC port and seems OK... but can't hurt trying another power supply if you have one.

keypink66, 2024-07-11:
Yes, you see well! D- and D+ are not wired because I have not found any indication on how they should be wired. Can you tell me how I should connect them? Maybe this is why the USB keyboard is not detected?

fruit-bat, 2024-07-11:
> Yes, you see well! D- and D+ are not wired because I have not found any indication on how they should be wired. Can you tell me how I should connect them? Maybe this is why the USB keyboard is not detected?

There are 2 test pads on the bottom of the Pi Pico. I think TP2 is D- and TP3 is D+ (these are tricky to solder to!). It's possible the PCB is arranged so contact is already made. Hard for me to see from the picture. If you have a continuity tester you could plug in a USB cable to the Pico and check the connections are made to the USB hub. 

Is the keyboard working any better with the new firmware? Is it all of the left keys that work (1-5, Q-T, A-G, CS-V)?

See https://components101.com/development-boards/raspberry-pi-pico-pinout-datasheet-specifications

keypink66, 2024-07-15:
> There are 2 test pads on the bottom of the Pi Pico. I think TP2 is D- and TP3 is D+ (these are tricky to solder to!). It's possible the PCB is arranged so contact is already made. Hard for me to see [...]
[image]

Thank you! You can't imagine the emotion of seeing this screen... and above all thanks to your instructions!

So, the 2 jumpers, as you pointed out, were not soldered. The YT video I used was very fast, and I hadn't seen that passage! :(

I soldered D- and D+ and flashed the fw you suggested, and... everything works. Spectrum keys and keys on the PCB, now everything is working correctly!

All that remains is the USB port which still doesn't show a keyboard or joystick, but that's okay for the moment...feeling the rubber buttons under your fingers is always a thrill! :)

Thanks again for your advice, this was my first board built by me and it was double satisfaction. Now I just need to find a way to cut the plastic on the back and shape the case. Can you suggest me a method?

Mariano

fruit-bat, 2024-07-15:
Awe, it's looking good! Glad to see it working. 

I used a dremel to cut my case.... and it was a bit messy.  Probably the best solution is a steel ruler,  a Stanley knife and lots of patience.

keypink66, 2024-07-15:
Yes, I too had thought of a ruler and a Stanley type knife, and I am
comforted to hear from you that it is a good idea!  ;)

keypink66, 2024-07-30:
[image]

Here's my latest effort..cutting the case! It looks like a good result and, I can assure you, it was harder than soldering the components..even if it was my first time with a circuit kit! 😅

fruit-bat, 2024-07-30:
Looks very tidy :-)

keypink66, 2025-02-07:
Hello, and welcome back! :) I haven't followed the threads for a while and
wanted to know if the problem of the volume being too low has been solved.
Thanks in advance
Mariano

fruit-bat, 2025-02-07:
Yes, it may be a little better now. I use a PIO  block to generate the PWM and the sound is clearer and a little louder. Definitely worth  trying a newer firmware.  Perhaps take a copy of your old one in case you don't like 🤔

keypink66, 2025-02-07:
Perfect! Can you post me the link of the new firmware?

 Thanks
Mariano

keypink66, 2025-04-07:
Hello and congratulations for all the things you are implementing on this
fabulous project!

I tried some newer firmwares for the audio issue, but I had stability
issues and finally reloaded the old one:

*ZxSpectrumPicomputerVgaAukBob_640x480x60Hz*

Can you please suggest me a firmware that generates a clearer and louder
sound?

Thanks in advance

fruit-bat, 2025-04-07:
> Hello and congratulations for all the things you are implementing on this
> fabulous project!
>
> I tried some newer firmwares for the audio issue, but I had stability
> issues and finally reloaded the old [...]

These are from the upcoming builds. I take you are using an RP2040, if so give these a go.

[sample_uf2.zip](https://github.com/user-attachments/files/19638189/sample_uf2.zip)

I've have just tested them and the extra keys are working ok for me. Let me know how you get on.

keypink66, 2025-04-09:
> Sì, ho costruito questa scheda con il Raspberry Pi Pico...credo sia il
> RP2040, puoi confermare?
>
>
>
> Il giorno lun 7 apr 2025 alle ore 21:36 fruit-bat <

Yes, I built this board with the Raspberry Pi Pico..I think it's the
RP2040, can you confirm?

fruit-bat, 2025-04-09:
> Yes, I built this board with the Raspberry Pi Pico..I think it's the
> RP2040, can you confirm?
> 
> 
> > Sì, ho costruito questa scheda con il Raspberry Pi Pico...credo sia il
> > RP2040, puoi confermare?
> >
> >
> >
> > Il giorno lun 7 apr 2025 alle ore 21:36 fruit-bat <
> >
> >> Hello and congratulations for all the things you are implementing on this
> >> fabulous project!
> >>
> >> I tried some newer firmwares for the audio issue, but I had stability
> >> issues and finally reloaded the old one:
> >>
> >> *ZxSpectrumPicomputerVgaAukBob_640x480x60Hz*
> >>
> >> Can you please suggest me a firmware that generates a clearer and louder
> >> sound?
> >>
> >> Thanks in advance
> >>
> >> Il giorno ven 7 feb 2025 alle ore 13:49 fruit-bat *@*.***>
> >> … <#m_2910970370617046105_m_-2727073579826888460_m_-8549586496744451430_>
> >>
> >> These are from the upcoming builds. I take you are using an RP2040, if so
> >> give these a go.
> >>
> >> sample_uf2.zip
> >>
> >> I've have just tested them and the extra keys are working ok for me. Let
> >> me know how you get on.
> >>
> >> (fruit-bat/pico-zxspectrum#126)
> >>
> >> Hello and congratulations for all the things you are implementing on this
> >> fabulous project!
> >>
> >> I tried some newer firmwares for the audio issue, but I had stability
> >> issues and finally reloaded the old one:
> >>
> >> *ZxSpectrumPicomputerVgaAukBob_640x480x60Hz*
> >>
> >> Can you please suggest me a firmware that generates a clearer and louder
> >> sound?
> >>
> >> Thanks in advance
> >>
> >> Il giorno ven 7 feb 2025 alle ore 13:49 fruit-bat *@*.***>
> >> … <#m_2910970370617046105_m_-2727073579826888460_m_-8549586496744451430_>
> >>
> >> These are from the upcoming builds. I take you are using an RP2040, if so
> >> give these a go.
> >>
> >> sample_uf2.zip
> >>
> >> I've have just tested them and the extra keys are working ok for me. Let
> >> me know how you get on.

Yup, that's correct