fruit-bat/pico-zxspectrum #130: Will this software work with RP2040-PICO-PC?

rog77, 2024-04-05:
For details https://www.olimex.com/Products/MicroPython/RP2040-PICO-PC/open-source-hardware

I am assuming it would need to use USB-OTG on the actual PICO board, as the USB on that carrier board just does power.

Thanks!

fruit-bat, 2024-04-05:
Yes to the OTG bit. And probably to will the software work. It will need a new build configuration.  Happy, to help out if you don't mind testing ... but be aware it may take multiple attempts.

ikjordan, 2024-04-05:
If the audio is sent over the HDMI pins it should be ok - but if you want audio over the 3.5mm socket then to quote the Readme from the version of PicoDVI that this project now uses:

_Stereo PWM sound for Olimexpc. The supplied PicoDVI code from Olimex uses PWM to generate the DVI clock on pins 12 and 13. The boards audio output is on pins 27 and 28. Pin 28 and pin 12 are on the same PWM channel (6A), so it is not possible to configure different PWM clock frequencies for the two pins. In effect this means that PWM stereo sound does not work with the supplied Olimex PicoDVI code. The code in this repo can optionally generate the DVI clock using a very simple PIO state machine, so that pins 27 and 28 can then be driven to generate PWM audio. To enable this set DVI_USE_PIO_CLOCK_

Note also that by default the RP2040-PICO-PC does not put 5V on  HDMI pin 18, so it is not detected by all TVs (it seems fine with monitors). If the board is not detected by your TV then soldering the jumpers on the board shown here will put 5V across Pin 18.
[image]

ikjordan, 2024-09-20:
With the submission of #173 this question can now be answered "Yes" and closed as completed :-)

rog77, 2024-09-22:
Thank you for the update, and your efforts. You have a great project going on here!

closed 2024-09-22 by rog77