fruit-bat/pico-zxspectrum #133: USB game controllers which don't work with emulator

tankist-git-2, 2024-04-14:
I provide two descriptions on USB game controllers which don't work with the emulator. The first one is Super Nintendo (chineese version) which hangs emulator immediatelly. The second one is Microsoft XBox game controller which do nothing in games. The only one which work fine for me is just Sony Playstation 3 game controller.

SNES game controller (https://www.brown.edu/Departments/Joukowsky_Institute/courses/13things/files/5980035.jpg)

```
Device USB\VID_0810&PID_E501\5&389ab2c2&0&7 configured.
Driver Name: input.inf
GUID Class: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.3636
Vendor: Microsoft
Driver Section: HID_Inst.NT
Driver Rank: 0xFF3202
Device ID: USB\Class_03
Low rank Drivers: 
Device has been updated: false
Parent Device: USB\ROOT_HUB30\4&33974a7c&0&0

Device HID\VID_0810&PID_E501\6&2e17453f&0&0000 configured.
Driver Name: input.inf
GUID Class: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.3636
Vendor: Microsoft
Driver Section: HID_Raw_Inst.NT
Driver Rank: 0xFF1003
Device ID: HID_DEVICE_SYSTEM_GAME
Low rank Drivers: input.inf:HID_DEVICE:00FF1005
Device has been updated: false
Parent Device: USB\VID_0810&PID_E501\5&389ab2c2&0&7
```

--------------------------------------------------------------
Microsoft XBox game controller (https://m.media-amazon.com/images/I/61ja-KkZjxL.jpg)

```
Device USB\VID_045E&PID_028E&IG_00\6&e794f6e&0&00 configured.
Driver Name: input.inf
GUID Class: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.3636
Vendor: Microsoft
Driver Section: HID_Inst.NT
Driver Rank: 0xFF3202
Device ID: USB\Class_03
Low rank Drivers: 
Device has been updated: false
Parent Device: USB\VID_045E&PID_028E\0449C11

Device HID\VID_045E&PID_028E&IG_00\7&fde769f&0&0000 configured.
Driver Name: input.inf
GUID Class: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.3636
Vendor: Microsoft
Driver Section: HID_Raw_Inst.NT
Driver Rank: 0xFF1002
Device ID: HID_DEVICE_SYSTEM_GAME
Low rank Drivers: input.inf:HID_DEVICE:00FF1004
Device has been updated: false
Parent Device: USB\VID_045E&PID_028E&IG_00\6&e794f6e&0&00
```

DPRCZ, 2024-04-17:
https://forums.libretro.com/t/solved-super-snes-joypads-dont-work-after-x-seconds/14709

fruit-bat, 2024-04-17:
Thanks for these. I'm currently working on more 50hz support, but will try to look at these soon.

DPRCZ, 2024-05-10:
I  did some research in USB game controllers.
[image]

It ended by creatting my own gamepad and after few day of  missery found the problem.  Pico-zxspectrum recognize as joystick HID descriptor with usage 4 - joystick and  not the usage 5 - gamepad :-( 

Sample of good HID descriptor:
code const uint8_t HidRepDesc[] = {
	0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
	0x09, 0x04,                    // USAGE (Joystick)
	0xa1, 0x01,                    // COLLECTION (Application)
	0x09, 0x01,                    //   USAGE (Pointer)
	0xa1, 0x00,                    //   COLLECTION (Physical)
	0x09, 0x30,                    //     USAGE (X)
	0x09, 0x31,                    //     USAGE (Y)
	0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
	0x26, 0xff, 0x00,              //     LOGICAL_MAXIMUM (255)
	0x75, 0x08,                    //     REPORT_SIZE (8)
	0x95, 0x02,                    //     REPORT_COUNT (2)
	0x81, 0x02,                    //     INPUT (Data,Var,Abs)
	0xc0,                          //   END_COLLECTION
	0x05, 0x09,                    //   USAGE_PAGE (Button)
	0x19, 0x01,                    //   USAGE_MINIMUM (Button 1)
	0x29, 0x01,                    //   USAGE_MAXIMUM (Button 1)
	0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
	0x25, 0x01,                    //   LOGICAL_MAXIMUM (1)
	0x75, 0x01,                    //   REPORT_SIZE (1)
	0x95, 0x01,                    //   REPORT_COUNT (1)
	0x81, 0x02,                    //   INPUT (Data,Var,Abs)
	0x75, 0x01,                    //   REPORT_SIZE (1)
	0x95, 0x07,                    //   REPORT_COUNT (7)
	0x81, 0x03,                    //   INPUT (Const,Var,Abs)
	0xc0                           // END_COLLECTION (application)
};

Non recognized gemepad ...
code const uint8_t HidRepDesc[] = {
	0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
	0x09, 0x05,                    // USAGE (Gamepad)
	0xa1, 0x01,                    // COLLECTION (Application)
	0x09, 0x01,                    //   USAGE (Pointer)
	0xa1, 0x00,                    //   COLLECTION (Physical)
	0x09, 0x30,                    //     USAGE (X)
	0x09, 0x31,                    //     USAGE (Y)
	0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
	0x26, 0xff, 0x00,              //     LOGICAL_MAXIMUM (255)
	0x75, 0x08,                    //     REPORT_SIZE (8)
	0x95, 0x02,                    //     REPORT_COUNT (2)
	0x81, 0x02,                    //     INPUT (Data,Var,Abs)
	0xc0,                          //   END_COLLECTION
	0x05, 0x09,                    //   USAGE_PAGE (Button)
	0x19, 0x01,                    //   USAGE_MINIMUM (Button 1)
	0x29, 0x01,                    //   USAGE_MAXIMUM (Button 1)
	0x15, 0x00,                    //   LOGICAL_MINIMUM (0)
	0x25, 0x01,                    //   LOGICAL_MAXIMUM (1)
	0x75, 0x01,                    //   REPORT_SIZE (1)
	0x95, 0x01,                    //   REPORT_COUNT (1)
	0x81, 0x02,                    //   INPUT (Data,Var,Abs)
	0x75, 0x01,                    //   REPORT_SIZE (1)
	0x95, 0x07,                    //   REPORT_COUNT (7)
	0x81, 0x03,                    //   INPUT (Const,Var,Abs)
	0xc0                           // END_COLLECTION (application)
};

fruit-bat, 2024-05-11:
@DPRCZ Many thanks for the suggested change. If you tell me which firmware you use I will build you a version to test (before I release it more widely).

DPRCZ, 2024-05-11:
Thank you, but I have already tested it on my board.

fruit-bat, 2024-05-11:
> I provide two descriptions on USB game controllers which don't work with the emulator. The first one is Super Nintendo (chineese version) which hangs emulator immediatelly. The second one is [...]

I think at least the X-box controller looks to be the same problem sorted by DPRCZ...
https://github.com/nefarius/ViGEmBus/issues/40

I've pushed some new builds. Please let me know how you get on.

fruit-bat, 2024-05-11:
> Thank you, but I have already tested it on my board.

Fantastic,  I will release a set of firmware.

tankist-git-2, 2024-05-12:
I tried my two joysticks again with the new firmware. There are no changes since the last test: Super Nintendo joystick hangs  the board, X-Box joystick do nothing. PS joystick work as before (a little bit strange because of using analog pad instead of discrete)

DPRCZ, 2024-05-12:
According https://github.com/hathach/tinyusb/discussions/1929  XBOX gamepad is not standart HID device.

tankist-git-2, 2024-05-15:
You should understand Xbox controller is not my goal. It's expensive and hard-to-get controllers. It would be great if you add support for cheap SNES controllers.

DPRCZ, 2024-05-17:
@tankist-git-2 
[image]
Try this.

DPRCZ, 2024-05-23:
@fruit-bat in tusb_config.h change:
#define CFG_TUH_ENUMERATION_BUFSIZE 256
to
#define CFG_TUH_ENUMERATION_BUFSIZE 512
It is necessary for long descriptors like PS4 gamepad has.

fruit-bat, 2024-05-27:
@DPRCZ I've made the change and released a set of firmware. Many thanks.