fruit-bat/pico-zxspectrum #135: Large list of files

LiCaNtRoPo-PCB, 2024-04-27:
Hi, when you put a large amount of files (.z80 or .tzx) in their respectives folders it doesn´t appear in the menu. Looks like handling long lists of files doesn´t work properly, maybe it is needed to read names in chunks?

Gringodingo, 2024-05-07:
It does work for me. No sub folders. 4000 plus files. Use the reindex option and be prepared to wait an hour or two for indexing to complete

fruit-bat, 2024-05-11:
> Hi, when you put a large amount of files (.z80 or .tzx) in their respectives folders it doesn´t appear in the menu. Looks like handling long lists of files doesn´t work properly, maybe it is needed to read names in chunks?

Which firmware are you using?

LiCaNtRoPo-PCB, 2024-07-07:
Ok, reindex worked for me. Took some time to do it but I got a weird behaviour when I power up the board with that microSD card, black screen and takes sometime (depends on how many files do you have into the folder) to appears the menu on screen.

> Which firmware are you using?
I tried each update you made on github, BTW last one I get: 
"hid_host.c:586:49: error: 'HID_PROTOCOL_DEFAULT' undeclared (first use in this function); did you mean 'HID_PROTOCOL_REPORT'?"

fruit-bat, 2024-07-07:
Sorry, my bad should be corrected now. You need to pull:
* tinusb
* pico-zxspectrum
* pico-emu-utils

I've just merged some code to emulate a 'Kempston mouse', but I have not got round to doing all of the builds so here is a version just for you! Let me know if this is any more stable. 

[ZxSpectrumBreadboardHdmiAudio.uf2.zip](https://github.com/user-attachments/files/16119774/ZxSpectrumBreadboardHdmiAudio.uf2.zip)

On boot the emulator looks for a file called .dcache and if it cannot find one creates it. Perhaps there is some problem writing this to the SD card and it does it on every power up(?). Mine is working ok. It will, be interesting to see if the problem is difference between my build and yours.. or maybe sd-card related.

You can use sub-folders if you want to.

LiCaNtRoPo-PCB, 2024-07-14:
Hi again, I tried the version you uploaded (thanks so much) and still having the same issue. Main thing is everytime at boot it tries to create a new .dcache file onto tapes (500 files, mixed .tzx and .tap) folder. The .dcache created for snapshots (about 5000 .z80 files) it´s ok and remains the same on every boot.
Also tried different microsd brands and capacities, all the same results.

fruit-bat, 2024-07-14:
I will have a look and see if I have done something silly. It is strange it rewrites one and not the other.

fruit-bat, 2024-07-14:
So I've had a play and my breadboard version is not re-writing the .dcache files.
Maybe turn on the debug in FatFsDirCache.cpp (by commenting the define) and see what arrives over the serial port.
Maybe we need a little delay before the initial attempt to mount the SD card to let things settle down (in main.cpp)?

LiCaNtRoPo-PCB, 2024-07-21:
Enabled debug into FastFsDirCache.cpp and nothing happens on serial port, silent.... (looks like debug isn´t enabled) until it finish writing .dcache file, then "Configuring DVI" and "Core1 Start" appears. Strange behaviour... I tried to put a small delay just before mounting SD Card and same results. Thought that long filenames with [(& characters could be a problem and remove those from files. No luck

fruit-bat, 2024-07-21:
Do you get correct behaviour with just one file in tapes and one file in snapshots?

I'm wondering if the Pico is running out of memory and basically just going mad, or I have made a logical in the startup (which is easy to do as the class constructors can trigger a whole bunch of activities).

LiCaNtRoPo-PCB, 2024-07-21:
I can send you both files, strange thing is that inside appears names or trace names that they doesn´t exists into folder, tried many microsd with some different partitions, etc... The issue only appears with tapes. Deleted all files in tapes folder and board startup quickly.

fruit-bat, 2024-07-21:
Check the date on the .cache or rename the tape file and see if the name changes in the menu.

LiCaNtRoPo-PCB, 2024-07-21:
Yes, it changes the name into menu.

fruit-bat, 2024-07-21:
OK, I've figured out what I have done wrong. The problem is in PicoExplorer from the pico-dvi-menu library. I'll fix it but i may need to think about it, bear with me. Thanks for helping me find the bug.

fruit-bat, 2024-07-21:
You need to pull:
* pico-dvi-menu
* pico-zxspectrum

(both on the main branch)

Hopefully, this will sort it out. Please let me know how you get on.