fruit-bat/pico-zxspectrum #136: Page up and page down option

Gringodingo, 2024-05-07:
Would be helpful to have a page up and down option on the rubber keyboard! Navigation up and down of hundred of files difficult 1 at a time. A simple first letter search would also be ideal.

Gringodingo, 2024-05-07:
I now see you have a key mapping file. I'll look at this tomorrow but it looks pretty complicated!

fruit-bat, 2024-05-11:
What about CAPS-4 for PageDown and CAPS-9 for PageUp?

[ZxSpectrumPicomputerVgaAukBob.zip](https://github.com/fruit-bat/pico-zxspectrum/files/15283371/ZxSpectrumPicomputerVgaAukBob.zip)

fruit-bat, 2024-05-11:
...and these have a simple first letter search.

Let me know if you like it... then I can release a set of builds.

[ZxSpectrumPicomputerVgaAukBob.zip](https://github.com/fruit-bat/pico-zxspectrum/files/15283764/ZxSpectrumPicomputerVgaAukBob.zip)

fruit-bat, 2024-05-14:
All sorted. Re open if you are still having problems

closed 2024-05-14 by fruit-bat